webcamoid/webcamoid #21: plasmoid-webcamoid-5.0.0b2-1.2.x86_64

Soyasi, 2015-03-05:
Application: plasma-desktop (4.11.16)
KDE Platform Version: 4.14.5
Qt Version: 4.8.6
Operating System: Linux 3.16.7-7-desktop x86_64
Distribution: "openSUSE 13.2 (Harlequin) (x86_64)"

-- Information about the crash:
- What I was doing when the application crashed:

plasmoid-webcamoid-5.0.0b2-1.2.x86_64
plasmoid-webcamoid-devel-5.0.0b2-1.2.x86_64.rpm

tengo 2 webcam
have 2 webcam

1º Bus 004 Device 002: ID 093a:2468 Pixart Imaging, Inc. SoC PC-Camera
2º Bus 001 Device 003: ID 046d:0825 Logitech, Inc. Webcam C270

cuando uso la ID 046d:0825 no tengo problemas
cuando uso la ID 093a:2468 se reinicia el escritorio y se redimensiona el plasmoide

when I use the ID 046d: 0825 I have no problems
when I use the 093rd ID: 2468 desktop restarts and resizes the plasmoid

-- Backtrace:
Application: Shell de escritorio Plasma (plasma-desktop), signal: Aborted
Using host libthread_db library "/lib64/libthread_db.so.1".
[Current thread is 1 (Thread 0x7ff4d5dba780 (LWP 1740))]

Thread 14 (Thread 0x7ff4ba52a700 (LWP 1749)):
#0  0x00007ff4d33d8c3d in poll () at /lib64/libc.so.6
#1  0x00007ff4d04ebbe4 in  () at /usr/lib64/libglib-2.0.so.0
#2  0x00007ff4d04ebf0a in g_main_loop_run () at /usr/lib64/libglib-2.0.so.0
#3  0x00007ff4bde4b576 in  () at /usr/lib64/libgio-2.0.so.0
#4  0x00007ff4d0510b85 in  () at /usr/lib64/libglib-2.0.so.0
#5  0x00007ff4d46740a4 in start_thread () at /lib64/libpthread.so.0
#6  0x00007ff4d33e106d in clone () at /lib64/libc.so.6

Thread 13 (Thread 0x7ff4b9d29700 (LWP 1750)):
#0  0x00007ff4d33d8c3d in poll () at /lib64/libc.so.6
#1  0x00007ff4d04ebbe4 in  () at /usr/lib64/libglib-2.0.so.0
#2  0x00007ff4d04ebcec in g_main_context_iteration () at /usr/lib64/libglib-2.0.so.0
#3  0x00007ff4d04ebd29 in  () at /usr/lib64/libglib-2.0.so.0
#4  0x00007ff4d0510b85 in  () at /usr/lib64/libglib-2.0.so.0
#5  0x00007ff4d46740a4 in start_thread () at /lib64/libpthread.so.0
#6  0x00007ff4d33e106d in clone () at /lib64/libc.so.6

Thread 12 (Thread 0x7ff497ec4700 (LWP 1785)):
#0  0x00007ff4d467aa9d in read () at /lib64/libpthread.so.0
#1  0x00007ff4d052c750 in  () at /usr/lib64/libglib-2.0.so.0
#2  0x00007ff4d04eb714 in g_main_context_check () at /usr/lib64/libglib-2.0.so.0
#3  0x00007ff4d04ebb7b in  () at /usr/lib64/libglib-2.0.so.0
#4  0x00007ff4d04ebcec in g_main_context_iteration () at /usr/lib64/libglib-2.0.so.0
#5  0x00007ff4d4a390de in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib64/libQtCore.so.4
#6  0x00007ff4d4a0ae6f in QEventLoop::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib64/libQtCore.so.4
#7  0x00007ff4d4a0b165 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib64/libQtCore.so.4
#8  0x00007ff4d49080bf in QThread::exec() () at /usr/lib64/libQtCore.so.4
#9  0x00007ff4d49ec783 in  () at /usr/lib64/libQtCore.so.4
#10 0x00007ff4d490a79f in  () at /usr/lib64/libQtCore.so.4
#11 0x00007ff4d46740a4 in start_thread () at /lib64/libpthread.so.0
#12 0x00007ff4d33e106d in clone () at /lib64/libc.so.6

Thread 11 (Thread 0x7ff496e8d700 (LWP 1788)):
#0  0x00007ff4d467805f in pthread_cond_wait@@GLIBC_2.3.2 () at /lib64/libpthread.so.0
#1  0x00007ff4c6ef18cb in  () at /usr/lib64/libQtScript.so.4
#2  0x00007ff4c6ef1909 in  () at /usr/lib64/libQtScript.so.4
#3  0x00007ff4d46740a4 in start_thread () at /lib64/libpthread.so.0
#4  0x00007ff4d33e106d in clone () at /lib64/libc.so.6

Thread 10 (Thread 0x7ff4976c3700 (LWP 10693)):
[... 126 lines cut ...]
#38 0x00007ff4d57afcea in KApplication::notify(QObject*, QEvent*) () at /usr/lib64/libkdeui.so.5
#39 0x00007ff4d4a0c2ad in QCoreApplication::notifyInternal(QObject*, QEvent*) () at /usr/lib64/libQtCore.so.4
#40 0x00007ff4d417fa07 in QGraphicsView::mouseReleaseEvent(QMouseEvent*) () at /usr/lib64/libQtGui.so.4
#41 0x00007ff4d3bd7c0a in QWidget::event(QEvent*) () at /usr/lib64/libQtGui.so.4
#42 0x00007ff4d3f745ee in QFrame::event(QEvent*) () at /usr/lib64/libQtGui.so.4
#43 0x00007ff4d4180429 in QGraphicsView::viewportEvent(QEvent*) () at /usr/lib64/libQtGui.so.4
#44 0x00007ff4d4a0c416 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) () at /usr/lib64/libQtCore.so.4
#45 0x00007ff4d3b8874c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib64/libQtGui.so.4
#46 0x00007ff4d3b8edea in QApplication::notify(QObject*, QEvent*) () at /usr/lib64/libQtGui.so.4
#47 0x00007ff4d57afcea in KApplication::notify(QObject*, QEvent*) () at /usr/lib64/libkdeui.so.5
#48 0x00007ff4d4a0c2ad in QCoreApplication::notifyInternal(QObject*, QEvent*) () at /usr/lib64/libQtCore.so.4
#49 0x00007ff4d3b8e5e3 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) () at /usr/lib64/libQtGui.so.4
#50 0x00007ff4d3bffc9b in  () at /usr/lib64/libQtGui.so.4
#51 0x00007ff4d3bfe70c in QApplication::x11ProcessEvent(_XEvent*) () at /usr/lib64/libQtGui.so.4
#52 0x00007ff4d3c255c2 in  () at /usr/lib64/libQtGui.so.4
#53 0x00007ff4d04eba04 in g_main_context_dispatch () at /usr/lib64/libglib-2.0.so.0
#54 0x00007ff4d04ebc48 in  () at /usr/lib64/libglib-2.0.so.0
#55 0x00007ff4d04ebcec in g_main_context_iteration () at /usr/lib64/libglib-2.0.so.0
#56 0x00007ff4d4a390be in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib64/libQtCore.so.4
#57 0x00007ff4d3c25676 in  () at /usr/lib64/libQtGui.so.4
#58 0x00007ff4d4a0ae6f in QEventLoop::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib64/libQtCore.so.4
#59 0x00007ff4d4a0b165 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib64/libQtCore.so.4
#60 0x00007ff4d4a105b9 in QCoreApplication::exec() () at /usr/lib64/libQtCore.so.4
#61 0x00007ff4c09a2b94 in kdemain () at /usr/lib64/libkdeinit4_plasma-desktop.so
#62 0x0000000000408815 in _start ()

hipersayanX, 2015-03-05:
You are testing a 2 years old release, please test latest version:

https://github.com/hipersayanX/webcamoid/releases
https://github.com/hipersayanX/webcamoid/wiki/Build-and-install-the-RPM-package

hipersayanX, 2015-04-11:
No response, this issue is about an old release anyway, closing.

closed 2015-04-11 by hipersayanX