webcamoid/webcamoid #22: crash on take photo

zhcj, 2015-04-11:
webcamoid version:6.0.0
distribution:gentoo unstable
QMake version:3.0
Qt version:5.4.1
gcc version:4.9.2

error information on take a photo:
linux@gentoo ~/下载/webcamoid-master $ webcamoid 
v 11866.80 A-V: -11865.915 q=1
QInotifyFileSystemWatcherEngine::addPaths: inotify_add_watch failed: 没有那个文件或目录
v 11866.97 A-V:   0.084 q=1
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
kf5.kservice.sycoca: Trying to open ksycoca from "/home/linux/.cache/ksycoca5"
Invalid URL:  QUrl( "照片2015-04-11 23-08-34.png" ) 
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
"Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
v 11867.09 A-V:   0.025 q=1
We got some errors while running testparm "Load smb config files from /etc/samba/smb.conf
Loaded services file OK.
ERROR: lock directory /var/lock/samba does not exist
"
v 11867.20 A-V:   0.015 q=1
Invalid URL:  QUrl( "照片2015-04-11 23-08-34.png" ) 
v 11867.33 A-V:  -0.005 q=1
v 11867.47 A-V:  -0.009 q=1
v 11867.59 A-V:   0.024 q=1
......

hipersayanX, 2015-04-11:
The warning:

    Invalid URL: QUrl( "照片2015-04-11 23-08-34.png" )

Isn't harmful at all, but I'll fix it.

    v 11867.20 A-V: 0.015 q=1
    Invalid URL: QUrl( "照片2015-04-11 23-08-34.png" )
    v 11867.33 A-V: -0.005 q=1
    v 11867.47 A-V: -0.009 q=1
    v 11867.59 A-V: 0.024 q=1
    ......

After this line, the program die with a segfault?

hipersayanX, 2015-04-11:
I can't reproduce the issue, but I remember the *Invalid URL* message, it was just a warning.
It literally crashes, ends with a segfault or similar? or its just that warning?

zhcj, 2015-04-12:
v 11867.20 A-V: 0.015 q=1
Invalid URL: QUrl( "照片2015-04-11 23-08-34.png" )
v 11867.33 A-V: -0.005 q=1
v 11867.47 A-V: -0.009 q=1
v 11867.59 A-V: 0.024 q=1
......

After this line,I use Ctrl+c to kill it.Because the console will show many infomation like "v 11867.59 A-V: 0.024 q=1",and the camera is still run,but program will stop at save photo dialog status.

The probolem is not about these line?

We got some errors while running testparm "Load smb config files from /etc/samba/smb.conf
Loaded services file OK.
ERROR: lock directory /var/lock/samba does not exist

hipersayanX, 2015-04-12:
Hi, can you test git version? I have disabled the sync log by default.
About the messages:

    QInotifyFileSystemWatcherEngine::addPaths: inotify_add_watch failed: 没有那个文件或目录
    "Interpolation {$[~stripAccel %1]} in {$[~stripAccel %1]} failed: Unregistered call to '~stripAccel'."
    We got some errors while running testparm "Load smb config files from /etc/samba/smb.conf
    Loaded services file OK.
    ERROR: lock directory /var/lock/samba does not exist

All these warning comes from your Plasma 5 installation, Plasma 5 is still buggy. Can you test with another DE?

zhcj, 2015-04-12:
I tested the newest git version,all the same probolem is also occured.Maybe this is kde5/plasma5's bug,I'll try some other DE later.

hipersayanX, 2015-06-26:
Hi, do you still have this bug?

zhcj, 2015-06-28:
OK!! I tried again just now,the bug is disappered.
Thank you very much ! 
webcamoid version:6.1.0
distribution:gentoo unstable
QMake version:3.0
Qt version:5.4.2
gcc version:5.1.0

zhcj, 2015-06-28:
Next problem i found is the install commond is wrong:
"make INSTALL_ROOT=/usr install"
in my machine this will install the program to /usr/usr,so i use this commond to be successed:
"make install"

hipersayanX, 2015-06-28:
INSTALL_ROOT is ok, it must point always to the literal root partition/folder install path, if you want to change the install prefix, let say for example to /opt, you can do:

    qmake-qt5 PREFIX=/opt Webcamoid.pro
    make
    make install

Other possible configuration options are listed in [commons.pri](https://github.com/hipersayanX/webcamoid/blob/master/commons.pri).

If you are creating a package, use INSTALL_ROOT to indicate the package folder, see [here](https://github.com/hipersayanX/webcamoid/blob/master/ports/archlinux/PKGBUILD).

hipersayanX, 2016-02-25:
As far as i tested, this bug is not present in latest release, closing.

closed 2016-02-25 by hipersayanX