webcamoid/webcamoid #23: Segmentation fault (Chakra Linux)

bravog, 2015-05-16:
I'm trying to update Webcamoid on CCR. It compiles and installation went fine, but it's not working. All I can get is "Segmentation fault (core dumped)".

Version is 6.0.0-3
Chakra Euler, kernel 3.19.4
Qt 4.8.6, qt5-base 5.4.1
Gcc 4.9.1

hipersayanX, 2015-05-16:
Hi, can you show me the PKGBUILD?

bravog, 2015-05-16:
# Maintainer: shadow <sombra AT openmailbox DOT org>
# AUR Maintainer: Gonzalo Exequiel Pedone <hipersayan DOT x AT gmail DOT com>

pkgname=webcamoid
pkgver=6.0.0
pkgrel=3
pkgdesc="Webcamoid is a full featured webcam capture application."
url='https://github.com/hipersayanX/webcamoid'
license=('GPL')
arch=('x86_64')
depends=('qt5-quickcontrols' 'qt5-multimedia' 'qt5-graphicaleffects' 'qt5-svg' 'ffmpeg' 'opencv')
makedepends=('qt5-tools')
provides=('webcamoid')
conflicts=('kdeplasma-applets-webcamoid' 'webcamoid-git')
source=("https://github.com/hipersayanX/${pkgname}/archive/${pkgver}.tar.gz")
md5sums=('866ac8f0243ee54c30e997448eb44e8a')

build() {
    cd "$srcdir/${pkgname}-${pkgver}"
    qmake-qt5 Webcamoid.pro
    make
}

package() {
    cd "$srcdir/${pkgname}-${pkgver}"
    make INSTALL_ROOT="${pkgdir}" install
}

hipersayanX, 2015-05-16:
change this line:

    qmake-qt5 Webcamoid.pro

to

    qmake-qt5 CONFIG+=debug Webcamoid.pro

and recompile. Then run webcamoid as:

    gdb webcamoid
    run

Try to do exactly the same thing you did when it crashed before.

bravog, 2015-05-16:
Starting program: /usr/bin/webcamoid 
warning: Could not load shared library symbols for linux-vdso.so.1.
Do you need "set solib-search-path" or "set sysroot"?
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[New Thread 0x7fffec3a3700 (LWP 12256)]
[New Thread 0x7fffeb751700 (LWP 12257)]

Program received signal SIGSEGV, Segmentation fault.
0x00007fffda943d4c in ?? () from /usr/lib/libQtTest.so.4

hipersayanX, 2015-05-16:
Why its linking against a Qt4 library?

    0x00007fffda943d4c in ?? () from /usr/lib/libQtTest.so.4

Can you upload the compile log? (please use [gist](https://gist.github.com/))

bravog, 2015-05-16:
Excuse my ignorance but where can I find the compile log?

hipersayanX, 2015-05-16:
The compile log is the output of the **makepkg** command (in this case), when it compile the package. Also I forgot to ask, did you did something special when it crashed? or it just crashed on start?

bravog, 2015-05-16:
OK, I'll try with makepkg --log and will paste the result. I wasn't doing anything special, it just crashed on startup.

bravog, 2015-05-16:
Build log: https://gist.github.com/bravog/78df523fefa541620da0
Package log: https://gist.github.com/bravog/70f2139fe7a1334e5702

hipersayanX, 2015-05-16:
I have found this:

https://forum.qt.io/topic/30108/mixed-qt4-and-qt5-lib-on-ubuntu-crash-my-program-i-think
http://www.qtcentre.org/threads/54158-OpenCV-Seg-fault

Try deleting the */usr/lib/Qb/libFaceDetect.so* file/plugin. And run it again.

bravog, 2015-05-16:
Ok, I'll give it a try tomorrow. Maybe the opencv included in Chakra was built against qt4.

bravog, 2015-05-17:
I did what you suggested and now it's working flawlessly. So what should I do next?

hipersayanX, 2015-05-17:
In this case, I think the best we can do is to remove the face detection plugin from build, because we have not idea if other packages will be affected if we modify the OpenCV package.

Replace your PKGBUILD with [this](https://gist.github.com/hipersayanX/f6e78955430220144b7f), and put [this file](https://gist.github.com/hipersayanX/b50d39654bc81394629e) in the same directory as the PKGBUILD.

OpenCV is causing me a lot of headaches in Debian/Ubuntu and Windows platforms as well, In this moment I'm working in an OpenCV replacement, so it probably it will be ready for the next release.

bravog, 2015-05-17:
Good job, Gonzalo. Now everything is perfect. Congrats and thanks for your work!

closed 2015-05-17 by bravog

hipersayanX, 2015-06-26:
Commit 808f4b6 fixes this issue and may fix build issues in Debian and Ubuntu.