webcamoid/webcamoid #25: Build packages with Travis

techtonik, 2015-08-17:
It is possible to use Travis to build, test and upload packages for Ubuntu. Just an idea.
http://docs.travis-ci.com/user/uploading-artifacts/

hipersayanX, 2015-08-17:
As far as I know, Travis build packages using Debian, but I don't know which repositories are available, I don't know if FFmpeg is available in Travis. Is a similar problem to OBS, they doesn't offers that very basic information.

techtonik, 2015-08-17:
Travis is just a build system. Why repositories are relevant? I thought the build scripts download dependencies and compile them regardless of repositories. Or maybe it is possible to compile the stuff statically..

hipersayanX, 2015-08-17:
>Why repositories are relevant?

Because I need to know if a certain package is available in the build system.

> I thought the build scripts download dependencies and compile them regardless of repositories.

Building a package in Travis, OBS or PPA, is exactly the same that building a package locally, you need to specify the name of all packages required for build, if a repository is missing and you require a dependency from that repository, your build will fail.

For example, Webcamoid depends on FFmpeg, but FFmpeg is banned from OBS repositories for [legal issues](https://en.opensuse.org/openSUSE:Build_Service_application_blacklist), for that reason I can't build Webcamoid using OBS. 

In the case of Debian, FFmpeg is available for *Stretch* [and higher](https://packages.debian.org/es/stretch/ffmpeg) in the main repository, I don't know if Travis provides a build environment for Debian Stretch.

hipersayanX, 2016-02-27:
Added build tests with [Travis](https://travis-ci.org/hipersayanX/webcamoid) and [OBS](https://build.opensuse.org/package/show/home:hipersayan_x/Webcamoid).

closed 2016-02-27 by hipersayanX