webcamoid/webcamoid #27: Project ERROR: libavdevice development package not found

hpfn, 2015-11-05:
Hi,

I have the libavdevice-dev installed. And also libavdevice-ffmpeg56.

What files does it is searching for ? I have these:
https://packages.debian.org/sid/amd64/libavdevice-dev/filelist

regards,
Herbert

hipersayanX, 2015-11-05:
Which is the error message?, the libavdevice-dev package is from libAV or FFmpeg? Webcamoid only supports FFmpeg, and I also working to support GStreamer right now.

hpfn, 2015-11-05:
cd Plugins/ && ( test -e Makefile || /usr/lib/x86_64-linux-gnu/qt5/bin/qmake /home/webcamoid/webcamoid-6.2.0/Qb/Plugins/Plugins.pro -o Makefile ) && make -f Makefile distclean
make[3]: Entering directory '/home/webcamoid/webcamoid-6.2.0/Qb/Plugins'
cd ACapsConvert/ && ( test -e Makefile || /usr/lib/x86_64-linux-gnu/qt5/bin/qmake /home/webcamoid/webcamoid-6.2.0/Qb/Plugins/ACapsConvert/ACapsConvert.pro -o Makefile ) && make -f Makefile distclean
Project ERROR: libavdevice development package not found

less Qb/Plugins/ACapsConvert/ACapsConvert.pro
isEmpty(FFMPEGLIBS) {
    CONFIG += link_pkgconfig

    PKGCONFIG += \
        libavdevice \
        libavfilter \
        libavformat \
        libavcodec \
        libpostproc \
        libswresample \
        libswscale \
        libavutil
}

Maybe it is looking for .pc file ?

# apt-cache search libavdevice
libavdevice-dev - FFmpeg library for handling input and output devices - development files
libavdevice-ffmpeg56 - FFmpeg library for handling input and output devices - runtime files

hipersayanX, 2015-11-05:
>Maybe it is looking for .pc file ?

Yes, that's right. That file is contained in libavdevice-dev package, did you installed **pkg-config**? If it's and still not working, you can disable pkg-config, and configure FFmpeg paths manually:

    qmake -qt=5 Webcamoid.pro \
        FFMPEGINCLUDES=/usr/include \       # The path where headers are located
        FFMPEGLIBS+=-L/usr/lib/x86_64-linux-gnu \ # The path where libraries are located
        FFMPEGLIBS+=-lavdevice \
        FFMPEGLIBS+=-lavfilter \
        FFMPEGLIBS+=-lavformat \
        FFMPEGLIBS+=-lavcodec \
        FFMPEGLIBS+=-lswresample \
        FFMPEGLIBS+=-lswscale \
        FFMPEGLIBS+=-lavutil

hpfn, 2015-11-08:
Ok, pkg-config was missing.

I had to add some more packages to Build-Depends.
It looks like that:
 debhelper,
 pkg-config,
 qtdeclarative5-dev,
 qttools5-dev-tools,
 qtmultimedia5-dev,
 linux-libc-dev,
 libavcodec-dev,
 libavdevice-dev,
 libavfilter-dev,
 libavformat-dev,
 libavresample-dev,
 libavutil-dev,
 libv4l-dev,
 libpostproc-dev,
 libswresample-dev,
 libswscale-dev,
 libqt5svg5-dev

Now is necessary to solve some lintian messages. When it is finished
I talk to you again.

hipersayanX, 2016-02-25:
Issue fixed, closing.

closed 2016-02-25 by hipersayanX