webcamoid/webcamoid #30: webcamoid.desktop

hpfn, 2015-11-10:
Hi,

Due lintian messages I made a patch for webcamoid.desktop:

-------------------------------------------------------------------------------------------------------------
Description: Remove deprecated key and add Keywords key
Author: Herbert Parentes Fortes Neto <hpfn@ig.com.br>
Last-Update: 2015-11-10
Index: webcamoid-6.2.0/webcamoid.desktop
===================================================================
--- webcamoid-6.2.0.orig/webcamoid.desktop
+++ webcamoid-6.2.0/webcamoid.desktop
@@ -1,5 +1,4 @@
 [Desktop Entry]
-Encoding=UTF-8
 Name=Webcamoid
 GenericName=Webcam Capture Software
 GenericName[ca]=Programari de Captura de Càmera web
@@ -29,6 +28,7 @@ Comment[pt]=Software de Captura de Webca
 Comment[ru]=Веб-камера захвата программного обеспечения
 Comment[zh_CN]=摄像头捕捉软件
 Comment[zh_TW]=攝像頭捕捉軟件
+Keywords=photo;video;webcam;
 Exec=webcamoid
 Icon=camera-web
 Terminal=false
---------------------------------------------------------------------------------------------------------

There are some messages from desktop-file-validate command about
redundance you may be interested. For example:

webcamoid-6.2.0/webcamoid.desktop: warning: value "Software de Captura de Webcam" for key "Comment[pt]" in group "Desktop Entry" looks redundant with value "Software de Captura de Webcam" of key "GenericName[pt]"

regards,
Herbert

hipersayanX, 2015-11-10:
Great! can you create a pull request? about **GenericName** and **Comment** in *.desktop*, adding:

    Comment=Webcamoid, ...

will be enough, according to [this](https://wiki.archlinux.org/index.php/Desktop_entries#Key_definition).

hpfn, 2015-11-14:
Hi,

> Great! can you create a pull request? about **GenericName** and **Comment** in *.desktop*, adding:
> 
>     Comment=Webcamoid, ...
> 
> will be enough, according to [this](https://wiki.archlinux.org/index.php/Desktop_entries#Key_definition).
> 

I never did a pull request. I don't know
how to do it.

The Debian package is almost done. I'm
a DM, not a DD, so I asked for a first review
at debian-mentors.

regards,
-- 
Herbert Parentes Fortes Neto (hpfn)

hipersayanX, 2015-11-14:
For doing a pull request:

1. In the top of the repository, you have 3 buttons: *Watch*, *Star* and *Fork*; click **Fork**.
2. Now in you copy of the repository, go to the file you want to edit, in this case *webcamoid.desktop*, in the top of the file preview, you have the buttons: *Raw*, *Blame*, *History*, and a pen, click the **pen (Edit)**.
3. Edit the file as you want.
4. Go to the bottom of the editor, write a message for your changes, and select **"Create a new branch for this commit..."**, and press **Commit changes**.
5. That's all. More info [here](https://help.github.com/articles/using-pull-requests/).

Then I will review the PR and merge the changes. Also, in this way you will get the credits for the changes.

hpfn, 2015-11-15:
Hi,

> For doing a pull request:
>
> 1. In the top of the repository, you have 3 buttons: *Watch*, *Star* and *Fork*; click **Fork**.
> 2. Now in you copy of the repository, go to the file you want to edit, in [...]

Ok. Just did it. But the last step was not
'Commit changes'.

I have one 'Pull Requests' on hpfn/webcamoid.

regards,
-- 
Herbert Parentes Fortes Neto (hpfn)

hipersayanX, 2015-11-15:
>Ok. Just did it. But the last step was not 'Commit changes'.

You must click **Propose file change**.

>I have one 'Pull Requests' on hpfn/webcamoid. 

You have a fork of Webcamoid's repository, now go to step 2, and edit the file. Once you made the changes to the file and click the **Propose file change** button, it will appears in the PR page, [here](https://github.com/hipersayanX/webcamoid/pulls?q=is%3Apr+is%3Aclosed) is an example.

hpfn, 2015-11-16:
I think the PR was created.

Do I have to sign CLAHub ?

regards,
-- 
Herbert Parentes Fortes Neto (hpfn)

hipersayanX, 2015-11-16:
>Do I have to sign CLAHub ?

Yes.

hpfn, 2015-11-17:
> >Do I have to sign CLAHub ?
> 
> Yes.

Just did it.

-- 
Herbert Parentes Fortes Neto (hpfn)

hipersayanX, 2016-02-25:
Fixed in latest release, closing.

closed 2016-02-25 by hipersayanX