webcamoid/webcamoid #32: cowbuilder fails

hpfn, 2015-11-23:
Hi,

From the normal tool (debuild) the package is builded. But using
'cowbuilder', to test deps and all, it fails to complete the process.

There is the error message:
In file included from include/videodisplay.h:27:0,
                 from src/main.cpp:29:
include/videoframe.h:25:22: fatal error: QGLContext: No such file or directory
compilation terminated.

A quick search:
grep -r QGLContext /home/webcamoid/webcamoid-6.2.0/
/home/webcamoid/webcamoid-6.2.0/src/videoframe.cpp:    QGLContext *context = const_cast<QGLContext *>(QGLContext::currentContext());
/home/webcamoid/webcamoid-6.2.0/src/videoframe.cpp:    QGLContext *context = const_cast<QGLContext *>(QGLContext::currentContext());
/home/webcamoid/webcamoid-6.2.0/include/videoframe.h:#include <QGLContext>

$ less /home/webcamoid/webcamoid-6.2.0/include/videoframe.h
#include <QSGTexture>
#include <QGLContext>

Looking for QSGTexture:
$ grep -r QSGTexture /home/webcamoid/webcamoid-6.2.0/
/home/webcamoid/webcamoid-6.2.0/include/videoframe.h:#include <QSGTexture>
/home/webcamoid/webcamoid-6.2.0/include/videoframe.h:class VideoFrame: public QSGTexture

It seems to be something in the source code. Can't help more, sorry.

regards,
Herbert

hpfn, 2015-11-23:
parts between \< are not been showing.

hpfn, 2015-11-23:
let's see now:

/home/webcamoid/webcamoid-6.2.0/src/videoframe.cpp:    QGLContext *context = const_cast\<QGLContext *\>(QGLContext::currentContext());
/home/webcamoid/webcamoid-6.2.0/src/videoframe.cpp:    QGLContext *context = const_cast\<QGLContext *\>(QGLContext::currentContext());
/home/webcamoid/webcamoid-6.2.0/include/videoframe.h:#include \<QGLContext\>

hpfn, 2015-11-23:
I guess I was wrong.

The file seems to be part of libqt5opengl5-dev package. Which is
missing from Build-Depends. A friend helped me.

I will check tomorrow

regards,
Herbert
regards,
Herbert

hpfn, 2015-11-24:
Yes, libqt5opengl5-dev was missing.

One more package to put in debian/control (Build-Depends).

regards,
Herbert

hipersayanX, 2015-11-25:
Dependency added, thanks!

closed 2015-11-25 by hipersayanX