webcamoid/webcamoid #33: webcamoid on NEW

hpfn, 2015-12-12:
Hi,

The webcamoid Debian package was uploaded
and is at NEW.

Let's see what the ftp-master will do now.

https://ftp-master.debian.org/new.html

regards,
Herbert

hipersayanX, 2015-12-12:
Great! :smile:

hpfn, 2015-12-30:
Hi,

The package was accepted.
https://packages.qa.debian.org/w/webcamoid.html

Was on experimental for final fixes. Tomorrow webcamoid
will be in 'sid'.

Happy New Year.

regards

hipersayanX, 2015-12-30:
So, is Webcamoid working well in Debian? Any special requirements for intalling it? I think it will be good idea to add all that information to [the wiki](https://github.com/hipersayanX/webcamoid/wiki).

hpfn, 2015-12-30:
> So, is Webcamoid working well in Debian? Any special requirements for intalling it? I think it will be good idea to add all that information to [the wiki](https://github.com/hipersayanX/webcamoid/wiki).
> 

I believe is working well. See:
https://buildd.debian.org/status/package.php?p=webcamoid&suite=experimental

No special requirements. The architectures which
are 'BD-Uninstallable' is not webcamoid problem.

regards,
-- 
Herbert Parentes Fortes Neto (hpfn)

hipersayanX, 2016-02-25:
I'll release Webcamoid 7.0.0 in few hours, so hopefully all problems with 6.x series will be solved and Webcamoid can be finally included in Debian repositories :smiley:

hipersayanX, 2016-02-25:
Webcamoid 7.0.0 released, Qt >= 5.5 and FFmpeg >= 2.8 is required, so probably Debian Sid or higher is required for build it. The logs are at [OBS](https://build.opensuse.org/package/show/home:hipersayan_x/Webcamoid).

hpfn, 2016-03-16:
Hi,

New version is at NEW again:
https://ftp-master.debian.org/new/webcamoid_7.0.0-1.html

I checked webcamoid-7.0.0/ports/debian/control and 
it helped me to remove some dependencies. Good!

But one is missing:  	libqt5opengl5-dev. Because of
QGLContext. With this package cowbuilder does not
complain.

regards,

hipersayanX, 2016-03-16:
>I checked webcamoid-7.0.0/ports/debian/control and
>it helped me to remove some dependencies. Good!

Yes, i've removed some dependencies since previous releases :smiley: . Also, you can check the [.travis](https://github.com/webcamoid/webcamoid/blob/master/.travis.yml) file and the [OBS](https://build.opensuse.org/package/show/home:hipersayan_x/Webcamoid) for more examples.

>But one is missing: libqt5opengl5-dev. Because of
>QGLContext. With this package cowbuilder does not
>complain.

Webcamoid doesn't use OpenGL right now (not directly), what is the error message?

hpfn, 2016-03-17:
> >But one is missing: libqt5opengl5-dev. Because of
> >QGLContext. With this package cowbuilder does not
> >complain.
> 
> Webcamoid doesn't use OpenGL right now (not directly), what is the error message?

src/videoframe.h:25:22: fatal error: QGLContext: File or directory not found

regards,
-- 
Herbert Parentes Fortes Neto (hpfn)

hipersayanX, 2016-03-17:
Oooooh, get it, i forgot about that :laughing: 
Thanks a lot!

hipersayanX, 2016-04-30:
Webcamoid is now included in Debian repositories, so closing.

closed 2016-04-30 by hipersayanX