webcamoid/webcamoid #34: Libqb requirement

Knusper, 2015-12-26:
I succesfully build 6.2.0 on openSUSE 42.1.
However, ldd on the binary gave me a 'libQb.so.6 => not found'. 

So - libQb is not needed to build - but to execute the program. Therefore it should be listed as a dependency in the documentation https://github.com/hipersayanX/webcamoid/wiki/Raw-build-and-install
, or not?

hipersayanX, 2015-12-26:
libQb is part of Webcamoid, you need it for build and run, the reason of the error is because you didn't installed webcamoid in your system. If you want to test Webcamoid without installing you can run it as:

    LD_PRELOAD='../Qb/Lib/libQb.so.6.2.0' ./webcamoid

You also need to configure your *~/.config/Webcamoid/Webcamoid.conf* file as:

    [PluginConfigs]
    paths\1\path=/home/your_user/path/to/Webcamoid/Qb/Plugins
    paths\size=1
    qmlPluginPath=/home/your_user/path/to/Webcamoid/Qb
    recursive=true

This will tell to Webcamoid where are located the plugins.

Knusper, 2015-12-26:
I did install it as described in the link. 
Although, not exactly... If I run `su -c 'make INSTALL_ROOT=/usr install'` everything got installed in /usr/usr/ ... I so just dropped the `INSTALL_ROOT` option.

Nevertheless - I got confused - and apparently it works when it is installed. I thought tht libqb refers to http://clusterlabs.github.io/libqb/ ... you can close this.

hipersayanX, 2015-12-26:
Oh thanks, I've removed the **INSTALL_ROOT** from the wiki instructions.

closed 2015-12-26 by hipersayanX