webcamoid/webcamoid #35: conflict in a file name

hpfn, 2016-01-04:
Hi,

Another package in Debian provide a file with the same name
in the same directory. And webcamoid is newer.

/usr/lib/x86_64-linux-gnu/libqb.so

The chat can be read here:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=809764

The file can be put in another directory when packaging but,
as said, will be a maintenance burden. The name of the file 
can be changed ?

regards,
Herbert

hpfn, 2016-01-04:
I have to let you know that I change to lowercase the qb part:

--- webcamoid-6.2.0.orig/Qb/commons.pri
+++ webcamoid-6.2.0/Qb/commons.pri
@@ -18,7 +18,7 @@
 # Web-Site: http://github.com/hipersayanX/webcamoid
 
 isEmpty(COMMONS_PRI_INCLUDE) {
-    COMMONS_APPNAME = "Qb"
+    COMMONS_APPNAME = "qb"
     COMMONS_TARGET = $${COMMONS_APPNAME}
     VER_MAJ = 6
     VER_MIN = 2

Removing this will fix, for now, the situation. But it still be
confusing. libqb libQb.

hipersayanX, 2016-01-04:
Tbh, this is my fault, I didn't know that other library with the same name existed before.

By the moment, you can try patching the Qb/Qb.pro, and change the line:

    TARGET = $${COMMONS_APPNAME}

to

    TARGET = Webcamoid

This will  create a *libWebcamoid.so* instead of *libQb.so*. Then in the file *StanAlone/StanAlone.pro*, change the line:

    !win32: LIBS += -L../Qb/Lib -lQb

to

    !win32: LIBS += -L../Qb/Lib -lWebcamoid

In *Qb/QbQml* and *Qb/Plugins/Plugin* do the same.

This will fixed for next release for sure. My deepest apologies.

hpfn, 2016-01-04:
> Tbh, this is my fault, I didn't know that other library with the same name existed before.

Nobody's fault.

> By the moment, you can try patching the Qb/Qb.pro, and change the line:
>
>     TARGET = $${COMMONS_APPNAME}
>
> to
>
>     TARGET = Webcamoid
>
> This will  create a *libWebcamoid.so* instead of *libQb.so*. [...]

I did patches for:

StandAlone.pro
Qb/Lib.pro
Qb/Plugins/*/*.pro
Qb/QbQml/QbQml.pro

As the name change, the packages will be treated
as NEW. I don't know how many days to go to 'sid'
again.

regards,
-- 
Herbert Parentes Fortes Neto (hpfn)

hpfn, 2016-01-04:
Please, change everything about 'qb' to webcamoid.

For exemple, when I packaged I used:

/usr/include/qb
/usr/lib/x86_64-linux-gnu/qb/libACapsConvert.so

I changed qb -> Webcamoid.

The package which webcamoid conflicted with is libqb.
Just in case:
https://packages.qa.debian.org/libq/libqb.html

regards,
-- 
Herbert Parentes Fortes Neto (hpfn)

hpfn, 2016-01-04:
I forgot Qb/commons.pri patch

hipersayanX, 2016-02-25:
Fixed in latest release, closing.

closed 2016-02-25 by hipersayanX