webcamoid/webcamoid #36: RPM build errors

greenrd, 2016-01-06:
```
RPM build errors:
    bogus date in %changelog: Wed Feb 23 2015 Gonzalo Exequiel Pedone <hipersayan DOT x AT gmail DOT com> 6.2.0-1
    Directory not found: /home/robin/rpmbuild/BUILDROOT/webcamoid-6.2.0-1.fc23.x86_64/usr/share/webcamoid
    Directory not found: /home/robin/rpmbuild/BUILDROOT/webcamoid-6.2.0-1.fc23.x86_64/usr/include/Qb
    File not found: /home/robin/rpmbuild/BUILDROOT/webcamoid-6.2.0-1.fc23.x86_64/usr/lib64/libWebcamoid.so*
```

hipersayanX, 2016-01-06:
>bogus date in %changelog: Wed Feb 23 2015 Gonzalo Exequiel Pedone <hipersayan DOT x AT gmail DOT com> 6.2.0-1

In *webcamoid.spec* change the **%changelog** part to:

    %changelog
    * Wed Jan 06 2016 Gonzalo Exequiel Pedone <hipersayan DOT x AT gmail DOT com> 6.2.0-1
    - Final Release.

Or just change the date to the output of this ccommand:

    LC_ALL=C date +"%a %b %d %Y"

> Directory not found: /home/robin/rpmbuild/BUILDROOT/webcamoid-6.2.0-1.fc23.x86_64/usr/share/webcamoid
> Directory not found: /home/robin/rpmbuild/BUILDROOT/webcamoid-6.2.0-1.fc23.x86_64/usr/include/Qb [...]

Change the **%files** part to:

    %files
    %defattr(-,root,root,-)
    %{_bindir}/webcamoid
    %{_datadir}/applications/webcamoid.desktop
    #%{_datadir}/webcamoid/
    %{_defaultdocdir}/webcamoid/
    #%{_includedir}/Qb/
    %{_libdir}/Qb/
    %{_libdir}/lib*Qb.so*
    #%{_libdir}/libWebcamoid.so*

ozky, 2016-01-30:
I got in Mageia 5 complain with edited spec file error that there is found installed but not packaged files.
/usr/lib64/qt5/qml/QbQml/libQbQml.so
/usr/lib64/qt5/qml/QbQml/qmldir
Then i got error about empty debug file what should be gone with this.
%define debug_package %{nil}
https://wiki.mageia.org/en/Packaging_problems_and_solutions#Empty_Debug_Files
Some info cmake is 3.0.2 make 4.0 qt 5.4.2 gcc 4.9.2 webcamoid desktop lxqt 0.9.0
Full log is here:
Empty %files file /home/ozky/rpmbuild/BUILD/webcamoid-6.2.0/debugfiles.list
Found-installed (but non-packaged) files:
/usr/lib64/qt5/qml/QbQml/libQbQml.so
 /usr/lib64/qt5/qml/QbQml/qmldir

hipersayanX, 2016-01-30:
@ozky, I've added *debug_package* to the spec file. About the non-packaged files, i'm not sure what that exactly means, maybe you need to specify the full path? something like this?

    %{_libdir}/qt5/qml/QbQml/libQbQml.so
    %{_libdir}/qt5/qml/QbQml/qmldir

in **%files** section. Let me know if that works for you, since i'm working blind with these distros.

ozky, 2016-01-30:
Thank you for all your help same in your language: `gracias por toda su ayuda`,can commit it to main branch spec file.

hipersayanX, 2016-01-30:
@ozky, yes please, do it :smile: , help from users of other distros is very much appreciated.

ozky, 2016-01-30:
This is why i am here :smile: i also help Mageia developers all my free time to fixing bugs,other things to make whole distro better.

hipersayanX, 2016-02-25:
Got working Webcamoid with [OBS](https://build.opensuse.org/package/show/home:hipersayan_x/Webcamoid), Qt >= 5.5 is required, there the packages for Fedora (GStreamer) and openSUSE (FFmpeg), i didn't tested with Mageia.

ozky, 2016-02-26:
Mageia 5 only have qt5.4 so it may not get it builded.
I forgot that fixed spec file totally i was having some busy time and then i lost that file.
I added these requires to it what was not in original spec.
`qtgraphicaleffects5 qtmultimedia5 qtquickcontrols5 `

hipersayanX, 2016-02-26:
According to [this](http://pkgs.org/search/libqt5base5-devel), Mageia has Qt 5.6 in Cauldron, so Webcamoid can be packaged for future versions of it.
Also, QtGraphicalEffects and QtMultimedia requirements where removed with 7.x series, Webcamoid requires less dependencies than before.

hipersayanX, 2016-04-30:
New release 7.1.0.

closed 2016-04-30 by hipersayanX

ozky, 2016-05-04:
Thanks i soon try to build this,i have moved to Mageia Cauldron aka Mageia 6 dev1 what have all build requires up to date.

hipersayanX, 2016-05-04:
@ozky, let use this issue for Fedora, that was the original subject, when you are done, open a new issue for Mageia. Also let me know when you are done, so i can upload the required info to [the wiki](https://github.com/webcamoid/webcamoid/wiki/Binary-packages).