webcamoid/webcamoid #39: Debian Build Issue

rstreeter, 2016-02-05:
I am having trouble building I have the dependencies installed and keep getting this error when attempting to build the package 

`~/webcamoid-6.1.0$ dpkg-buildpackage -us -uc
dpkg-buildpackage: source package webcamoid
dpkg-buildpackage: source version 6.1.0-1
dpkg-buildpackage: source distribution unstable
dpkg-buildpackage: source changed by Gonzalo Pedone <hipersayan.x@gmail.com>
dpkg-buildpackage: host architecture amd64
 dpkg-source --before-build webcamoid-6.1.0
 fakeroot debian/rules clean
dh_testdir
dh_testroot
rm -f build-stamp
# Add here commands to clean up after the build process.
/usr/bin/make clean || true
make[1]: Entering directory `/home/robert/webcamoid-6.1.0'
cd Qb/ && ( test -e Makefile || /usr/lib/x86_64-linux-gnu/qt5/bin/qmake /home/robert/webcamoid-6.1.0/Qb/Qb.pro -o Makefile ) && /usr/bin/make -f Makefile clean
make[2]: Entering directory `/home/robert/webcamoid-6.1.0/Qb'
( test -e Makefile.Lib || /usr/lib/x86_64-linux-gnu/qt5/bin/qmake /home/robert/webcamoid-6.1.0/Qb/Lib.pro -o Makefile.Lib ) && /usr/bin/make -f Makefile.Lib clean
Project ERROR: Unknown module(s) in QT: qml
make[2]: *** [sub-Lib-pro-clean-ordered] Error 3
make[2]: Leaving directory `/home/robert/webcamoid-6.1.0/Qb'
make[1]: *** [sub-Qb-clean-ordered] Error 2
make[1]: Leaving directory `/home/robert/webcamoid-6.1.0'
# --- end custom part for cleaning up
dh_clean
	rm -f debian/webcamoid.substvars
	rm -f debian/webcamoid.*.debhelper
	rm -rf debian/webcamoid/
	rm -f debian/*.debhelper.log
	rm -f debian/files
	find .  \( \( -type f -a \
	        \( -name '#*#' -o -name '.*~' -o -name '*~' -o -name DEADJOE \
		 -o -name '*.orig' -o -name '*.rej' -o -name '*.bak' \
		 -o -name '.*.orig' -o -name .*.rej -o -name '.SUMS' \
		 -o -name TAGS -o \( -path '*/.deps/*' -a -name '*.P' \) \
		\) -exec rm -f {} + \) -o \
		\( -type d -a -name autom4te.cache -prune -exec rm -rf {} + \) \)
	rm -f *-stamp
 dpkg-source -b webcamoid-6.1.0
dpkg-source: info: using source format `3.0 (quilt)'
dpkg-source: info: building webcamoid using existing ./webcamoid_6.1.0.orig.tar.gz
dpkg-source: info: local changes detected, the modified files are:
 webcamoid-6.1.0/Makefile
 webcamoid-6.1.0/Qb/Makefile
dpkg-source: info: you can integrate the local changes with dpkg-source --commit
dpkg-source: error: aborting due to unexpected upstream changes, see /tmp/webcamoid_6.1.0-1.diff.ueFm0Z
dpkg-buildpackage: error: dpkg-source -b webcamoid-6.1.0 gave error exit status 2
robert@robert-K56CA:~/webcamoid-6.1.0$ 
`

hipersayanX, 2016-02-05:
Try installing the package **qtdeclarative5-dev**, let me know if that works.

rstreeter, 2016-02-05:
It was going better then I hit a snag here is the error message this time

make[3]: *** [build/Qt5.2.1/gcc/release/obj/qbcaps.o] Error 1
make[3]: Leaving directory `/home/robert/webcamoid-6.1.0/Qb'
make[2]: *** [sub-Lib-pro-make_first-ordered] Error 2
make[2]: Leaving directory `/home/robert/webcamoid-6.1.0/Qb'
make[1]: *** [sub-Qb-make_first-ordered] Error 2
make[1]: Leaving directory `/home/robert/webcamoid-6.1.0'
make: *** [build-stamp] Error 2
dpkg-buildpackage: error: debian/rules build gave error exit status 2
robert@robert-K56CA:~/webcamoid-6.1.0$

hipersayanX, 2016-02-05:
There must be the real error before that. 

Also, 6.1.0 is an old version, try compiling [latest version](https://github.com/hipersayanX/webcamoid/releases/tag/6.2.0), or even better current git version. 

And you are using a very old version of Qt, there are a lot of differences between Qt 5.2 and Qt 5.5 API, this could lead to compiling problems. You must update your system.

hipersayanX, 2016-02-25:
This issue become invalid with 7.x series, closing.

closed 2016-02-25 by hipersayanX