webcamoid/webcamoid #40: module "QtQuick*" is not installed

hpfn, 2016-04-03:
Hi,

A user reported and I confirm:

$ webcamoid 
QQmlApplicationEngine failed to load component
qrc:/Webcamoid/share/qml/main.qml:25 module "QtQuick.Layouts" is not installed
qrc:/Webcamoid/share/qml/main.qml:24 module "QtQuick.Controls" is not installed
qrc:/Webcamoid/share/qml/main.qml:21 module "QtQuick" is not installed
qrc:/Webcamoid/share/qml/main.qml:23 module "QtQuick.Window" is not installed
qrc:/Webcamoid/share/qml/main.qml:25 module "QtQuick.Layouts" is not installed
qrc:/Webcamoid/share/qml/main.qml:24 module "QtQuick.Controls" is not installed
qrc:/Webcamoid/share/qml/main.qml:21 module "QtQuick" is not installed
qrc:/Webcamoid/share/qml/main.qml:23 module "QtQuick.Window" is not installed
qrc:/Webcamoid/share/qml/main.qml:25 module "QtQuick.Layouts" is not installed
qrc:/Webcamoid/share/qml/main.qml:24 module "QtQuick.Controls" is not installed
qrc:/Webcamoid/share/qml/main.qml:21 module "QtQuick" is not installed
qrc:/Webcamoid/share/qml/main.qml:23 module "QtQuick.Window" is not installed
qrc:/Webcamoid/share/qml/main.qml:25 module "QtQuick.Layouts" is not installed
qrc:/Webcamoid/share/qml/main.qml:24 module "QtQuick.Controls" is not installed
qrc:/Webcamoid/share/qml/main.qml:21 module "QtQuick" is not installed
qrc:/Webcamoid/share/qml/main.qml:23 module "QtQuick.Window" is not installed

The user said that he fixed the problem with qml-module-qtquick-dialogs package.
I installed the package but the errors still happen.

Webcamoid was builded with qtdeclarative5-dev. (log - i386)
https://buildd.debian.org/status/fetch.php?pkg=webcamoid&arch=i386&ver=7.0.0-2&stamp=1458754915

Do you have something in mind ?

regards,

hpfn, 2016-04-03:
Hi,
 - qml-module-qtquick-dialogs
 - qml-module-qtquick-controls
- qml-module-qtquick-layouts
- qml-module-qtquick-window2

Have to be installed.

regards,

hipersayanX, 2016-04-03:
Dependencies added, thanks!

closed 2016-04-03 by hipersayanX

AnikSR, 2018-06-10:
Thanks from 2018.

AnikSR, 2018-06-10:
Thanks from 2018.

FlorinChertes, 2018-08-09:
Dear Herbert, thank you very much, you suggestion solved my problem on RPi3, Qt5.3. with QtQuick 2.3.

guohaoooo, 2020-06-17:
Thanks from 2020.