webcamoid/webcamoid #41: issuing a wishlist

hpfn, 2016-05-25:
Hi,

I am packaging the last release for Debian and doing a final
test. To see Webcamoid running. As I did that on a VM (KVM), 
I do not really have a webcam. But it is good to see the window 
popping up, without a image, but the program is running. The
dependencies are ok.

If I run Webcamoid from Gnome Shell, I have only a blank
window. From a terminal I got this message:

'QFileSystemWatcher::addPAths: list is empty'

Two files has :
"libAvKys/Plugins/VideoCApture/src/v4l2/(capture|cameraout).cpp"

...QFileSystemWatcher(QStringList() << "/dev", this)

To see the behavior of other webcam application, I ran 'cheese'.
A blank window again. But with a message that tell something like:

"Not device found"

It is clear that the program ran fine and the webcam is missing. And
it was not necessary run the program from a terminal.

Can we have that on Webcamoid ?

regards,

hipersayanX, 2016-05-25:
>If I run Webcamoid from Gnome Shell, I have only a blank window.

Just a plain blank window? no icons, no button, no nothing?

>From a terminal I got this message:
>
>'QFileSystemWatcher::addPAths: list is empty'

This is just a warning but Webcamoid will start normally.

hpfn, 2016-05-25:
> >If I run Webcamoid from Gnome Shell, I have only a blank window.
> 
> Just a plain blank window? no icons, no button, no nothing?
> 

I'm sorry, buttons at the bottom. And they are operational.

> >From a terminal I got this message:
> >
> >'QFileSystemWatcher::addPAths: list is empty'
> 
> This is just a warning but Webcamoid will start normal

Ok.

regards,
-- 
Herbert Parentes Fortes Neto (hpfn)

hipersayanX, 2016-05-25:
Feature added :smile:

hpfn, 2016-05-26:
Em 25/05/2016 17:42, hipersayanX escreveu:
> Feature added 😄
>

Thanks !

hipersayanX, 2016-07-19:
Feature added in 0.7.2, closing.

closed 2016-07-19 by hipersayanX