webcamoid/webcamoid #42: VirtualCamera - 7.2.0

hpfn, 2016-08-02:
Hi,

VirtualCamera plugin is  GPL-incompatible[0].

[0] - https://en.wikipedia.org/wiki/Shared_source#Microsoft_Limited_Public_License_.28Ms-LPL.29

I can not put a file with Ms-LPL license in Debian. I removed
the directory, edited Plugins.pro, tried "COMPILE_VCAM_DRIVER=0".
The package builds, but it does not work.

I click on 'Play' but no image at all. The error message:

$ LANG=C webcamoid 
Error loading plugin  "VirtualCamera" : "The shared library was not found."
"Error: Element 'VirtualCamera' doesn't exist."

VirtualCamera is for Windows only. What do I have to
do to disable it ?

regards,
Herbert

hipersayanX, 2016-08-02:
**COMPILE_VCAM_DRIVER** is just for Windows, not for GNU/Linux. In GNU/Linux the virtual camera plugin uses v4l2loopback, so you don't need to remove the plugin, or add any special option, just remove [dshow](https://github.com/webcamoid/webcamoid/tree/master/libAvKys/Plugins/VirtualCamera/src/dshow) folder from source distribution and that's all.

hpfn, 2016-08-02:
Em Ter, 2016-08-02 às 12:22 -0700, hipersayanX escreveu:
> COMPILE_VCAM_DRIVER is just for Windows, not for GNU/Linux. In GNU/Linux the virtual camera plugin uses v4l2loopback, so you don't need to remove the plugin, or add any special option, just remove
> dshow folder from source distribution and that's all.
> —

Quick replay. Thanks!

Webcamoid is running!

regards,
-- 
Herbert Parentes Fortes Neto (hpfn)

hipersayanX, 2016-08-02:
Closing then :smile:

closed 2016-08-02 by hipersayanX

hipersayanX, 2016-10-13:
**UPDATE** Just in case other system developers have the same problem packaging Webcamod I've added a [README](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VirtualCamera/src/dshow/VirtualCameraFilter/BaseClasses/src/README) about this issue, BaseClasses is not required for redistribution outside Windows, so feel free to remove it.