webcamoid/webcamoid #43: VirtualCamera - 7.2.0 - linux

hpfn, 2016-08-19:
Hi,

Someone having problems with root password when trying to '+add' a device ?
(virtual camera)

regards,
Herbert

hipersayanX, 2016-08-19:
Can you describe more what was the problem exactly? Any error message in console?

hpfn, 2016-08-20:
Em Sex, 2016-08-19 às 11:15 -0700, hipersayanX escreveu:
> Can you describe more what was the problem exactly? Any error message in console?

# ordinary user
$ webcamoid 
qrc:/Webcamoid/share/qml/OutputConfig.qml:38:5: QML GroupBox: Binding loop detected for property "implicitWidth"
qrc:/Webcamoid/share/qml/OutputConfig.qml:38:5: QML GroupBox: Binding loop detected for property "implicitWidth"
qrc:/Webcamoid/share/qml/OutputConfig.qml:38:5: QML GroupBox: Binding loop detected for property "implicitHeight"
qrc:/Webcamoid/share/qml/OutputConfig.qml:38:5: QML GroupBox: Binding loop detected for property "implicitHeight"
qrc:/Webcamoid/share/qml/OutputConfig.qml:38:5: QML GroupBox: Binding loop detected for property "implicitWidth"
QQmlExpression: Expression qrc:/VirtualCamera/share/qml/main.qml:155:22 depends on non-NOTIFYable properties:
    VirtualCameraElement::needRoot
QQmlExpression: Expression qrc:/VirtualCamera/share/qml/main.qml:148:22 depends on non-NOTIFYable properties:
    VirtualCameraElement::needRoot

After enter root password: "Wrong Password" (red color)

# root
# webcamoid 
Qt: Session management error: None of the authentication protocols specified are supported
qrc:/Webcamoid/share/qml/OutputConfig.qml:38:5: QML GroupBox: Binding loop detected for property "implicitWidth"
qrc:/Webcamoid/share/qml/OutputConfig.qml:38:5: QML GroupBox: Binding loop detected for property "implicitWidth"
qrc:/Webcamoid/share/qml/OutputConfig.qml:38:5: QML GroupBox: Binding loop detected for property "implicitHeight"
qrc:/Webcamoid/share/qml/OutputConfig.qml:38:5: QML GroupBox: Binding loop detected for property "implicitHeight"
qrc:/Webcamoid/share/qml/OutputConfig.qml:38:5: QML GroupBox: Binding loop detected for property "implicitWidth"
QQmlExpression: Expression qrc:/VirtualCamera/share/qml/main.qml:155:22 depends on non-NOTIFYable properties:
    VirtualCameraElement::needRoot
QQmlExpression: Expression qrc:/VirtualCamera/share/qml/main.qml:148:22 depends on non-NOTIFYable properties:
    VirtualCameraElement::needRoot

After enter root password: ok.

I did a search about 'Session management error', but 
did not find a solution yet.

regards,
-- 
Herbert Parentes Fortes Neto (hpfn)

hipersayanX, 2016-08-20:
As normal user, is this command working?

    echo 'ROOT_PASSWORD_HERE' | su -c 'modprobe v4l2loopback video_nr=12 card_label="Virtual Camera Test"'
    ls /dev/video12

hipersayanX, 2016-08-20:
Also, perhaps, is the user using Ubuntu? because [this](https://help.ubuntu.com/community/RootSudo) could be the reason.

hpfn, 2016-08-20:
Em Sáb, 2016-08-20 às 11:19 -0700, hipersayanX escreveu:
> echo 'ROOT_PASSWORD_HERE' | su -c 'modprobe v4l2loopback video_nr=12 card_label="Virtual Camera Test"'

returns:
su: must be run from a terminal

regards,
-- 
Herbert Parentes Fortes Neto (hpfn)

hpfn, 2016-08-20:
Em Sáb, 2016-08-20 às 11:31 -0700, hipersayanX escreveu:
> Also, perhaps, is the user using Ubuntu? because this could be the reason.

I am doing tests myself. Debian testing.

regards,
-- 
Herbert Parentes Fortes Neto (hpfn)

hipersayanX, 2016-08-20:
What about this?:

    echo 'ROOT_PASSWORD_HERE' | sudo -S modprobe v4l2loopback video_nr=13 card_label="Virtual Camera Test"
    ls /dev/video13

hpfn, 2016-08-21:
Em Sáb, 2016-08-20 às 14:00 -0700, hipersayanX escreveu:
> echo 'ROOT_PASSWORD_HERE' | sudo -S modprobe v4l2loopback video_nr=13 card_label="Virtual Camera Test"

I do not use sudo but: I did not have time to 
enter a password. It was like a timeout.

We trust you have received the usual lecture from the local System
Administrator. It usually boils down to these three things:

    #1) Respect the privacy of others.
    #2) Think before you type.
    #3) With great power comes great responsibility.

[sudo] password for zaza: Sorry, try again.
[sudo] password for zaza: 
sudo: 1 incorrect password attempt

regards,
-- 
Herbert Parentes Fortes Neto (hpfn)

hipersayanX, 2016-08-21:
I've tried the sudo command my self, it works well. I'll upload the patch later.

>We trust you have received the usual lecture from the local System Administrator. It usually boils down to these three things:
>
>     #1) Respect the privacy of others.
>     #2) Think before you type.
>     #3) With great power comes great responsibility.

It's ok, that message appears in first usage of sudo.

>sudo: 1 incorrect password attempt

Because you wrote the command literally, you must change **ROOT_PASSWORD_HERE** with your admin password.

hpfn, 2016-08-21:
Em Dom, 2016-08-21 às 11:25 -0700, hipersayanX escreveu:
> I've tried the sudo command my self, it works well. I'll upload the patch later.

I do not know how many users uses sudo. In Ubuntu
probably a lot of.

> We trust you have received the usual lecture from the local System Administrator. It usually boils down to these three things:
> #1) Respect the privacy of others.
> #2) Think before you type.
> #3) With [...]

I did the change.

regards,
-- 
Herbert Parentes Fortes Neto (hpfn)

hipersayanX, 2016-08-22:
Hi, I made the changes to support **sudo** instead of **su**, I've added **ROOT_METHOD** at compile time, look at [Build options](https://github.com/webcamoid/webcamoid/wiki/Raw-build-and-install).

I can confirm it's working in Ubuntu 16.04, can you test the git version to confirm it's working or not in Debian?, not install required but it will required some tweaks to *~/.config/Webcamoid/Webcamoid.conf* file, so let me know if you can do it.

>I did the change.

Even so, I need to discard every possible option.

hpfn, 2016-08-23:
I can.

I need to:
 - use sudo 
 - edit ~/.config/Webcamoid/Webcamoid.conf 

That's all ?

hipersayanX, 2016-08-23:
Good!

1) First at all, install all build and run dependencies listed at *control* file, and the v4l2loopback-utils package or equivalent.
2) clone the repository with:

    git clone https://github.com/webcamoid/webcamoid.git

3) **cd** inside the *webcamoid* folder and execute as normally:

    qmake -qt=5 Webcamoid.pro ROOT_METHOD=sudo
    make

Note the **ROOT_METHOD** flag.

4) Edit *~/.config/Webcamoid/Webcamoid.conf* to add this options:

    [OutputConfigs]
    enableVirtualCamera=true

    [PluginConfigs]
    paths\1\path=/home/USER/path/to/webcamoid/libAvKys/Plugins
    paths\size=1
    qmlPluginPath=/home/USER/path/to/webcamoid/libAvKys
    recursive=true

Just replace */home/USER/path/to* with the path where you cloned the repository. And remove the entire **PluginsCache** entry with the paths, so it will be refreshed with paths in the cloned repository instead of the installed plugins paths.

5) And launch Webcamoid as:

    LD_PRELOAD='./libAvKys/Lib/libavkys.so' ./StandAlone

Don't run Webcamoid as root.

6) Test if you can add/remove/change description, normally.

I've also added the code to show the output of su/sudo commands at CLI,

hpfn, 2016-08-25:
Hi,

Em Ter, 2016-08-23 às 06:43 -0700, hipersayanX escreveu:
> Good!
> 1) First at all, install all build and run dependencies listed at control file, and the v4l2loopback-utils package or equivalent.
> 2) clone the repository with:
> git clone [...]

First I did everything in a chroot env. And
tried to run webcamoid inside it:

$ LD_PRELOAD='./libAvKys/Lib/libavkys.so' ./StandAlone/webcamoid 
libGL error: MESA-LOADER: could not create udev device for fd 9
libGL error: MESA-LOADER: could not create udev device for fd 9
qrc:/Webcamoid/share/qml/OutputConfig.qml:38:5: QML GroupBox: Binding loop detected for property "implicitWidth"
qrc:/Webcamoid/share/qml/OutputConfig.qml:38:5: QML GroupBox: Binding loop detected for property "implicitWidth"
qrc:/Webcamoid/share/qml/OutputConfig.qml:38:5: QML GroupBox: Binding loop detected for property "implicitHeight"
qrc:/Webcamoid/share/qml/OutputConfig.qml:38:5: QML GroupBox: Binding loop detected for property "implicitHeight"
qrc:/Webcamoid/share/qml/OutputConfig.qml:38:5: QML GroupBox: Binding loop detected for property "implicitWidth"
QQmlExpression: Expression qrc:/VirtualCamera/share/qml/main.qml:155:22 depends on non-NOTIFYable properties:
    VirtualCameraElement::needRoot
QQmlExpression: Expression qrc:/VirtualCamera/share/qml/main.qml:148:22 depends on non-NOTIFYable properties:
    VirtualCameraElement::needRoot
sudo: unable to resolve host debian-free
rmmod: ERROR: ../libkmod/libkmod.c:514 lookup_builtin_file() could not open builtin file '/lib/modules/4.6.0-1-amd64/modules.builtin.bin'
rmmod: ERROR: Module v4l2loopback is not currently loaded

The problem with 'Module v4l2loopback' is because
chroot. But no error message about wrong password.
I create an user inside chroot to run webcamoid.

I also tried to run webcamoid outside chroot, but
there is no menu to create the device. Only a
checkbox selected: (I do not understand why)

$ LD_PRELOAD='./libAvKys/Lib/libavkys.so' ./StandAlone/webcamoid 
"Error: Element 'MultiSrc' doesn't exist."
qrc:/Webcamoid/share/qml/OutputConfig.qml:38:5: QML GroupBox: Binding loop detected for property "implicitWidth"
qrc:/Webcamoid/share/qml/OutputConfig.qml:38:5: QML GroupBox: Binding loop detected for property "implicitWidth"
qrc:/Webcamoid/share/qml/OutputConfig.qml:38:5: QML GroupBox: Binding loop detected for property "implicitHeight"
qrc:/Webcamoid/share/qml/OutputConfig.qml:38:5: QML GroupBox: Binding loop detected for property "implicitHeight"
qrc:/Webcamoid/share/qml/OutputConfig.qml:38:5: QML GroupBox: Binding loop detected for property "implicitWidth"

regards,
-- 
Herbert Parentes Fortes Neto (hpfn)

hpfn, 2016-08-25:
Em Qui, 2016-08-25 às 09:02 -0700, hipersayanX escreveu:
> The problem with 'Module v4l2loopback' is because chroot. But no error message about wrong password. I create an user inside chroot to run webcamoid.
> v4l2loopback is a kernel module, it will never work in a chroot environment, if you want to test software in a sandbox (Webcamoid or any other), try it in VirtualBox. Also, forgot to mention, you
> must add your user to /etc/sudoers to use sudo.

Yes. I tried with KVM. Is not enough ?

Using sudo without problems.

-- 
Herbert Parentes Fortes Neto (hpfn)

hipersayanX, 2016-08-25:
That's  odd :confused: , I tested it in Debian Testing with same instructions and **sudo** method and get it working.

[image]

>Only a checkbox selected: (I do not understand why)

I don't understand it at all, the virtual camera manage buttons didn't appeared or something?

hpfn, 2016-08-25:
Em Qui, 2016-08-25 às 11:35 -0700, hipersayanX escreveu:
> That's odd 😕 , I tested it in Debian Testing with same instructions and sudo method and get it working.
> 
> Only a checkbox selected: (I do not understand why)
> I don't understand it at all, the virtual camera manage buttons didn't appeared or something?

See attached. No, only Virtual camera checked.

-- 
Herbert Parentes Fortes Neto (hpfn)

hipersayanX, 2016-08-25:
Then the virtual camera plugin didn't not get loaded, if you go to the effects configurations, can you see the effect list? if not, then you didn't configured *~/.config/Webcamoid/Webcamoid.conf* properly.

hpfn, 2016-08-25:
Em Qui, 2016-08-25 às 12:11 -0700, hipersayanX escreveu:
> 
> 
> Then the virtual camera plugin didn't not get loaded, if you go to the effects configurations, can you see the effect list? if not, then you didn't configured ~/.config/Webcamoid/Webcamoid.conf
> properly.

~/.config/Webcamoid/Webcamoid.conf now properly 
configured. :)

See attached the patch working.

-- 
Herbert Parentes Fortes Neto (hpfn)

hipersayanX, 2016-08-25:
Great! :smile:

hpfn, 2016-08-26:
Em Qui, 2016-08-25 às 14:52 -0700, hipersayanX escreveu:
> Great! 😄
> —

When the patch is ready, let me know. So I can
put in the Debian package.

regards,
-- 
Herbert Parentes Fortes Neto (hpfn)

hipersayanX, 2016-08-26:
Yes, no problem, I'll make a new release this Tuesday.

hipersayanX, 2016-08-30:
Bug fixed in 7.2.1.

closed 2016-08-30 by hipersayanX