webcamoid/webcamoid #44: Support JACK audio

wolftune, 2016-09-28:
Anyone doing real audio stuff on Linux uses JACK as it's much more flexible. Webcamoid appears to only support Pulse Audio. It should add direct ALSA support and full JACK support as options.

hipersayanX, 2016-09-28:
Thanks a lot for the idea!, my idea is to add audio and camera system detection at compile and runtime for 8.x series. Oss, Alsa and Jack will be supported for sure :smiley:

mikhailnov, 2016-10-02:
Is it really worth spending time on the dead OSS?

mikhailnov, 2016-10-02:
But Jack support will be a really great feature, especially when using Webcamoid as a virtual webcam.

mikhailnov, 2016-10-02:
Because currently the sound from the video played as a virtual webcam must be manually redirected using pavucontrol, for example.

wolftune, 2016-10-02:
Yeah, zero time should be given to OSS, nobody needs that. All they need are Pulse, ALSA, and JACK.

hipersayanX, 2016-10-03:
Supporting OSS is because I want Webcamoid to work in platforms like FreeBSD and similar, and I'm not sure if ALSA, JACK and PulseAudio are supported there.

hipersayanX, 2017-05-16:
Support for optional audio systems has been added in latest release.

closed 2017-05-16 by hipersayanX

hipersayanX, 2017-06-26:
@mikhailnov 👍