webcamoid/webcamoid #46: Virtual webcam very high ffmpeg's CPU usage

mikhailnov, 2016-10-02:
Hello,
When a video is played for a virtual webcam, ffmpeg creates multiple decoding/encoding streams depending on number of CU cores.

If to do
$ ffmpeg -re -i 40min.mp4 -f v4l2 /dev/video1 -f alsa default
(after modprobbing v4l2loopback which is used by Webcamoid)
then the CPU usage is low and it is not multistream.

Webcams are normally used with other video applications which also require quite a lot of system resources.

hipersayanX, 2016-10-03:
Thanks for reporting!, this issue seems to be more noticeable when capturing from desktop, because the [capture interval](https://github.com/webcamoid/webcamoid/blob/dbff1ddca4a7efab56365f77a4e34588c879331a/libAvKys/Plugins/DesktopCapture/src/desktopcaptureelement.cpp#L46) is set to its default value (that is 0), so increasing the the capture interval (depending on the desired FPS) reduces CPU usage. This will be solved for sure.

Webcams doesn't seems to be affected at all, and playing from videos causes this issue when playing HD videos and that's normal because you are converting from YUV to RGB, for video processing (e.g. effects), and then converting back to YUV for sending the processed frames to the fake webcam, maybe I find a fix for this but not sure.

All the tests were made without applying any effect, using effect is normal to trigger CPU usage.

mikhailnov, 2016-10-03:
I suggest to make an option to disable the video playback in the Webcamoid video because it is not needed in most cases. As for effects, personally I do not need them, a low CPU usage is much more important to me. If I send the same HD video via ffmpeg directly as I wrote, the CPU usage is low.

mikhailnov, 2016-10-03:
Now the CPU usage is sow high that it impacts the system performance quite high even on Intel i5 4 virtual cores.

hipersayanX, 2016-10-03:
>I suggest to make an option to disable the video playback in the Webcamoid video

No need for that, it's already optional, you add and play a video if you want, if not you don't and that's all.

>because it is not needed in most cases. 

What if you have not a webcam and you want to test the effects anyway?

>As for effects, personally I do not need them, a low CPU usage is much more important to me.

Effects are optional, you use if you want them, if not you don't. Removing a feature is never an option, such kind of decision must be never be driven by personal preferences.

Webcamoid was designed with flexibility in mind, to give to the user the maximum freedom to do this or that with the software. If I start removing things it will be not much different of what others webcam software does, I don't want another remake :)

mikhailnov, 2016-10-03:
You understood me incorrectly:) I did not say about removing a feauture, I meant an oltion to disable it because you said that the video is being converted between colorspaces even when no effects are applied.>No need for that, it's already optional, you add and play a video if you want, if not you don't and that's all.Do you mean that it is possible to pass the video to a virtual webcam but not play it in the Webcamoid window?

hipersayanX, 2016-10-03:
>I meant an oltion to disable it because you said that the video is being converted between colorspaces even when no effects are applied.

Will try reducing the number of conversions for sure.

>Do you mean that it is possible to pass the video to a virtual webcam but not play it in the Webcamoid window?

Doesn't, but maybe adding an option to stop rendering when the window is minimized could be useful here.

mikhailnov, 2016-10-03:
If you need any help in testing, let me know

hipersayanX, 2016-10-03:
Sure :+1:

hipersayanX, 2016-10-13:
Desktop capture now, may work well. Fixing this issue for other sources, will need more research.

mikhailnov, 2016-10-15:
Reducing fps greatly reduces cpu usage by ffmpeg, maybe make a setting to control the fps and number of ffmpeg threads

hipersayanX, 2017-05-16:
Fixed in latest release.

closed 2017-05-16 by hipersayanX