webcamoid/webcamoid #47: install directory is not followed by libavkys

bignaux, 2016-10-21:
```
>>> Install webcamoid-7.2.1 into /var/tmp/portage/media-video/webcamoid-7.2.1/image/ category media-video
make -j1 DESTDIR=/var/tmp/portage/media-video/webcamoid-7.2.1/image/ install 
cd libAvKys/ && ( test -e Makefile || /usr/lib64/qt5/bin/qmake /var/tmp/portage/media-video/webcamoid-7.2.1/work/webcamoid-webcamoid-3710641/libAvKys/libAvKys.pro 'QMAKE_AR=x86_64-pc-linux-gnu-ar cqs' QMAKE_CC=x86_64-pc-linux-gnu-gcc QMAKE_LINK_C=x86_64-pc-linux-gnu-gcc QMAKE_LINK_C_SHLIB=x86_64-pc-linux-gnu-gcc QMAKE_CXX=x86_64-pc-linux-gnu-g++ QMAKE_LINK=x86_64-pc-linux-gnu-g++ QMAKE_LINK_SHLIB=x86_64-pc-linux-gnu-g++ QMAKE_OBJCOPY=x86_64-pc-linux-gnu-objcopy QMAKE_RANLIB= QMAKE_STRIP= 'QMAKE_CFLAGS=-march=core2 -O2 -pipe' QMAKE_CFLAGS_RELEASE= QMAKE_CFLAGS_DEBUG= 'QMAKE_CXXFLAGS=-O2 -pipe' QMAKE_CXXFLAGS_RELEASE= QMAKE_CXXFLAGS_DEBUG= 'QMAKE_LFLAGS=-Wl,-O1 -Wl,--as-needed' QMAKE_LFLAGS_RELEASE= QMAKE_LFLAGS_DEBUG= PREFIX=/var/tmp/portage/media-video/webcamoid-7.2.1/image//usr -o Makefile ) && make -f Makefile install
make[1]: Entering directory '/var/tmp/portage/media-video/webcamoid-7.2.1/work/webcamoid-webcamoid-3710641/libAvKys'
cd Lib/ && ( test -e Makefile || /usr/lib64/qt5/bin/qmake /var/tmp/portage/media-video/webcamoid-7.2.1/work/webcamoid-webcamoid-3710641/libAvKys/Lib/Lib.pro 'QMAKE_AR=x86_64-pc-linux-gnu-ar cqs' QMAKE_CC=x86_64-pc-linux-gnu-gcc QMAKE_LINK_C=x86_64-pc-linux-gnu-gcc QMAKE_LINK_C_SHLIB=x86_64-pc-linux-gnu-gcc QMAKE_CXX=x86_64-pc-linux-gnu-g++ QMAKE_LINK=x86_64-pc-linux-gnu-g++ QMAKE_LINK_SHLIB=x86_64-pc-linux-gnu-g++ QMAKE_OBJCOPY=x86_64-pc-linux-gnu-objcopy QMAKE_RANLIB= QMAKE_STRIP= 'QMAKE_CFLAGS=-march=core2 -O2 -pipe' QMAKE_CFLAGS_RELEASE= QMAKE_CFLAGS_DEBUG= 'QMAKE_CXXFLAGS=-O2 -pipe' QMAKE_CXXFLAGS_RELEASE= QMAKE_CXXFLAGS_DEBUG= 'QMAKE_LFLAGS=-Wl,-O1 -Wl,--as-needed' QMAKE_LFLAGS_RELEASE= QMAKE_LFLAGS_DEBUG= PREFIX=/var/tmp/portage/media-video/webcamoid-7.2.1/image//usr -o Makefile ) && make -f Makefile install
make[2]: Entering directory '/var/tmp/portage/media-video/webcamoid-7.2.1/work/webcamoid-webcamoid-3710641/libAvKys/Lib'
install -m 755 -p libavkys.so.7.2.1 /var/tmp/portage/media-video/webcamoid-7.2.1/image/usr/lib/libavkys.so.7.2.1
ln -f -s libavkys.so.7.2.1 /var/tmp/portage/media-video/webcamoid-7.2.1/image/usr/lib/libavkys.so
ln -f -s libavkys.so.7.2.1 /var/tmp/portage/media-video/webcamoid-7.2.1/image/usr/lib/libavkys.so.7
ln -f -s libavkys.so.7.2.1 /var/tmp/portage/media-video/webcamoid-7.2.1/image/usr/lib/libavkys.so.7.2
make[2]: Leaving directory '/var/tmp/portage/media-video/webcamoid-7.2.1/work/webcamoid-webcamoid-3710641/libAvKys/Lib'
cd AkQml/ && ( test -e Makefile || /usr/lib64/qt5/bin/qmake /var/tmp/portage/media-video/webcamoid-7.2.1/work/webcamoid-webcamoid-3710641/libAvKys/AkQml/AkQml.pro 'QMAKE_AR=x86_64-pc-linux-gnu-ar cqs' QMAKE_CC=x86_64-pc-linux-gnu-gcc QMAKE_LINK_C=x86_64-pc-linux-gnu-gcc QMAKE_LINK_C_SHLIB=x86_64-pc-linux-gnu-gcc QMAKE_CXX=x86_64-pc-linux-gnu-g++ QMAKE_LINK=x86_64-pc-linux-gnu-g++ QMAKE_LINK_SHLIB=x86_64-pc-linux-gnu-g++ QMAKE_OBJCOPY=x86_64-pc-linux-gnu-objcopy QMAKE_RANLIB= QMAKE_STRIP= 'QMAKE_CFLAGS=-march=core2 -O2 -pipe' QMAKE_CFLAGS_RELEASE= QMAKE_CFLAGS_DEBUG= 'QMAKE_CXXFLAGS=-O2 -pipe' QMAKE_CXXFLAGS_RELEASE= QMAKE_CXXFLAGS_DEBUG= 'QMAKE_LFLAGS=-Wl,-O1 -Wl,--as-needed' QMAKE_LFLAGS_RELEASE= QMAKE_LFLAGS_DEBUG= PREFIX=/var/tmp/portage/media-video/webcamoid-7.2.1/image//usr -o Makefile ) && make -f Makefile install
make[2]: Entering directory '/var/tmp/portage/media-video/webcamoid-7.2.1/work/webcamoid-webcamoid-3710641/libAvKys/AkQml'
 * ACCESS DENIED:  mkdir:        /usr/lib64/qt5/qml/AkQml
mkdir: cannot create directory ‘/usr/lib64/qt5/qml/AkQml’: Permission denied

```

This is annoying to get webcamoid on gentoo.

hipersayanX, 2016-10-21:
Need a light on this. Are you installing it through ebuild? can you gist it?

bignaux, 2016-10-21:
I try to write the ebuild since i don't find anyone. basically , portage installs in a sandbox to verify that it doesn't install stuff in a bad way,  make -f Makefile install in AkQml doesn't follow . another boring stuff is that each time you release a new version, the directory has differents name ( webcamoid-webcamoid-3710641 in last) , that is not fine for such system.

hipersayanX, 2016-10-21:
Please, post the ebuild script using [gist](https://gist.github.com/), so I can spot better were is the problem.

bignaux, 2016-10-23:
https://gist.github.com/bignaux/5dda471354f00b833c598c3f2f5dcb06

hipersayanX, 2016-10-23:
**PREFIX**, in many cases, must point to */usr*, that is the path where the program files will be installed, */usr* is the default value. If you want to change the path where the files will be copied you must use the **INSTALL_ROOT** variable in **make**, [check this](https://github.com/webcamoid/webcamoid/blob/master/ports/archlinux/PKGBUILD#L33).

>another boring stuff is that each time you release a new version, the directory has differents name ( webcamoid-webcamoid-3710641 in last) , that is not fine for such system.

According to [this](https://devmanual.gentoo.org/ebuild-writing/variables/), just change 

    S="${WORKDIR}/webcamoid-webcamoid-3710641"

to something like this

    S="${WORKDIR}/${PN}-${PV}"

Also, not sure if it could be useful but, a long time ago, I wrote [this ebuild](https://github.com/webcamoid/webcamoid/blob/a9b7cf6477ce7966a199e755e80cb8efab3bdfd4/ports/gentoo/webcamoid-5.1.0.ebuild), but never tested, may be you can adapt it for latest version.

hipersayanX, 2017-05-16:
This issue seems to be solved, closing.

closed 2017-05-16 by hipersayanX