webcamoid/webcamoid #48: Videocapture.dll has crashed

Steamskunk, 2016-10-26:
I am trying to run webcamoid in Windows. When I try to run it crashes and it says *webcamoid has stopped working" Here is the data from the event logs: Log Name:      Application
Source:        Application Error
Date:          10/26/2016 12:58:11 PM
Event ID:      1000
Task Category: (100)
Level:         Error
Keywords:      Classic
User:          N/A
Computer:      Steamskunk-PC
Description:
Faulting application name: webcamoid.exe, version: 7.2.1.0, time stamp: 0x57c4a4a1
Faulting module name: VideoCapture.dll, version: 7.2.1.0, time stamp: 0x57c4a44d
Exception code: 0xc0000005
Fault offset: 0x0000000000005340
Faulting process id: 0x2b18
Faulting application start time: 0x01d22faa1d0ee942
Faulting application path: C:\Program Files (x86)\Webcamoid\bin\webcamoid.exe
Faulting module path: C:\Program Files (x86)\Webcamoid\lib\avkys\VideoCapture.dll
Report Id: edd6c90e-6daa-44fa-b553-d847450d2e22
Faulting package full name: 
Faulting package-relative application ID: 
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Application Error" />
    <EventID Qualifiers="0">1000</EventID>
    <Level>2</Level>
    <Task>100</Task>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2016-10-26T16:58:11.475955900Z" />
    <EventRecordID>3140</EventRecordID>
    <Channel>Application</Channel>
    <Computer>Steamskunk-PC</Computer>
    <Security />
  </System>
  <EventData>
    <Data>webcamoid.exe</Data>
    <Data>7.2.1.0</Data>
    <Data>57c4a4a1</Data>
    <Data>VideoCapture.dll</Data>
    <Data>7.2.1.0</Data>
    <Data>57c4a44d</Data>
    <Data>c0000005</Data>
    <Data>0000000000005340</Data>
    <Data>2b18</Data>
    <Data>01d22faa1d0ee942</Data>
    <Data>C:\Program Files (x86)\Webcamoid\bin\webcamoid.exe</Data>
    <Data>C:\Program Files (x86)\Webcamoid\lib\avkys\VideoCapture.dll</Data>
    <Data>edd6c90e-6daa-44fa-b553-d847450d2e22</Data>
    <Data>
    </Data>
    <Data>
    </Data>
  </EventData>
</Event>

hipersayanX, 2016-10-26:
What Windows version? This happens when starting the program o when playing the camera?

hipersayanX, 2017-01-28:
Tested in Windows 7, 8, and 10, 32 and 64 bits, can't reproduce the problem. But I had a crash in Windows 8 due to the [DEP](https://support.microsoft.com/en-us/help/875352/a-detailed-description-of-the-data-execution-prevention-dep-feature-in-windows-xp-service-pack-2,-windows-xp-tablet-pc-edition-2005,-and-windows-server-2003) that could be related to the user problem, setting **QMLSCENE_DEVICE=softwarecontext** environment variable solve the problem. Upgrading to Qt 5.8.0 may solve the problem without requiring **QMLSCENE_DEVICE** fix.

hipersayanX, 2017-05-16:
As said before, tested in Windows 7, 8, and 10, 32 and 64 bits, and everything seems to be ok, software render in Qml is enabled by default, maybe there could be some crashes related to DEP in Windows 8, didn't tried updating the system.

closed 2017-05-16 by hipersayanX