webcamoid/webcamoid #50: [QUESTION] Stream Screen through VideoCall apps

jonaspm, 2016-12-07:
Can I use webcamoid to stream my desktop through videocall services like Skype or Google Hangouts? Because I already have tried using virtual webcam without luck.
Thanks in advance!

hipersayanX, 2016-12-07:
Yes, virtual camera feature was made for that purpose.

>Because I already have tried using virtual webcam without luck.

Need more info. Which operating system? Skype and Google Hangouts didn't recognized the camera? Which other programs did you tested?. Please, include any other useful information.

jonaspm, 2016-12-09:
Software: Webcamoid 7.2.1 x64 Windows installable
OS: Windows 10 x64 SL
Device: Laptop with HD Camera
I managed to make the virtual camera work with FaceFlow.com (see picture 1)
[image]

However, Hangouts does not detect the virtual camera for some reason. (see picture 2)
[image]

jonaspm, 2016-12-09:
Skype Preview for Windows 10 also doesn't detect ManyCam nor Webcamoid
[image]

hipersayanX, 2016-12-09:
I've implemented the virtual camera as a DirectShow filter (based on all the documentation out there), for that reason Firefox has not problem detecting it. However, my guess, Google Hangouts doesn't detect devices loaded as a DirectShow filter. 

The reason because Google Hangouts it's detecting the ManyCam camera is because they are probably implementing the virtual camera as a [KMDF o UMDF](https://en.wikipedia.org/wiki/Windows_Driver_Frameworks). I can try writing a UMDF driver, if MinGW allow me to do that, but KMDF requires code signing 💵 .

hipersayanX, 2017-10-19:
Ok, I have good news!

I been experimenting with UMDF and KMDF drivers, and even when I was able to create a bare bone UMDF driver in GCC, at the end discarded the idea because not enough documentation to do something useful, and as I said before, KMDF requires signing, so these drivers are discarded.
Then I went for an [AVStream](https://github.com/Microsoft/Windows-driver-samples/tree/master/avstream/avshws) driver, the problem is that these drivers requires wdm.h, so these drivers are plain kernel drivers that also requires signing, so discarded too.

At end decided to install and test the virtual camera in Skype, and I can confirm that the **virtual webcam works out of the box in Skype**, but with one condition, **you must install the 32 bits version of Webcamoid**, and the reason is because, Skype as provided by Microsoft, is a 32 bits program only.

So I will update website and wiki, and close this. At the end it was a very easy to solve issue, I just over complicated things 😛

hipersayanX, 2017-10-19:
Website and wiki updated so closing.

closed 2017-10-19 by hipersayanX

jonaspm, 2017-10-20:
Great news! 💃