webcamoid/webcamoid #52: Build without PulseAudio

reagentoo, 2017-01-10:
In some distributions, users are accustomed to do without PulseAudio.
Possible build options:
USE_GSTREAMER: 0, 1; default: 0

hipersayanX, 2017-01-10:
Hi, this has been also discussed [here](https://github.com/webcamoid/webcamoid/issues/44). Next versions will support other audio systems too. You can also try the git version were the changes has been applied already.

reagentoo, 2017-01-10:
Initial webcamoid ebuild for Gentoo is now [here](https://github.com/reagentoo/gentoo-overlay/tree/master/media-video/webcamoid). Only for build from master git repo yet.
Please check dependencies. Also should make possibility to disable build with libs (such as jack, pulseaudio) without removing them from the system.

hipersayanX, 2017-01-10:
Great! 😄
About dependencies, for ffmpeg i will suggest to go for 3.1 as minimum requirement, since i will be pushing some changes that will break code compatibility with older versions. If possible, also add v4l2loopback as optional package (runtime only), for enabling the virtual camera. Everything else seems to be ok.

>Also should make possibility to disable build with libs (such as jack, pulseaudio) without removing them from the system.

Taking note.

reagentoo, 2017-01-11:
> Also should make possibility to disable build with libs (such as jack, pulseaudio) without removing them from the system.

I mean that it is necessary to make additional build options USE_JACK/PULSEAUDIO. Otherwise I will have to make sed *.pro files inside ebuild for disable build with that libs. Also I see some non unix libs check in libAvKys/libAvKys.pro (avfoundation, coreaudio, dshow, vcamwin, wasapi). It is necessary to disable for unix/linux.

hipersayanX, 2017-01-11:
>I mean that it is necessary to make additional build options USE_JACK/PULSEAUDIO. Otherwise I will have to make sed *.pro files inside ebuild for disable build with that libs.

I understand, I'll add a NO____ options for disabling libraries, ie. NOGSTREAMER, NOALSA, NOPULSEAUDIO, and so on.

>Also I see some non unix libs check in libAvKys/libAvKys.pro (avfoundation, coreaudio, dshow, vcamwin, wasapi). It is necessary to disable for unix/linux.

That's not needed, libraries are auto-detected ([this](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/libAvKys.pro#L39) and [this](https://github.com/webcamoid/webcamoid/tree/master/libAvKys/Tests)). Also you can check in the automated build that no patch is required ([this](https://travis-ci.org/webcamoid/webcamoid) and [this](https://ci.appveyor.com/project/hipersayanX/webcamoid)).

reagentoo, 2017-01-11:
Please tell me, does a webcamoid need a libuvc under linux?

hipersayanX, 2017-01-11:
Required, no, because in GNU/linux v4l2 is preferred. libuvc is an alternative I added to give a better support in OSX, but it still has some bugs that need to be solved by upstream. You can add libuvc to the git build if you want, for the release version I'm thinking in bundling a copy of libuvc with the patches applied depending on upstream move. I'll wait to the last moment and see what libuvc devs will do.

reagentoo, 2017-01-11:
I think a support of libuvc would be useful in future porting to Android. In Android libuvc proved to be quite good for some applications. But in Linux, it is desirable to have more alternatives.
Also, please tell me, how can I change webcamoid libraries install path from /usr/lib to /usr/lib64?

hipersayanX, 2017-01-11:
>I think a support of libuvc would be useful in future porting to Android.

That's is my plan for 9.x series 😉 

>how can I change webcamoid libraries install path from /usr/lib to /usr/lib64?

    qmake Webcamoid.pro LIBDIR=/usr/lib64 # other options here ...

More info [here](https://github.com/webcamoid/webcamoid/wiki/Raw-build-and-install).

reagentoo, 2017-01-13:
Unfortunately, I can't record video in Gentoo. Output video file is empty.
```
$ webcamoid 
qrc:/Webcamoid/share/qml/OutputConfig.qml:39:5: QML GroupBox: Binding loop detected for property "implicitWidth"
qrc:/Webcamoid/share/qml/OutputConfig.qml:39:5: QML GroupBox: Binding loop detected for property "implicitWidth"
qrc:/Webcamoid/share/qml/OutputConfig.qml:39:5: QML GroupBox: Binding loop detected for property "implicitHeight"
qrc:/Webcamoid/share/qml/OutputConfig.qml:39:5: QML GroupBox: Binding loop detected for property "implicitHeight"
qrc:/Webcamoid/share/qml/OutputConfig.qml:39:5: QML GroupBox: Binding loop detected for property "implicitWidth"
kf5.kio.core: Refilling KProtocolInfoFactory cache in the hope to find "mtp"
kf5.kio.core: Invalid URL: QUrl("Video 2017-01-13 10-49-06.mp4")
kf5.kio.core: KSambaShare: Could not find smb.conf!
kf5.kio.core: Invalid URL: QUrl("Video 2017-01-13 10-49-06.mp4")
Can't write header:  Error number 22 occurred
```

hipersayanX, 2017-01-13:
I recommend to open a new issue because it has nothing to do with the original subject.

hipersayanX, 2017-01-23:
The [**NOLIBRARY** options](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/libAvKys.pro#L42) can be used as:

    qmake Webcamoid.pro NOPULSEAUDIO=1 NOGSTREAMER=1 NOV4LUTILS=1 # and so on...

hipersayanX, 2017-05-16:
Support for other audio systems has been added in latest release.

closed 2017-05-16 by hipersayanX