webcamoid/webcamoid #53: Output video file is empty

reagentoo, 2017-01-13:
Can't record video in Gentoo. Output video file is empty.
My system Gentoo (~amd64). media-video/ffmpeg-3.2.2 flags:
```
USE="X alsa amr bzip2 encode fontconfig gpl hardcoded-tables iconv mp3 network opus postproc theora threads truetype vaapi vorbis x264 x265 zlib
(-altivec) -amrenc (-armv5te) (-armv6) (-armv6t2) (-armvfp) -bluray -bs2b -cdio -celt -chromaprint -cpudetection -debug -doc -ebur128 -fdk -flite -frei0r -fribidi -gcrypt -gme -gmp -gnutls -gsm -iec61883 -ieee1394 -jack -jpeg2k -kvazaar -ladspa -libass -libcaca -libilbc -librtmp -libsoxr -libv4l -lzma (-mipsdspr1) (-mipsdspr2) (-mipsfpu) (-mmal) -modplug (-neon) -nvenc -openal -opengl -openh264 -openssl -oss -pic (-pulseaudio) -rubberband -samba -schroedinger -sdl -snappy -speex -ssh -static-libs {-test} -twolame -v4l* -vdpau -vpx -wavpack -webp -xcb -xvid -zimg -zvbi" CPU_FLAGS_X86="aes avx avx2 mmx mmxext sse sse2 sse3 sse4_1 sse4_2 ssse3 -3dnow -3dnowext -fma3 -fma4 -xop"
FFTOOLS="aviocat cws2fws ffescape ffeval ffhash fourcc2pixfmt graph2dot ismindex pktdumper qt-faststart sidxindex trasher"
```
Console output:
```
$ webcamoid 
qrc:/Webcamoid/share/qml/OutputConfig.qml:39:5: QML GroupBox: Binding loop detected for property "implicitWidth"
qrc:/Webcamoid/share/qml/OutputConfig.qml:39:5: QML GroupBox: Binding loop detected for property "implicitWidth"
qrc:/Webcamoid/share/qml/OutputConfig.qml:39:5: QML GroupBox: Binding loop detected for property "implicitHeight"
qrc:/Webcamoid/share/qml/OutputConfig.qml:39:5: QML GroupBox: Binding loop detected for property "implicitHeight"
qrc:/Webcamoid/share/qml/OutputConfig.qml:39:5: QML GroupBox: Binding loop detected for property "implicitWidth"
kf5.kio.core: Refilling KProtocolInfoFactory cache in the hope to find "mtp"
kf5.kio.core: Invalid URL: QUrl("Video 2017-01-13 10-49-06.mp4")
kf5.kio.core: KSambaShare: Could not find smb.conf!
kf5.kio.core: Invalid URL: QUrl("Video 2017-01-13 10-49-06.mp4")
Can't write header:  Error number 22 occurred
```

hipersayanX, 2017-05-16:
Fixed in latest release.

closed 2017-05-16 by hipersayanX

PonasKovas, 2020-05-10:
Can reproduce with the latest build. (https://aur.archlinux.org/packages/webcamoid-git/)
OS: Archlinux, Webcamoid 8.7.1

```
> webcamoid                                                                                                                [0]
libv4l2: error getting pixformat: Invalid argument
Qt Quick Layouts: Detected recursive rearrange. Aborting after two iterations.
Qt Quick Layouts: Detected recursive rearrange. Aborting after two iterations.
Qt Quick Layouts: Detected recursive rearrange. Aborting after two iterations.
Qt Quick Layouts: Detected recursive rearrange. Aborting after two iterations.
libv4l2: error getting pixformat: Invalid argument
libv4l2: error getting pixformat: Invalid argument
Can't write header:  Error number 22 occurred
zsh: segmentation fault (core dumped)  webcamoid
```

hipersayanX, 2020-05-11:
@PonasKovas could be related to #297.