webcamoid/webcamoid #54: The usual upside downs

hamidrjafari, 2017-01-17:
Hello,

https://linuxtv.org/wiki/index.php/Libv4l_Upside_Down_Webcams
At the time this was the soft solution against the adventure of actually flipping the camera another 180 degrees. On a Kubuntu 16.10 with 4.8.0-34 kernel I ran the variant:

`$export LIBV4LCONTROL_FLAGS=3 && LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libv4l/v4l1compat.so webcamoid &`

and the output is:

file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/TextField.qml:635:5: QML TextInputWithHandles: Binding loop detected for property "text"
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/TextField.qml:635:5: QML TextInputWithHandles: Binding loop detected for property "text"
libv4l2: error setting pixformat: Device or resource busy
libv4l2: error setting pixformat: Device or resource busy
VideoCapture: No streams available.

Is it a 32 vs 64 bit mismatch? What can I do to make it run?

Hamid

hipersayanX, 2017-01-17:
What is the output of this command?

    v4l2-ctl -d /dev/video0 --list-formats-ext

hamidrjafari, 2017-01-17:
`$ v4l2-ctl -d /dev/video0 --list-formats-ext`
ioctl: VIDIOC_ENUM_FMT
        Index       : 0
        Type        : Video Capture
        Pixel Format: 'YUYV'
        Name        : YUYV 4:2:2
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                Size: Discrete 352x288
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                Size: Discrete 176x144
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                Size: Discrete 160x120
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)

hipersayanX, 2017-01-17:
When running Webcamoid, those resolutions are listed in the [right bar](https://raw.githubusercontent.com/webcamoid/webcamoid.github.io/master/data/screenshots/cap01.png), right?
Without LIBV4LCONTROL_FLAGS and LD_PRELOAD, you can play the webcam?

hamidrjafari, 2017-01-17:
Yes, running from the desktop works and the resolutions are listed but the picture is 180 degrees rotated.

P.S. I notice it adds up ~ %25 CPU overhead when running. Is that normal?

hipersayanX, 2017-01-17:
>Yes, running from the desktop works and the resolutions are listed but the picture is 180 degrees rotated.

Then, probably, Webcamoid is not detecting the resolutions offered by v4l1compat.so. It must be fixed in the next release, it's possible for you to test git version? If not, a quick patch wold be using the **Matrix Transform** effect with the following values:

    1  0 0
    0 -1 0

>P.S. I notice it adds up ~ %25 CPU overhead when running. Is that normal?

Not for sure, that's a performance issue, I will see what I can do.

hamidrjafari, 2017-01-17:
Resolutions seem picked up correctly. That transformation should be implemented which requires you finding out if a webcam is upside down or not. Does the webcam report or are you going to add some controls in the user interface?

I can build according to:
 [https://github.com/webcamoid/webcamoid/wiki/Build-and-install-the-Debian-package]

Is that the git version for test? Is the change applied currently to make a build?

hipersayanX, 2017-01-17:
>Does the webcam report or are you going to add some controls in the user interface?

It will use [v4lconvert by default](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VideoCapture/src/v4l2/src/capturev4l2.h#L32), I think it will solve this issue, at least in my case it adds the flip options to the controls when LIBV4LCONTROL_FLAGS is set.

>Is that the git version for test? Is the change applied currently to make a build?

Almost, but you doesn't need to install it for testing. follow [these instructions](https://github.com/webcamoid/webcamoid/issues/43#issuecomment-241734913). For step 1 the required dependencies are:

    sudo apt-get install pkg-config linux-libc-dev qt5-qmake qtdeclarative5-dev qtmultimedia5-dev libqt5opengl5-dev libqt5svg5-dev libavcodec-dev libavdevice-dev libavformat-dev libavutil-dev libswresample-dev libswscale-dev libgstreamer-plugins-base1.0-dev libpulse-dev libjack-dev libasound2-dev libv4l-dev

hamidrjafari, 2017-01-18:
I can afford to wait for the update if in the span of a month. Can use another system for the moment but would like to fix the issue.

hipersayanX, 2017-01-23:
Ok, I've found [some information](https://01.org/linuxgraphics/gfx-docs/drm/media/uapi/v4l/vidioc-enuminput.html#id3) that can be useful, the fields **V4L2_IN_ST_HFLIP** and **V4L2_IN_ST_VFLIP** can be used to detect if the image is flipped.

@hamidrjafari can you give me the output of this command?

    v4l2-ctl --device=/dev/video0 --list-inputs

hamidrjafari, 2017-01-25:
`$v4l2-ctl --device=/dev/video0 --list-inputs`

ioctl: VIDIOC_ENUMINPUT
        Input       : 0
        Name        : Camera 1
        Type        : 0x00000002
        Audioset    : 0x00000000
        Tuner       : 0x00000000
        Standard    : 0x0000000000000000 ()
        Status      : 0x00000000 (ok)
        Capabilities: 0x00000000 (not defined)

BTW I would like to solve this at the system level, where do you think I'd better raise the issue?

hipersayanX, 2017-01-25:
Status : 0x00000000 (ok)

According to the documentation Status must be returning some number != 0, but it doesn't, so the FLIP flags doesn't seems useful here. Anyway, you may not have any problem flipping the image with LIBV4LCONTROL_FLAGS=3 in next release. Also, did you tested the webcam with other programs? does it have the same problem?

>BTW I would like to solve this at the system level, where do you think I'd better raise the issue?

V4L is maintained by the LinuxTV team so probably the best place to report bugs is in their [mailing list and IRC](https://linuxtv.org/lists.php).

hipersayanX, 2017-05-16:
Webcamoid may recognize LIBV4LCONTROL_FLAGS=3 in latest release.

closed 2017-05-16 by hipersayanX