webcamoid/webcamoid #55: Question on source laid out.

PrincessAnnabelle, 2017-01-30:
Hi, so I'm trying to write a pet project where I can select the desktop and stream it as a webcam (like to skype or some other application) and I'm a bit lost reading the source. I'm trying to write my own example based off your code and would appreciate a bit of help.

I was wondering which classes/files I would need to look at/through to see just the desktop grabbing code, as well as the code needed to transfer/register the application as an actual webcam skype can react to.

It would help alot if you guys can help me out. Thanks.

hipersayanX, 2017-01-30:
It depends on the target operating system, if we are talking about GNU/Linux, Webcamoid use [v4l2loopback](https://github.com/umlaeute/v4l2loopback) for creating the fake webcam. Webcamoid follow the scheme:

* open
* read/write
* close

So in this case you have:

* [open](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VirtualCamera/src/v4l2/src/cameraoutv4l2.cpp#L525)
* [read/write](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VirtualCamera/src/v4l2/src/cameraoutv4l2.cpp#L198)
* [close](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VirtualCamera/src/v4l2/src/cameraoutv4l2.cpp#L574)

In Windows this is a more complicated task. You need to create a DirectShow filter where the entry point is [this](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VirtualCamera/src/dshow/VirtualCameraFilter/VirtualCameraSource/src/dllmain.cpp), and you have some documentation [here](https://msdn.microsoft.com/en-us/library/windows/desktop/dd391013(v=vs.85).aspx). You need to compile the [BaseClasses library](https://github.com/webcamoid/webcamoid/tree/master/libAvKys/Plugins/VirtualCamera/src/dshow/VirtualCameraFilter/BaseClasses/src) from [here](https://github.com/Microsoft/Windows-classic-samples/tree/master/Samples/Win7Samples/multimedia/directshow/baseclasses), and an [IPC bridge](https://github.com/webcamoid/webcamoid/tree/master/libAvKys/Plugins/VirtualCamera/src/dshow/VirtualCameraFilter/ipc/src) to send the frames to the filter, you have more info [here](https://msdn.microsoft.com/en-us/library/windows/desktop/aa365574(v=vs.85).aspx#base.using_a_file_mapping_for_ipc).

The corresponding ORWC scheme is this:

* [open](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VirtualCamera/src/dshow/src/cameraoutdshow.cpp#L420)
* [read/write](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VirtualCamera/src/dshow/src/cameraoutdshow.cpp#L197)
* [close](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VirtualCamera/src/dshow/src/cameraoutdshow.cpp#L428)

There is another way of creating a fake webcam in Windows that is creating a [UMDF or KMDF driver](https://msdn.microsoft.com/en-us/windows/hardware/drivers/wdf/writing-a-simple-kmdf-driver), but I didn't made any research in that field yet.

And finally in Mac, I didn't made any research yet, but you can start [here](https://github.com/rsodre/ofxFakam).

Hope this helps 😃

hipersayanX, 2017-01-30:
Oh, I forgot, for desktop capturing I just use [QScreen classes from Qt](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/DesktopCapture/src/desktopcaptureelement.cpp#L280t).

closed 2017-01-30 by hipersayanX

PrincessAnnabelle, 2017-01-30:
How exactly do you use a filter after compiling it? I mean you take the frames from QT classes, send to IPC bridge which sends to the er, directshow filter and then what? Does the filter automatically send to skype?

hipersayanX, 2017-01-30:
You must open the CMD as **administrator**, and then register the filter with:

    regsvr32 path/to/filter.dll

If you want to unregister the filter then:

    regsvr32 /u path/to/filter.dll

The target program must have support for DirectShow filters that is [not the case for Skype](https://github.com/webcamoid/webcamoid/issues/50), but well you can start hacking from there.

PrincessAnnabelle, 2017-01-30:
Ohh, so what would be a method to try and get it to work with Skype?

hipersayanX, 2017-01-30:
As I said before, start researching on [UMDF and KMDF drivers](https://msdn.microsoft.com/en-us/windows/hardware/drivers/wdf/writing-a-simple-kmdf-driver).