webcamoid/webcamoid #56: Does not install on OpenSUSE Tumbleweed (as of 2017.03.08)

AlexCzar, 2017-03-07:
```
$ sudo zypper ar -f http://download.opensuse.org/repositories/home:/hipersayan_x/openSUSE_Tumbleweed/ webcamoid
$ sudo zypper install webcamoid akqml avkys-plugins

Problem: nothing provides v4l2loopback-utils needed by webcamoid-7.2.1-5.17.x86_64
 Solution 1: do not install webcamoid-7.2.1-5.17.x86_64
 Solution 2: break webcamoid-7.2.1-5.17.x86_64 by ignoring some of its dependencies
```
Irrelevant output is omitted for brevity.

hipersayanX, 2017-03-08:
If zypper allow that, select Solution 2.

BTW, I will stop maintaining OBS repositories after 8.0.0 release, and start providing bundles for GNU/Linux instead. I possible contact OpenSUSE developers to include and keep maintaining an official RPM package in their repositories.

AlexCzar, 2017-03-20:
It got installed. But
1) virtual camera does not work, probably should have been expected, since it uses v4l2loopback
2) It shows all screens at once when I click play, I can't select to show only one screen
I understand that it is a burden to maintain packages for many distributions, but each time someone chooses to abandon SUSE or Fedora in favour of *buntu, a little kitten dies in my head :'(

hipersayanX, 2017-03-20:
>It shows all screens at once when I click play, I can't select to show only one screen

Does it show the screens side by side, or alternate between frames in one screen and the other?
<br />
<br />
<hr />

>I understand that it is a burden to maintain packages for many distributions, but each time someone chooses to abandon SUSE or Fedora in favour of *buntu, a little kitten dies in my head :'(

Of course I'm not abandoning support for OpenSUSE or any other distro. What I'll do now is:

1. Provide [automated builds](https://github.com/webcamoid/webcamoid/blob/master/ports/ci/travis/install_deps.sh#L112) for those who want to package Webcamoid for any distro, so they can see exactly which dependencies are needed. And
2. Provide bundles (probably with [AppImage](https://github.com/probonopd/AppImageKit)), and installers (with [QtIFW](https://github.com/qtproject/installer-framework)), so users will be able to install latest version, when is out, without the need of wait the distro maintainers to package it.

I have many strong reasons to do this.

1. Webcamoid was never part of the official repository in OpenSUSE and Fedora, you need to add the repository and run some [very specific commands](https://github.com/webcamoid/webcamoid/wiki/Binary-packages) to make it work there, and that looks very ugly for non-technical users.
2. When you go to the [Downloads area](http://webcamoid.github.io/#downloads), in Webcamoid website, the downloads area for GNU/Linux looks awful with all the mess of commands for each distro and versions of that distro, and it look very awful to users of any other operating system, so they will think two times if they want to install Webcamoid or even GNU/Linux. I don't want to show that to those users because is bad for us.
3. I'm not a Debian, Fedora, OpenSUSE or Ubuntu user, I can't keep track of all the changes and new releases these distros have, and I can't apply the very specific changes these distros requires for packaging. It will be a lot much better if someone who know better the distro create the package for whebcamoid.
4. It will reduce my work load so I can concentrate in fixing the problems and adding new features, and derive the work of packaging to distro maintainers, as it must be.
5. The changes to come between 7.x and 8.x series and the changes to come in future releases are big enough that reporting a bug or feature in an old release will result in a lot of [invalid issues](https://github.com/webcamoid/webcamoid/issues?q=is%3Aissue+label%3A%22Verdict%3A+Invalid%22+is%3Aclosed), and because I'm the only software developer of the project, I can't provide code backports or patches for old releases, it will stall the development in new versions, users needs to use latest version available.

All the changes I'm working in for 8.x will be a lot much better for everyone, developers, distro developers, and users, and it will be allow me to provide a better solution that was not provided until now in this field.

# TL;DR #

## New and fancy features for everyone a lot much faster! ##

AlexCzar, 2017-03-21:
It shows the screens side by side, as if it was one very wide screen. Here's a screenshot: [image]

hipersayanX, 2017-03-21:
Can you give me the output of [this script](https://gist.github.com/hipersayanX/e9bb6e892ed9ddfe9be0062c50326869)? you need to install PyQt5.

AlexCzar, 2017-03-23:
```python thisscript.py                                 
Screen info (0):
()
geometry(): PyQt5.QtCore.QRect(0, 360, 1920, 1080)
availableGeometry(): PyQt5.QtCore.QRect(40, 360, 1880, 1080)
virtualGeometry(): PyQt5.QtCore.QRect(0, 0, 7040, 1440)
availableVirtualGeometry(): PyQt5.QtCore.QRect(40, 0, 7000, 1440)
()
Desktop info (0):
()
screenGeometry(): PyQt5.QtCore.QRect(0, 360, 1920, 1080)
availableGeometry(): PyQt5.QtCore.QRect(40, 360, 1880, 1080)
()
Screen info (0):
()
geometry(): PyQt5.QtCore.QRect(1920, 0, 2560, 1440)
availableGeometry(): PyQt5.QtCore.QRect(1920, 0, 2560, 1440)
virtualGeometry(): PyQt5.QtCore.QRect(0, 0, 7040, 1440)
availableVirtualGeometry(): PyQt5.QtCore.QRect(40, 0, 7000, 1440)
()
Desktop info (1):
()
screenGeometry(): PyQt5.QtCore.QRect(1920, 0, 2560, 1440)
availableGeometry(): PyQt5.QtCore.QRect(1920, 0, 2560, 1440)
()
Screen info (0):
()
geometry(): PyQt5.QtCore.QRect(4480, 0, 2560, 1440)
availableGeometry(): PyQt5.QtCore.QRect(4480, 0, 2560, 1440)
virtualGeometry(): PyQt5.QtCore.QRect(0, 0, 7040, 1440)
availableVirtualGeometry(): PyQt5.QtCore.QRect(40, 0, 7000, 1440)
()
Desktop info (2):
()
screenGeometry(): PyQt5.QtCore.QRect(4480, 0, 2560, 1440)
availableGeometry(): PyQt5.QtCore.QRect(4480, 0, 2560, 1440)
```

hipersayanX, 2017-03-24:
@AlexCzar fixed, thanks 😄

hipersayanX, 2017-05-16:
Webcamoid may capture desktop as expected now. Releasing AppImages as promised.

closed 2017-05-16 by hipersayanX