webcamoid/webcamoid #57: linuxdeployqt

probonopd, 2017-03-21:
Hi, I assume you are aware of https://github.com/probonopd/linuxdeployqt but nevertheless wanted to implement it in bash. Would you consider contributing it for others to use, independently of webcamoid?

hipersayanX, 2017-03-22:
Sure, no problem 😄 it still requires some small adjustments, but I think it will be much better porting the code in Bash to C++, because solving dependencies require a stack that is quite complex to implement in Bash, and my current implementation can solve almost 90% of the dependencies but still need to run once more the get the 10% remaining, because of that.

Also, thanks for AppImage it's quite useful 👍

probonopd, 2017-03-22:
Thanks @hipersayanX, closing here. The C++ implementation currently also needs 2 runs, would love to have you review its code and possibly suggest what should be ported over from your bash implementation.

hipersayanX, 2017-03-22:
Yes, no problem, anyway I will be able to review it after Webcamoid 8.0.0 release, I don't want to delay it more.

probonopd, 2017-04-30:
Is an AppImage available for download somewhere?

hipersayanX, 2017-04-30:
No yet, it will be available for 8.0.0, hopefully next week.

hipersayanX, 2017-05-16:
AppImages [are ready](https://github.com/webcamoid/webcamoid/releases/tag/8.0.0), I've compiled in Trusty and tested in newer distros, and worked flawlessly 😄

probonopd, 2017-05-17:
Thank you very much, added to https://github.com/probonopd/AppImageKit/wiki/AppImages.

hipersayanX, 2017-05-18:
great!

closed 2017-05-18 by hipersayanX