webcamoid/webcamoid #60: webcamoid 7.2.1 can't add virtual device: wrong password

adamlazok, 2017-03-31:
Debian stretch, 
Webcamoid 7.2.1,
Qt 5.7.1,
FFmpeg 3.2.4-1

running from terminal as root:
bash: -c: row: 0: syntax error „(” unexpected token
bash: -c: row: 0: `modprobe v4l2loopback video_nr=1,2 card_label=Dummy\ video\ device\ (0x0000),Virtual\ Camera\ 2'

su:"must run from terminal", 
or such, i had to translate.
I've tried from terminal as root, as user, user was in the sudoers, everything. So am i missing something?

hipersayanX, 2017-03-31:
First at all, never run Webcamoid ar root, it does not need that.
About your problem, it seems to be this:

modprobe v4l2loopback video_nr=1,2 card_label=Dummy\ video\ device\ **(0x0000)**,Virtual\ Camera\ 2

Because the parentheses are not escaped, it's a bug, for the moment just set the device **Description** by hand to some thing like *My camera*, "virtual camera", or similar, without quotes or parentheses.

adamlazok, 2017-03-31:
[image]

I've just entered root password, i tried my own password, same result.

adamlazok, 2017-03-31:
I also tried w/o Description.

hipersayanX, 2017-03-31:
And removing all devices first, and then add a new webcam again?

adamlazok, 2017-03-31:
Nothing happens, just the password input field with the buttons disappears.

hipersayanX, 2017-03-31:
And the previous device is still there?

adamlazok, 2017-03-31:
Yes, the video1 device remains there.

hipersayanX, 2017-03-31:
Wait, try 

    sudo rmmod v4l2loopback

from console, close and reopen Webcamoid, and create a new virtual camera.

adamlazok, 2017-03-31:
"wrong password"

Devices field is empty. I removed the module before i started webcamoid (I'm sure: "rmmod: ERROR: Module v4l2loopback is not currently loaded").

hipersayanX, 2017-03-31:
And trying adding the camera from console like this?

    sudo modprobe v4l2loopback video_nr=1 card_label=Virtual \Camera

adamlazok, 2017-03-31:
Virtual (/dev/video1) appears in Preferences/Output, but still "wrong password".

hipersayanX, 2017-03-31:
Ok, I will test it later, thank for reporting this bug!

adamlazok, 2017-03-31:
Thank You very much, and let me know if I can help You in anything!

adamlazok, 2017-05-03:
[image]

Installing libjack-dev would like to remove libavdevice-dev, jackd2, pulseaudio-module-jack too.
libavdevice-dev is also on the dependency list. I'm using jackd2 so i can't uninstall. Can i help more?

hipersayanX, 2017-05-03:
Its possible to specify a build dependency to something like _libjack-dev | libjack-jackd2-dev_ in Debian, so it can build with one or the other? if that's the case just do that, if not just remove jack as dependency.

hipersayanX, 2017-05-16:
Webcamoid now support kdesu, gksu, and other graphical variants, that may solve the problem with the virtual camera.

closed 2017-05-16 by hipersayanX

Camaro77, 2022-03-17:
What password are they asking for?

hipersayanX, 2022-03-19:
@Camaro77 Use [latest version](https://github.com/webcamoid/webcamoid/releases/latest), old versions has no support.