webcamoid/webcamoid #62: Don't apply effects to Desktop Capture

xbox360pad, 2017-04-03:
First of all, excuse my bad english. I use ur program on Linux and i like it but, It's possible to apply effects only on webcam output? I use my webcam using some effect, but when i swith to desktop capture the effect is still applied, which does not make much sense. It's possible separet effects? Is it possible to configure the effects for each device individually?

Thanks in advance.

hipersayanX, 2017-04-03:
Is not possible to set a per device effect, but is planned in the very log term, maybe after Webcamoid 10.x or higher, but I do not promise anything, I want to add support for layouts first and then add that feature.

Storing/Restoring device configs will come **after** Webcamoid 8.0.0 release, maybe 8.1 o 8.2, but is planned in the short term.

hipersayanX, 2017-05-16:
As said before, implementing more complex audio and video combinations will take a very long time. But I can implement per device configs storing/Restoring, so changing the title of this issue.

hipersayanX, 2019-03-05:
Save/Restore webcam format, resolution and FPS may now be working well in 8.6.0.

closed 2019-03-05 by hipersayanX