webcamoid/webcamoid #63: keeps crashing

Tapatapatapatap, 2017-04-05:
OS, Ubuntu 16.10

Just installed webcamoid using the ppa, it opens fine, but randomly stops responding and either it closes on it's own or I must force quit the program or sometimes it freezes everything and so I have to reboot my system.
I was wondering what dependencies are needed for the program work? 
I have installed the guvcview webcam program and it works well so I know the webcam works with sound. Your program looks interesting and has some nice effects if I could just get it to work.

Thank you very much

hipersayanX, 2017-04-05:
I understand perfectly well your concerns, I will not say Webcamoid is perfect and don't crash, because is not true, but the reason because it, no so, randomly crashes is because it's multi-theaded and modular nature.

Let's say, for example, in the case of webcam capture, it needs a thread for reading the packets that comes from your webcam, and other for decompressing and convert the frames to some format that can be processed and presented in your screen, ie. you set Webcamoid to capture from your webcam in H264 format, then the frames will be decompressed in YUV420, because that's the most common format in which the frames are stored in that format , and finally the frames are converted to RGB32.

Let say you want to apply some heavy effect like Denoise with radius 5 or more, to a high resolution frame, let say 1920x1080, it will take more than 1 sec to process the frame, if you where using just one thread for everything the GUI will become absolutely unresponsive, with the frame processed in it's own thread you are able to use Webcamoid without much lag.

But here is the other thing, because Webcamoid has a modular design, camera capture is made in one module, and Denoise is made in other, both threads, for packet reading and packet processing are created in the camera capture module, and Denoise process the frames in the thread created by camera capture module, so when you remove or change an effect Webcamoid will crash because Denoise module is being destroyed when a buffer was being transferred from capture module.

The same exact happens when recording, that also has it's own thread, but in this case it can also freeze the GUI when writing the packets to the output file, and this normally happens with buggy codecs. The same happens with videos, the same happens with desktop capture.

But even with these problems, the current design gives me a lot of advantages, like for example I can easily port Webcamoid to any platform in which Qt is available or add new features with minimal effort. And anyway these problems can be fixed, and I been working a lot to fix them for 8.0.0, so maybe it may not crash as much as before, maybe (at least in my case it didn't crashed that much).

I hope it clarifies a bit this problem.

# TL;DR #

Webcamod 8.0.0 may not crash as much as before, hopefully.

Tapatapatapatap, 2017-04-05:
Thank you for your reply and also thank you for your continued effort to create a good piece of software. I will continue to play around with it and see if I can get it to work more reliably.

Good day and good luck on Webcamoid 8.0.0

hipersayanX, 2017-05-16:
Let say it's closed, maybe it's not but, because catching this kind of bugs is quite tricky I need more eyes that can see things I can't. Anyway, keep me informed if you find other bugs.

hipersayanX, 2017-05-18:
Will reopen this. In fact, it crash less compared to previous version, at least in GNU/Linux (in my experience), but it crash randomly in Windows, at this point I'm not sure if this a problem related to Webcamoid it self or it's something related to Qml/Qt, I need to think a battery of tests to hunt the bug.

wsteele, 2017-05-31:
On OSX Sierra it crashes on start.  I cannot get it to run.  Here is my crash log...

Process:               webcamoid.run [866]
Path:                  /Applications/webcamoid/webcamoid.app/Contents/MacOS/webcamoid.run
Identifier:            ASDF.webcamoid
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        ??? [863]
Responsible:           webcamoid.run [866]
User ID:               501

Date/Time:             2017-05-30 23:34:42.453 -0400
OS Version:            Mac OS X 10.12.5 (16F73)
Report Version:        12
Anonymous UUID:        63E2EBD6-7B1F-215A-1197-E6CB41BCC38C

Time Awake Since Boot: 200 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
Assertion failed: (mach_vm_map(mach_task_self(), (mach_vm_address_t*)&addr, size, 0, VM_FLAGS_ANYWHERE, dataPort, 0, false, VM_PROT_DEFAULT, VM_PROT_DEFAULT, VM_INHERIT_DEFAULT) == KERN_SUCCESS), function SLSHWCaptureDesktop, file Services/Windows/CGSCapture.c, line 495.
 

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff8bd19d42 __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff8be07457 pthread_kill + 90
2   libsystem_c.dylib             	0x00007fff8bc7f420 abort + 129
3   libsystem_c.dylib             	0x00007fff8bc46893 __assert_rtn + 320
4   com.apple.SkyLight            	0x00007fff88053c34 SLSHWCaptureDesktop + 4558
5   com.apple.SkyLight            	0x00007fff87fc23b3 SLDisplayCreateImageForRect + 93
6   libqcocoa.dylib               	0x0000000107f021a2 0x107ef9000 + 37282
7   org.qt-project.QtGui          	0x00000001047b7c74 QScreen::grabWindow(unsigned long long, int, int, int, int) + 340
8   libDesktopCapture.dylib       	0x000000010f395d50 DesktopCaptureElement::readFrame() + 384
9   org.qt-project.QtCore         	0x0000000104e50f1b QMetaObject::activate(QObject*, int, int, void**) + 2347
10  org.qt-project.QtCore         	0x0000000104e49176 QObject::event(QEvent*) + 102
11  org.qt-project.QtWidgets      	0x00000001057fbf52 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 306
12  org.qt-project.QtWidgets      	0x00000001057fd2a8 QApplication::notify(QObject*, QEvent*) + 360
13  org.qt-project.QtCore         	0x0000000104e2093f QCoreApplication::notifyInternal2(QObject*, QEvent*) + 159
14  org.qt-project.QtCore         	0x0000000104e7467a QTimerInfoList::activateTimers() + 890
15  libqcocoa.dylib               	0x0000000107f1d592 0x107ef9000 + 148882
16  com.apple.CoreFoundation      	0x00007fff76491321 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
17  com.apple.CoreFoundation      	0x00007fff76472197 __CFRunLoopDoSources0 + 423
18  com.apple.CoreFoundation      	0x00007fff76471716 __CFRunLoopRun + 934
19  com.apple.CoreFoundation      	0x00007fff76471114 CFRunLoopRunSpecific + 420
20  com.apple.HIToolbox           	0x00007fff759d2ebc RunCurrentEventLoopInMode + 240
21  com.apple.HIToolbox           	0x00007fff759d2bf9 ReceiveNextEventCommon + 184
22  com.apple.HIToolbox           	0x00007fff759d2b26 _BlockUntilNextEventMatchingListInModeWithFilter + 71
23  com.apple.AppKit              	0x00007fff73f6ba54 _DPSNextEvent + 1120
24  com.apple.AppKit              	0x00007fff746e77ee -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 2796
25  com.apple.AppKit              	0x00007fff73f603db -[NSApplication run] + 926
26  libqcocoa.dylib               	0x0000000107f1e21f 0x107ef9000 + 152095
27  org.qt-project.QtCore         	0x0000000104e1c8d2 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 418
28  org.qt-project.QtCore         	0x0000000104e20ff7 QCoreApplication::exec() + 343
29  webcamoid.run                 	0x000000010415a7ec main + 1228
30  libdyld.dylib                 	0x00007fff8bbeb235 start + 1

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff8bd1a44e __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8be0448e _pthread_wqthread + 1023
2   libsystem_pthread.dylib       	0x00007fff8be0407d start_wqthread + 13

Thread 2:
0   libsystem_kernel.dylib        	0x00007fff8bd1a44e __workq_kernreturn + 10
[... 494 lines cut ...]
CoreUI image file                  184K        5 
Dispatch continuations            8192K        2 
Foundation                           4K        2 
Image IO                            12K        4 
JS JIT generated code                4K        2 
JS VM register file               8192K        4 
JS garbage collector              28.4M       26 
Kernel Alloc Once                    8K        2 
MALLOC                           279.0M       57 
MALLOC guard page                   48K       10 
Memory Tag 242                      12K        2 
OpenGL GLSL                        256K        4 
STACK GUARD                       56.1M       16 
Stack                             15.1M       16 
VM_ALLOCATE                       21.3M       71 
__DATA                            35.4M      324 
__GLSLBUILTINS                    2588K        2 
__IMAGE                            528K        2 
__LINKEDIT                       122.4M       79 
__TEXT                           213.9M      313 
__UNICODE                          556K        2 
mapped file                       65.9M      361 
shared memory                     16.4M       13 
===========                     =======  ======= 
TOTAL                            949.4M     1349

hipersayanX, 2017-05-31:
@wsteele according with this:

    4 com.apple.SkyLight 0x00007fff88053c34 SLSHWCaptureDesktop + 4558
    5 com.apple.SkyLight 0x00007fff87fc23b3 SLDisplayCreateImageForRect + 93
    6 libqcocoa.dylib 0x0000000107f021a2 0x107ef9000 + 37282
    7 org.qt-project.QtGui 0x00000001047b7c74 QScreen::grabWindow(unsigned long long, int, int, int, int) + 340
    8 libDesktopCapture.dylib 0x000000010f395d50 DesktopCaptureElement::readFrame() + 384

It crash when grabbing the desktop, in my case when capturing the desktop it give me this result:

[image]

It seems both problems are related, I used to think it was a problem of my graphic driver, but it seem to be a bad implementation inside Qt. In GNU/Linux and Windows it doesn't happen.

Anyway, if Webcamoid doesn't start, you can try disabling webcam playback on start, go to the file _~/Library/Preferences/com.webcamoid.Webcamoid.plist_ and add a Boolean variable **StreamConfigs.playOnStart** with value **NO**, and restart Webcamoid.

wsteele, 2017-05-31:
@hipersayanX  Thanks for the suggestion.  I just tried changing the value to NO, but it still crashes and the plist file gets reset back to the old values after the crash.

hipersayanX, 2017-10-10:
I've, fixed many bugs for 8.1.0, I hope it works smoother now.

closed 2017-10-10 by hipersayanX