webcamoid/webcamoid #68: Can you make a virutal webcam with this program?

rogueyoshi, 2017-06-19:
As in, can you feed output to a program like XSplit or OBS. If so, I'd like to modify it to enable feeding QML output into such programs.

hipersayanX, 2017-06-19:
It depends, if you mean sending the output directly to this programs, the answer is yes if they provide an API for that; Otherwise it depends in the operating system and how these programs detects the webcams:

**If GNU/Linux 💚**

If they list the webcams from _/dev/video*_ and they are are not very picky about resolutions or formats (some programs are, Kamoso for example), then it's already possible installing **v4l2loopback**.

If the programs detects the webcams by listing USB devices (like Cheese), it's not possible yet, but I will be working on that for 8.2 release so both problems will be solved.

**If Mac ❤️**

I be experimenting with [kext programming](https://github.com/hipersayanX/QmakeKext) for the virtual camera, but it requires code signing to load and test the kext, I've seen [many projects](https://github.com/pje/WavTap) struggling with this and most of them died right after Apple enforced code signing. Not sure virtual camera will be possible in Mac.

**If Windows 💛**

As explained [here](https://github.com/webcamoid/webcamoid/issues/50#issuecomment-265911857), if the programs has compatibility for reading cameras from DirectShow filters, then it's already possible. Otherwise, I will try writing a UMDF driver for 8.2 or 8.3, that may work, if not it will share the same destiny as Mac.

💚 = Possible
💛 = May possible
❤️ = Kinda impossible

hipersayanX, 2017-07-24:
After some research, will change the status for **Mac** to kinda possible 💛, I will try adding support for [Syphon](http://syphon.v002.info/), that more or less seems to be an acceptable solution.

theonlygusti, 2017-08-28:
Hey I'd just like to clarify if it's possible to use webcamoid to stream some application through Skype?

hipersayanX, 2017-08-28:
@theonlygusti, Webcamoid 8.0.0 does not supports video output, but video output through Syphon was actually added in development version and will be available for 8.1.0 soon, so [all programs supporting Syphon](http://syphon.v002.info/) (see Applications section) will be be able to capture the output from Webcamoid, if Skype supports Syphon input or not i don't know.

hipersayanX, 2017-09-25:
Created [Virtual camera support](https://github.com/webcamoid/webcamoid/wiki/Virtual-camera-support) page in wiki, so closing issue.

hipersayanX, 2017-11-02:
Will reopen this. I have some advances for virtual camera in Mac. Will inform later.

hipersayanX, 2017-11-08:
I can confirm, Mac gonna have it's green badge too, so all major operating systems will have virtual camera support.

theonlygusti, 2017-11-08:
Is it possible to correct the title of this issue so that it comes up in searches for "virtual"?

hipersayanX, 2017-11-08:
Fixed.

camAtGitHub, 2017-11-23:
Can confirm using 32bit installable version on 64bit Windows 10, works as expected and the output 'device' was able to be used by other video conferencing software.

hipersayanX, 2018-01-10:
@theonlygusti virtual webcam working in Mac, and will be available for 8.5.0.
Now is the turn to rewrite Windows virtual webcam and fix issue #90 too.

TiborUdvari, 2018-05-06:
Hi, I managed to compile the `devel` branch on MacOS High Sierra. From the wiki I understood that this should be able to create a virtual webcam, but in the Preferences of webcamoid under `Virtual camera` it says this system is not supported yet. Is there any way of getting this to work? I have tried with `Photo booth` and `Skype` to check if it is working.

hipersayanX, 2018-05-06:
The "Wrong password" is actually misleading (I must fix that). In Qt Creator go to Projects > Active Project > Run, in  command lines arguments add __--vcam PATH/webcamoid/libAvKys/Plugins/VirtualCamera/src/cmio/VirtualCamera/AkVirtualCamera.plugin__, replace PATH with the path where Webcamoid actually is, and now it must be able to install and use the driver.

hipersayanX, 2018-05-06:
Just in Photo Booth? is Skype working?

hipersayanX, 2018-05-07:
Oh, I did not figured out at first, but you may have selected the debug camera, that camera doesn't receive frames from Webcamoid, it is used just as a reference. In Photo Booth for example, you must go to the menu in Camera option and select the camera you created, for example you created a camera called "My Virtual Camera", then you must select that camera in Photo Booth.
Also, for now, the cameras are removed when you close Webcamoid.

LinkMJB, 2018-07-06:
webcamoid  8.5.0 (development) freezes, and then presents with a memory leak when I attempt to create a virtual camera on macOS High Sierra 10.13.3. It eventually consumes all memory on the system if it isn't force killed when this condition is hit.

hipersayanX, 2018-07-06:
@LinkMJB, are _/Library/CoreMediaIO/Plug-Ins/DAL/AkVirtualCamera.plugin_ and  _~/Library/LaunchAgents_/AkVCamAssistant.plist created? if that is the case,  is the camera detected and working in Photo Booth or other webcam software?

LinkMJB, 2018-07-06:
Hey @hipersayanX, no those 2 files appear to be missing. I am not able to detect the virtual camera in other software.

LinkMJB, 2018-07-06:
It does look like the following directory structure was created today, however:

> ls -al /Library/CoreMediaIO/Plug-Ins/DAL/VirtualCamera/
total 4456
drwxr-xr-x   9 root  wheel      288 Jul  5 15:52 .
drwxr-xr-x   4 root  wheel      128 Jul  5 15:52 ..
drwxr-xr-x   3 root  wheel       96 Jul  5 15:52 AkVirtualCamera.plugin
-rw-r--r--   1 root  wheel     1604 Jul  5 15:52 Info.plist
-rw-r--r--   1 root  wheel    39703 Jul  5 15:52 Makefile
-rw-r--r--   1 root  wheel     2601 Jul  5 15:52 VirtualCamera.pro
drwxr-xr-x   3 root  wheel       96 Jul  5 15:52 build
-rwxr-xr-x   1 root  wheel  2229596 Jul  5 15:52 libAkVirtualCamera.dylib
drwxr-xr-x  21 root  wheel      672 Jul  5 15:52 src

hipersayanX, 2018-07-06:
@LinkMJB wow, that's actually weird :confused: , it copied the VirtualCamera source directory into the DAL instead of AkVirtualCamera.plugin. Maybe the copy function did not worked right.

Try deleting the VirtualCamera folder in _ /Library/CoreMediaIO/Plug-Ins/DAL/_. Then add this lines [here](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VirtualCamera/src/cmio/VCamIPC/src/ipcbridge.mm#L1432):

```
AkLoggerLog("driverPath: ", std::string(driverPath.begin(), driverPath.end()));
AkLoggerLog("dstPath: ", std::string(dstPath.begin(), dstPath.end()));
```

And rebuild (in Qt Creator) the whole [VirtualCamera tree](https://github.com/webcamoid/webcamoid/tree/master/libAvKys/Plugins/VirtualCamera).
And try installing the camera again. Those new lines may appear somewhere in the console after **IpcBridgePrivate::checkDaemon()**.

hipersayanX, 2019-01-22:
Fixed in 8.5.0.

**GNU/Linux**: Added a [new driver](https://github.com/webcamoid/akvcam) option with better compatibility with certain programs.
**Mac**: Removed Syphon and reimplemented the driver as a CoreMediaIO plugin. Some programs like PhotoBoth may require binary signing to work which is not possible right now, but is still a big advance anyway.
**Windows**: Driver still works with DirectShow but is much better than the previous driver.

Also added [compatibility tables](https://github.com/webcamoid/webcamoid/wiki/Virtual-camera-support).

closed 2019-01-22 by hipersayanX

malikAr, 2019-11-08:
Hello Guys,

Is it possible to only build a virtual camera for mac from this project? If Yes, please help me.

Any help is appreciable.
Thanks in advance.

hipersayanX, 2019-11-08:
@malikAr yes, you can pass **VIRTUALCAMERAONLY=1** when compiling [VirtualCamera.pro](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VirtualCamera/VirtualCamera.pro)

malikAr, 2019-11-10:
@hipersayanX: Thanks for soon reply.

I have got this error :-1: error: library not found for -lavkys.

Will you help me, please.

hipersayanX, 2019-11-11:
@malikAr, how did you ran the command?

malikAr, 2019-11-11:
I  have compiled through qt creator, and it shows me this error.

Please help.

hipersayanX, 2019-11-11:
@malikAr, open **VirtualCamera.pro** and in _Projects > Build Settings > Build Steps > Additional argumments_ add **VIRTUALCAMERAONLY=1**.

malikAr, 2019-11-13:
@hipersayanX Thanks, buddy. Compiled Virtual Module successfully. Will you please help me with how to load that plugin in the system, so that it shows in skype and zoom etc.

hipersayanX, 2019-11-14:
@malikAr just call [deviceCreate()](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VirtualCamera/src/ipcbridge.h#L152) from your program to install the camera and it will be detected by the target program. You can check in the [compatibility tables](https://github.com/webcamoid/webcamoid/wiki/Virtual-camera-support) if the target program is supported, not all versions of Skype are supported.
Also, you can check [virtualcameraelement.cpp](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VirtualCamera/src/virtualcameraelement.cpp) to unsderstand how to use the bridge.

malikAr, 2019-11-18:
**@hipersayanX**: should I copy AkVirtualCamera.plugin to /Library/CoreMediaIO/DAL or do use dylib

Thanks

hipersayanX, 2019-11-18:
@malikAr always use the bridge to manage the camera. A manual copy is not enough because you need to activate the assistant and write the plist files to _Library/Preferences folder_, the bridge does all of that for you.
You could eventually do it manually but you need to know [launchd](https://www.launchd.info/) and write the plist files manually if you know the format, but at the end is too much hassle.

malikAr, 2019-11-18:
@hipersayanX : will you please tell me about "always use the bridge to manage the camera." and how to use it.
Thanks

hipersayanX, 2019-11-18:
@malikAr as I say in my previous comment, call deviceCreate() and check the source code of virtualcameraelement.cpp.