webcamoid/webcamoid #69: Webcamoid does not open after a new install

alynur, 2017-06-24:
After installing Webcamoid into my Peppermint 8 desktop, the application will not open. Only the activity circle appears for about 30 seconds and then disappears and nothing happens. Any fix to this?

alynur, 2017-06-24:
I should have mentioned that I installed the app from peppermint Software Manager. I removed it thru the Software Manager and downloaded the 64 bit runable package. Now it does open and I can create a video clip but the clip has no sound. So I guess I'm half way there.

hipersayanX, 2017-06-24:
Hi, when installing from repositories, try running from terminal so you will be able to see which dependencies are missing.

And in latest version audio recording is disabled by default because there are some known bugs I'm trying to fix now.

alynur, 2017-06-25:
Thanks hipersayanX, that explains the no audio. I'll reverse my installations and try as you suggest, running in terminal. I'll report back here my results.

alynur, 2017-06-25:
Hello again hipersayanX, okay when I try to run webcamoid from the terminal, this is what I get
``albert@albert-desktop ~ $ webcamoid
QQmlApplicationEngine failed to load component
qrc:/Webcamoid/share/qml/main.qml:22 module "QtQuick.Dialogs" is not installed
qrc:/Webcamoid/share/qml/main.qml:21 module "QtQuick" is not installed
qrc:/Webcamoid/share/qml/main.qml:24 module "QtQuick.Controls" is not installed
qrc:/Webcamoid/share/qml/main.qml:23 module "QtQuick.Window" is not installed
qrc:/Webcamoid/share/qml/main.qml:26 module "QtGraphicalEffects" is not installed
qrc:/Webcamoid/share/qml/main.qml:25 module "QtQuick.Layouts" is not installed
qrc:/Webcamoid/share/qml/main.qml:22 module "QtQuick.Dialogs" is not installed
qrc:/Webcamoid/share/qml/main.qml:21 module "QtQuick" is not installed
qrc:/Webcamoid/share/qml/main.qml:24 module "QtQuick.Controls" is not installed
qrc:/Webcamoid/share/qml/main.qml:23 module "QtQuick.Window" is not installed
qrc:/Webcamoid/share/qml/main.qml:26 module "QtGraphicalEffects" is not installed
qrc:/Webcamoid/share/qml/main.qml:25 module "QtQuick.Layouts" is not installed
qrc:/Webcamoid/share/qml/main.qml:22 module "QtQuick.Dialogs" is not installed
qrc:/Webcamoid/share/qml/main.qml:21 module "QtQuick" is not installed
qrc:/Webcamoid/share/qml/main.qml:24 module "QtQuick.Controls" is not installed
qrc:/Webcamoid/share/qml/main.qml:23 module "QtQuick.Window" is not installed
qrc:/Webcamoid/share/qml/main.qml:26 module "QtGraphicalEffects" is not installed
qrc:/Webcamoid/share/qml/main.qml:25 module "QtQuick.Layouts" is not installed
qrc:/Webcamoid/share/qml/main.qml:22 module "QtQuick.Dialogs" is not installed
qrc:/Webcamoid/share/qml/main.qml:21 module "QtQuick" is not installed
qrc:/Webcamoid/share/qml/main.qml:24 module "QtQuick.Controls" is not installed
qrc:/Webcamoid/share/qml/main.qml:23 module "QtQuick.Window" is not installed
qrc:/Webcamoid/share/qml/main.qml:26 module "QtGraphicalEffects" is not installed
qrc:/Webcamoid/share/qml/main.qml:25 module "QtQuick.Layouts" is not installed
qrc:/Webcamoid/share/qml/main.qml:22 module "QtQuick.Dialogs" is not installed
qrc:/Webcamoid/share/qml/main.qml:21 module "QtQuick" is not installed
qrc:/Webcamoid/share/qml/main.qml:24 module "QtQuick.Controls" is not installed
qrc:/Webcamoid/share/qml/main.qml:23 module "QtQuick.Window" is not installed
qrc:/Webcamoid/share/qml/main.qml:26 module "QtGraphicalEffects" is not installed
qrc:/Webcamoid/share/qml/main.qml:25 module "QtQuick.Layouts" is not installed
qrc:/Webcamoid/share/qml/main.qml:22 module "QtQuick.Dialogs" is not installed
qrc:/Webcamoid/share/qml/main.qml:21 module "QtQuick" is not installed
qrc:/Webcamoid/share/qml/main.qml:24 module "QtQuick.Controls" is not installed
qrc:/Webcamoid/share/qml/main.qml:23 module "QtQuick.Window" is not installed
qrc:/Webcamoid/share/qml/main.qml:26 module "QtGraphicalEffects" is not installed
qrc:/Webcamoid/share/qml/main.qml:25 module "QtQuick.Layouts" is not installed

alynur, 2017-06-26:
I'm thinking that maybe since this program has known issues and I am unable to use it without audio, I should just uninstall it and call it a day. How do I uninstall the package I downloaded from the website?

hipersayanX, 2017-06-26:
You must install the following packages:

    sudo apt-get install qml-module-qtquick-dialogs qml-module-qtquick-controls qml-module-qtgraphicaleffects qml-module-qtquick-layouts

Also report this issue to the developers of your distro so they can fix the problem with packages dependencies.

>How do I uninstall the package I downloaded from the website?

* If you downloaded the AppImage just delete the file.
* If you downloaded the portable just delete the folder.
* If you downloaded the installer, you most probably installed Webcamoid in _~/Webcamoid_, so run the Maintenance Tool installed there.

Also, if you want to start with a fresh config delete the configuration folder (_~/.config/Webcamoid_):

    rm -rvf ~/.config/Webcamoid

hipersayanX, 2017-09-25:
I think the question has been solved, so closing.

closed 2017-09-25 by hipersayanX