webcamoid/webcamoid #71: My Camera Crashes When Turning to 1080p

rotospacestudios, 2017-07-18:
Hello,
I am using this webcam from amazon. I haven't changed any settings and just installed the program. I run Windows 10 x64
https://www.amazon.ca/gp/product/B017SRHRR6/ref=oh_aui_detailpage_o05_s00?ie=UTF8&psc=1

hipersayanX, 2017-07-18:
Not sure but could be related to the buffer size of the frame grabber. What is the maximum resolution and what are the listed formats?

Elmi77, 2017-07-24:
Same here. It happens with the integrated camera of my Laptop - and also resizing of the applications window causes this crash.

hipersayanX, 2017-07-24:
@Elmi77 same problem related to high resolution? any possibility to test if this also fails in GNU/Linux?

I'm not sure but the problem could be related on how the frames are rendered by Qml and not the buffer size.

Elmi77, 2017-07-25:
@hipersayanX using v8 and Fedora 23 the software works well. And yes, it seems to be related to a higher camera resolution.

hipersayanX, 2017-07-25:
@Elmi77 thank! so now I can narrow the problem to the DirectShow capture in Windows.
A last question, what capture formats supports your webcam and what is the maximum resolution?

hipersayanX, 2017-09-23:
Did not tried yet, but probably found the problem was maybe related to how video frames were displayed, and not DirectShow.

hipersayanX, 2017-10-10:
May be fixed in 8.1.0.

closed 2017-10-10 by hipersayanX