webcamoid/webcamoid #72: Feature Request: Zoom/crop function

Elmi77, 2017-07-24:
Hi,

there are a lot of (more or less) useful effects in Webcamoid, but one feature is missing (or at least I could not find it): the possibility to zoom into a view/to select a part of the image and to show only the selected part.

I have a camera with quite a huge resolution beyond of 3000 pixels width where I often want to see only a part of the whole image, so such a function would be very useful.

hipersayanX, 2017-07-24:
Plugins will requires the ability of modify the UI that is not possible right now, but is planned. So it will take a while but yes I like the idea.

larshb, 2017-12-07:
Would be even nicer paired with desktop output.

hairypaulsack, 2018-01-10:
@larshb when you say 
> Would be even nicer paired with desktop output.
are you referring to a feature where instead of the whole desktop screen being the output source, but rather being able to zoom or crop (essentially using a section of the desktop specified by the user)?

@hipersayanX any chance you've had a chance to consider this proposal? Is it an option?

hipersayanX, 2018-01-10:
@hairypaulsack sure, not for 8.5.0 (next release), but before 9.x series for sure.

system123, 2019-04-05:
Any updates on this?

hipersayanX, 2019-04-05:
@system123 nop, right now I'm focusing all efforts into Android port, and later I will go for UI redesign, both are very heavy tasks so I will only pay attention to bugs that can be solved quickly, and let features for later, unless someone want to work on that of course :wink:

jcolson, 2019-10-11:
why does the UI need to change in order for this to be accommodated?  just allow a bounding box entry as config to the filter?

hipersayanX, 2019-10-12:
@jcolson my idea is to add a control that create a selection box on top of the frame, something similar to this:

[image]

but then I need to add a control layer on top of the frame, so the plugins can add controls on top of it. And even if I can do that right now, it make no sense to do it at this point because I'm already planning to rewrite the UI for Android, so I won't write a functionality to rewrite it again in the near future.
So yes, I'll add this feature but this is not the right moment to do so.

jcolson, 2019-10-12:
ah!  going for the fancyness ;-)  understood

beppe9000, 2020-04-21:
+1 for cropping!

but also for progrssive functionality: if a plugin can take topleft and bottomright coordinates as number entries it could be enhanced at a future point with the ability to fill those field using a GUI control

tamo, 2020-07-09:
FYI, I use "Matrix Transform" plugin with [ 1 0 0 ; 0 0.8 0 ] to convert my 16:9 camera to the virtual camera (4:3) for Zoom conferences.

If you want a simple 2x zooming, you can set [ 2 0 0 ; 0 2 0 ].

DurvalMenezes, 2020-10-05:
Hello @tamo, 
> If you want a simple 2x zooming, you can set [ 2 0 0 ; 0 2 0 ].

I would like to limit the output to the left half of the screen. Is that possible with Matrix Transform? How?

Also, could you please provide a link to documentation on how to use Matrix Transform? I searched but could not find anything. 
Thanks in advance!

tamo, 2020-10-10:
Neither did I understand how to use it at first. But those who are familiar with mathematics may not realize the need for documentation. For those who are not, here is an intuitive demo: 
https://understanding-transform-matrix.ginpei.info/
Click "()" at the bottom.

DurvalMenezes, 2020-10-10:
Thanks for the link, @tamo. Great for understanding translation, rotation and resizing/zoom, but unfortunately could not figure out anything about cropping... Am I missing something?

tamo, 2020-10-10:
You can move the green point. Your [ 1 0 220; 0 1 0 ] may be equivalent to moving the point 50px aside. (Assume your width is 440 and the demo image width is 100)
[image]

DurvalMenezes, 2020-10-10:
Hello @tamo 

I tried it but it only resulted in a translation (ie, simple movement of the entire picture) to the right, ie no cropping: 
[image]

I  must be missing something... perhaps you mean that, on webcamoind, there would be an automatic crop to the orange square in the picture you included, which doesn't happen in the intuitive demo website?

Cheers, 
-- Durval.

tamo, 2020-10-11:
> I must be missing something... perhaps you mean that, on webcamoid, there would be an automatic crop to the orange square in the picture you included, which doesn't happen in the intuitive demo website?

Beautiful! That's exactly what I meant. Thanks for your perseverance and understanding my poor English!

DurvalMenezes, 2020-10-11:
Thank you @tamo-san for your patience, and by the way there's nothing wrong with your English.

Slion, 2021-04-14:
Still no actual cropping with the matrix transform though. It's just the zoom, translation and skew right?

spackmat, 2021-04-14:
I moved on to OBS Studio, which can do a lot more than cropping and zooming and for that simple tasks is quite simple to use: Just start the virtual camera, add your camera as a camerasource and transform it as required.

iam-TJ, 2022-06-04:
This is functionality I'm investigating implementing. Initially I considered adding a new effect plug-in but after reviewing the **Scale** effect it seems to me it would be far more efficient (less function call indirection for stream chains) to integrate crop and zoom into Scale since the two operations are directly related to scaling (zoom is a scale operation with the same aspect ratio).

`ScaleElement::iVideoStream()` [already calls `QImage.scaled(width, height, aspectRatio, mode)`](https://github.com/webcamoid/webcamoid/blob/d885fd36ef662786ae9c5a28aa5df1a58f897df9/libAvKys/Plugins/Scale/src/scaleelement.cpp#L112) - for cropping we'd add a call to `QImage.copy(QRect ...)`.

I'd implement it in two stages:
1. User manually provides top-left coordinate (x,y) via GUI for crop, and a decimal multiplier for zoom, via text boxes and/or number spinners or sliders.
2. Implement a stand-alone rubber-band selection operation that operates on top of the main view, and then link that into this plug-in (and any others that need the functionality).

I'd keep the option to manually enter the values to allow easy input of precise values.

rwperrott, 2022-10-10:
I would like to see full resolution support for higher than screen resolution cameras, via vertical and horizontal image/video scrollbars, and a drop-down resolution selector. It is very annoying and ridiculous that only Android USB camera apps currently seem able to zoom in and pan across a full resolution video from my 5 M pixel USB microscope. I may also be useful to have a separate pane/window showing a small scale image, with a block showing the visible section, which can be drag panned for faster coarse positioning than via both scrollbars; this could also be useful for scaled-up zoomed images. To maximise available screen space, the left and right borders will need to go, together with all the controls on them; the controls could be accessed/revealed via a top or bottom menu bar, or edge of window mouse hover triggers.

I'm using my 5 M pixel USB microscope camera on a Linux Mint 21 MATE 64-bit machine, and webcamoid seems to provide the highest frame rate of the Linux webcam viewers I've tried. I was able to view full 5M pixel resolution in Java with https://github.com/sarxos/webcam-capture with a Swing ScrollPane, but sadly the frame rate was under 2 frames per second, so a lot slower than webcamoid.

If I can build webcamoid, using KDevelop, I'll see if I can find the QT equivalent of a Java Swing ScrollPane and integrate this, then look at adding a dropdown selector for powers of 2 divided resolutions, supported by the camera.

hipersayanX, 2022-10-15:
@rwperrott please, open a new issue for your problem, this issue has nothing to do with what you want.

hipersayanX, 2023-06-10:
Plugin added in Webcamoid 9.1.0.

closed 2023-06-10 by hipersayanX