webcamoid/webcamoid #73: Webcam not found - Mac OSX

aryus96, 2017-08-04:
Hello,

The webcam is not recognize on a Imac7.1 2007, on OSX El Capitan 10.11.6.
Tried in portable version and installation.
It doesn't find my webcam, and I need to uninstall it if I want to use my webcam again (for Photobooth or Facetime), because it crashes the others apps.

Regards

hipersayanX, 2017-08-04:
Webcamoid use libuvc for webcam capture by default with AV Foundation as optional backend, first at all I need to know if you can make it work with AV Foundation, in Webcamoid go to Preference button > General Options > Video capture and select **avfoundation**, and restart Webcamoid, and let me know if it works for you.

>I want to use my webcam again (for Photobooth or Facetime), because it crashes the others apps.

One possible way Webcamoid will crash other apps is because you are trying to access the webcam  in other programs while playing the webcam in Webcamoid, because it doesn't modify nothing in the system. The other way is probably due to a bug in **libuvc** or even **libusb**. In both cases, a simple restart of the system may solve the problem.

Anyway, for uninstalling Webcamoid just delete the bundle or go to the Maintenance Tool created by the installer and uninstall it.

hipersayanX, 2017-08-04:
Great! 😄 
Next release will use AV Foundation by default, libuvc is causing seems to be causing many problems.

hipersayanX, 2017-10-10:
Fixed in Webcamoid 8.1.0, now uses AVFoundation by default.