webcamoid/webcamoid #76: Building on Mac - what do I do after opening with Qt Creator?

theonlygusti, 2017-08-28:
I can't figure out how to complete step 6 of the mac build instructions:

> In Build, in Build environment add /usr/local/bin to PATH variable.

This is what I see when I open `Webcamoid.pro` in Qt Creator:

[image]

hipersayanX, 2017-08-28:
Click in **Configure project**, wait until the project is loaded, then go to the left bar in **Projects** > **Build & Run** > **Build** > **Build Environment** > **PATH** and click **Edit**.

theonlygusti, 2017-08-28:
Do you mean click the button on the right that says "Configure Project"? It's disabled and I can't click it.

The screen shot I've sent is exactly what I see after opening Webcamoid.pro, and it's been like this for hours.

hipersayanX, 2017-08-28:
Click in **Details**, check all checkboxes and then click in **Configure Project**.

theonlygusti, 2017-08-28:
Details has no checkboxes:

[image]

hipersayanX, 2017-08-28:
You must install [Xcode](https://developer.apple.com/xcode/) and [command line tools](http://railsapps.github.io/xcode-command-line-tools.html).

theonlygusti, 2017-08-28:
I've already done that

theonlygusti, 2017-08-28:
Oh, I just found out about the GH pages site.

All I want to do is have webcamoid on my computer for streaming certain apps through things like Skype, 

can I just download the installer from webcamoid.github.io?

hipersayanX, 2017-09-01:
Answered [here](https://github.com/webcamoid/webcamoid/issues/68#issuecomment-325493470), so closing this issue.

closed 2017-09-01 by hipersayanX