webcamoid/webcamoid #77: Could you make it integrate into the system?

Lvaskz, 2017-09-01:
Hi, this is more a request than a problem, although perhaps for some people it is.
Appimage is not integrated into the linuxmint system 18.2. In some appimages at the time of making the application executable, the application asked if you would like integrate it into the system, but I see that it does not happen now. Nor the official version of krita. Could you make it possible in any way?
When you have many appimage not integrated in the system, it gets some tedious fill the desktop with shortcuts
PD: Can you integrate a icon in the system?
thanks

hipersayanX, 2017-09-01:
Which type of integration do you want exactly? How Webcamoid may interact with the desktop? How those shortcuts are supposed to work? I need more information to see what features are viable and what features are not.

>PD: Can you integrate a icon in the system?

The installer can already do that, but for AppImage is not possible, unless I put an AppImage inside an installer, and modify the system to do that in the installer, which will be like a Frankenstein package, not sure.

Lvaskz, 2017-09-01:
Hi, in some applications at the time of making the application executable,
a dialog box appears, asking if you want to integrate the application on
the desktop, to appear in the menu of installed programs.

Hola, en algunas appimage al momento de hacer la aplicacion ejecutable, le
aparece un cuadro de dialogo, preguntandole si desea integrar la aplicación
en el escritorio, para que aparezca en el menú de programas
instalados.Several appimages have this system (?) By default, like
kdevelop, openshot, freecad, etc. Obviously, appimage is a portable
application, but somehow, several appimages ask if it wants to appear in
the list of installed applications, even if appimage is in downloads or in
my documents, etc.

I hope I have explained the situation, if not, I will send some images.

Thank you

hipersayanX, 2017-09-01:
Yes, right now I understand what do you want, but that's already implemented in the [installable package](http://webcamoid.github.io/#downloads).

AppImages were designed with the idea of making applications portable, that is moving the application to a different folders, carrying it in an USB drive, etc., and then click and make it run without modifying the environment.

Installables were designed to put the application in a fixed place and configure the environment (icons, shortcuts, services, etc.) according to that path, because the installed files will not move from there. When you don't need the application anymore and do you want to uninstall it, the uninstaller will revert all changes it did in the environment, leaving a clean environment.

I don't think the application must be doing the job of the installer, because if you move the application to another place you will need to reconfigure again, and if you delete the AppImage, all the modifications it did before will be still there pointing to an application that's not anymore there, I don't want the application to mess with the system.

So what do you want is the installer, not the AppImage.

Lvaskz, 2017-09-01:
Ok, thanks 😀

hipersayanX, 2017-09-01:
Closing then 😄

closed 2017-09-01 by hipersayanX