webcamoid/webcamoid #82: Can't build on mac

theonlygusti, 2017-09-26:
I'm trying to follow the Mac build and install instructions.

I am stuck after opening Qt Creator.

[image]

I can't see how to add /usr/local/bin to PATH variable.

hipersayanX, 2017-09-27:
In the menu, go to _Qt Creator > Preferences > Build & Run_ and take a screenshot of Kits, Qt Versions, Compilers and Debuggers.

theonlygusti, 2017-09-27:
[image]
[image]
[image]
[image]

hipersayanX, 2017-09-27:
In **Qt Versions**, click in **Add** and then press ⇧⌘G and write _/usr/local/Cellar/qt_, go into the folder of the Qt version you have installed, then select _bin/qmake_.
Now go to **Kits**, select de Desktop kit, and then configure as follow:

**Device type**: Desktop
**Compiler C/C++**: Clang (x86 64 bit in /usr/bin)
**Debugger**: System LLDB at /usr/bin/lldb
**Qt version**: the Qt version you configured just before

And then you will have a kit ready to build Webcamoid or any other project.

theonlygusti, 2017-09-27:
During building I get

    [libAging.dylib] Error 2 (ignored)

I press CMD+R afterwards to Run and it asks if I should ignore errors, i say yes, but webcamoid has no windows. Just says "webcamoid" in menu bar.

[image]

hipersayanX, 2017-09-27:
The problem with clang in Mac is that sometimes errors are not very descriptive at all 😕, so unless you dive into the build log is quite impossible to know why that error. The error may be marked red somewhere there.

>I press CMD+R afterwards to Run and it asks if I should ignore errors, i say yes, but webcamoid has no windows. Just says "webcamoid" in menu bar.

If you can launch Webcamoid then go to _Projects > Active Project > Run_, then in **Command line arguments** write _-q /path/to/webcamoid/libAvKys_ (replace /path/to with the path where is located the project).
Also at the beginning of compile must appear a some lines similar to this:

    Checking for alsa... no
    Checking for avfoundation... yes
    Checking for coreaudio... yes
    Checking for dshow... no
    Checking for ffmpeg... yes
    Checking for ffmpeg_avcodec_contextframerate... yes
    Checking for ffmpeg_avcodec_extracodecformats... yes
    Checking for ffmpeg_avcodec_freecontext... yes
    Checking for ffmpeg_avcodec_packetref... yes
    Checking for ffmpeg_avcodec_rescalets... yes
    Checking for ffmpeg_avcodec_sendrecv... yes
    Checking for ffmpeg_avcodec_subtitledata... yes
    Checking for ffmpeg_avformat_allocoutputcontext... yes
    Checking for ffmpeg_avformat_codecpar... yes
    Checking for ffmpeg_avutil_extraoptions... yes
    Checking for ffmpeg_avutil_extrapixformats... yes
    Checking for ffmpeg_avutil_framealloc... yes
    Checking for ffmpeg_avutil_sampleformat64... yes
    Checking for ffmpeg_avresample... yes
    Checking for ffmpeg_swresample... yes
    Checking for gstreamer... yes
    Checking for jack... yes
    Checking for libuvc... yes
    Checking for libuvcdev... yes
    Checking for oss... no
    Checking for pulseaudio... yes
    Checking for qtaudio... yes
    Checking for syphon... yes
    Checking for v4l2... no
    Checking for vcamwin... no
    Checking for wasapi... no

Can you past the checking lines for you? you need at least the avfoundation, coreaudio and ffmpeg lines to be yes to get Webcamoid working properly.

theonlygusti, 2017-09-27:
I don't see any lines like that. Under Compile Output I get

```
19:48:15: Running steps for project Webcamoid...
19:48:15: Configuration unchanged, skipping qmake step.
19:48:15: Starting: "/usr/bin/make" 
cd libAvKys/ && ( test -e Makefile || /usr/local/Cellar/qt/5.9.1/bin/qmake -o Makefile /Users/George/Installations/webcamoid/libAvKys/libAvKys.pro -spec macx-g++ CONFIG+=debug CONFIG+=x86_64 CONFIG+=qml_debug ) && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile 
cd Lib/ && ( test -e Makefile || /usr/local/Cellar/qt/5.9.1/bin/qmake -o Makefile /Users/George/Installations/webcamoid/libAvKys/Lib/Lib.pro -spec macx-g++ CONFIG+=debug CONFIG+=x86_64 CONFIG+=qml_debug ) && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile 
make[2]: Nothing to be done for `first'.
cd AkQml/ && ( test -e Makefile || /usr/local/Cellar/qt/5.9.1/bin/qmake -o Makefile /Users/George/Installations/webcamoid/libAvKys/AkQml/AkQml.pro -spec macx-g++ CONFIG+=debug CONFIG+=x86_64 CONFIG+=qml_debug ) && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile 
make[2]: Nothing to be done for `first'.
cd Plugins/ && ( test -e Makefile || /usr/local/Cellar/qt/5.9.1/bin/qmake -o Makefile /Users/George/Installations/webcamoid/libAvKys/Plugins/Plugins.pro -spec macx-g++ CONFIG+=debug CONFIG+=x86_64 CONFIG+=qml_debug ) && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile 
cd ACapsConvert/ && ( test -e Makefile || /usr/local/Cellar/qt/5.9.1/bin/qmake -o Makefile /Users/George/Installations/webcamoid/libAvKys/Plugins/ACapsConvert/ACapsConvert.pro -spec macx-g++ CONFIG+=debug CONFIG+=x86_64 CONFIG+=qml_debug ) && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile 
cd src/ && ( test -e Makefile || /usr/local/Cellar/qt/5.9.1/bin/qmake -o Makefile /Users/George/Installations/webcamoid/libAvKys/Plugins/ACapsConvert/src/src.pro -spec macx-g++ CONFIG+=debug CONFIG+=x86_64 CONFIG+=qml_debug ) && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile 
make[4]: Nothing to be done for `first'.
cd src/ffmpegav/ && ( test -e Makefile || /usr/local/Cellar/qt/5.9.1/bin/qmake -o Makefile /Users/George/Installations/webcamoid/libAvKys/Plugins/ACapsConvert/src/ffmpegav/ffmpegav.pro -spec macx-g++ CONFIG+=debug CONFIG+=x86_64 CONFIG+=qml_debug ) && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile 
make[4]: Nothing to be done for `first'.
cd src/ffmpegsw/ && ( test -e Makefile || /usr/local/Cellar/qt/5.9.1/bin/qmake -o Makefile /Users/George/Installations/webcamoid/libAvKys/Plugins/ACapsConvert/src/ffmpegsw/ffmpegsw.pro -spec macx-g++ CONFIG+=debug CONFIG+=x86_64 CONFIG+=qml_debug ) && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile 
make[4]: Nothing to be done for `first'.
cd AudioDevice/ && ( test -e Makefile || /usr/local/Cellar/qt/5.9.1/bin/qmake -o Makefile /Users/George/Installations/webcamoid/libAvKys/Plugins/AudioDevice/AudioDevice.pro -spec macx-g++ CONFIG+=debug CONFIG+=x86_64 CONFIG+=qml_debug ) && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile 
cd src/ && ( test -e Makefile || /usr/local/Cellar/qt/5.9.1/bin/qmake -o Makefile /Users/George/Installations/webcamoid/libAvKys/Plugins/AudioDevice/src/src.pro -spec macx-g++ CONFIG+=debug CONFIG+=x86_64 CONFIG+=qml_debug ) && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile 
make[4]: Nothing to be done for `first'.
cd src/coreaudio/ && ( test -e Makefile || /usr/local/Cellar/qt/5.9.1/bin/qmake -o Makefile /Users/George/Installations/webcamoid/libAvKys/Plugins/AudioDevice/src/coreaudio/coreaudio.pro -spec macx-g++ CONFIG+=debug CONFIG+=x86_64 CONFIG+=qml_debug ) && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile 
make[4]: Nothing to be done for `first'.
cd src/jack/ && ( test -e Makefile || /usr/local/Cellar/qt/5.9.1/bin/qmake -o Makefile /Users/George/Installations/webcamoid/libAvKys/Plugins/AudioDevice/src/jack/jack.pro -spec macx-g++ CONFIG+=debug CONFIG+=x86_64 CONFIG+=qml_debug ) && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile 
make[4]: Nothing to be done for `first'.
cd src/pulseaudio/ && ( test -e Makefile || /usr/local/Cellar/qt/5.9.1/bin/qmake -o Makefile /Users/George/Installations/webcamoid/libAvKys/Plugins/AudioDevice/src/pulseaudio/pulseaudio.pro -spec macx-g++ CONFIG+=debug CONFIG+=x86_64 CONFIG+=qml_debug ) && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile 
make[4]: Nothing to be done for `first'.
cd src/qtaudio/ && ( test -e Makefile || /usr/local/Cellar/qt/5.9.1/bin/qmake -o Makefile /Users/George/Installations/webcamoid/libAvKys/Plugins/AudioDevice/src/qtaudio/qtaudio.pro -spec macx-g++ CONFIG+=debug CONFIG+=x86_64 CONFIG+=qml_debug ) && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile 
make[4]: Nothing to be done for `first'.
cd AudioGen/ && ( test -e Makefile || /usr/local/Cellar/qt/5.9.1/bin/qmake -o Makefile /Users/George/Installations/webcamoid/libAvKys/Plugins/AudioGen/AudioGen.pro -spec macx-g++ CONFIG+=debug CONFIG+=x86_64 CONFIG+=qml_debug ) && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile 
make[3]: Nothing to be done for `first'.
cd Bin/ && ( test -e Makefile || /usr/local/Cellar/qt/5.9.1/bin/qmake -o Makefile /Users/George/Installations/webcamoid/libAvKys/Plugins/Bin/Bin.pro -spec macx-g++ CONFIG+=debug CONFIG+=x86_64 CONFIG+=qml_debug ) && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Mak [... cut ...] cations/Xcode.app/Contents/Developer/usr/bin/gcc/debug/obj/moc_updates.o   -F/usr/local/Cellar/qt/5.9.1/lib -L/Users/George/Installations/webcamoid/StandAlone/../libAvKys/Lib -lavkys -framework QtQuick -framework QtGui -framework QtCore -framework DiskArbitration -framework IOKit -framework QtQml -framework QtNetwork -framework QtOpenGL -framework QtWidgets -framework QtSvg -framework OpenGL -framework AGL 
19:48:28: The process "/usr/bin/make" exited normally.
19:48:28: Elapsed time: 00:13.
```

hipersayanX, 2017-09-27:
_git clean -xfd_ the project folder and rebuild. Also next time use [gist](https://gist.github.com/) to paste long outputs.

theonlygusti, 2017-09-27:
What do I need to do now? I git cleaned and rebuilt.

Exactly the same situation as before.

hipersayanX, 2017-09-27:
Now you must see the full build log, past it to [gist](https://gist.github.com/).

theonlygusti, 2017-09-27:
How do I see the full log?

hipersayanX, 2017-09-27:
In the **Compile Output**.

theonlygusti, 2017-09-27:
https://gist.github.com/theonlygusti/282ea3b76fe194031aa98c691b695cb4

hipersayanX, 2017-09-27:
The problem is that you are not doing a clean build, you need to:

    cd webcamoid
    git clean -xfd

and then press the build button and the paste that output. You must see a log similar to [this](https://travis-ci.org/webcamoid/webcamoid/jobs/280193575).

theonlygusti, 2017-09-27:
I did that before this build. I will screen record this time.

hipersayanX, 2017-09-27:
Wait, you did cloned the repository or just downloaded the bare source code?

theonlygusti, 2017-09-27:
I cloned it.

here's the latest

https://gist.github.com/theonlygusti/8429ac1a1f0fd41f46bd58328e799376

I will send video in a second

theonlygusti, 2017-09-27:
https://youtu.be/7__yMeD9Mic

hopefully it helps

hipersayanX, 2017-09-27:
Try configuring Build & Run like this:

[image]

Note that **Default build directory** must be ".".
And in **Build Settings**, disable shadow build. Then git clean and build, I hope it works now 😕.

theonlygusti, 2017-09-27:
Ok it works now :)

hipersayanX, 2017-09-27:
Good! 😄
I will update the documentation later then.

hipersayanX, 2017-10-04:
Wiki page updated.

closed 2017-10-04 by hipersayanX