webcamoid/webcamoid #83: App doesn't work at all (no windows launched) on MacOS 10.12.3

SandroGrzicic, 2017-10-13:
I tried both the Portable and Installer versions, and when the app is launched, I see the app icon in the dock but no windows are launched. When I right click on the app icon in the dock, and press Show all windows, I get No available windows.

hipersayanX, 2017-10-13:
I tested in OS X 10.12.4 and it's working in my side, maybe updating your Mac will solve this problem? 😕

SandroGrzicic, 2017-10-16:
I updated to 10.12.6, the same thing happens.

closed 2017-10-16 by SandroGrzicic

hipersayanX, 2017-10-16:
Updated minimum requirements for Mac, thanks!

theswamy, 2020-04-13:
I'm unable to install it on MacOS since the system says it isnt from a trustedd source. I have allowed Apps from Appstore and Known Developers in my MacOS settings. Appreciate any help.

System: MacOS Catalina 10.15.5 Beta (19F53f)
[image]

hipersayanX, 2020-04-16:
@theswamy right click, open.