webcamoid/webcamoid #85: Question : RTSP Streaming Latency

doumdi, 2017-10-16:
Hi,

I would like to know if there is a way to reduce the latency when streaming cameras from RTSP sources ?

I get about 2 seconds of latency when adding a source with the following URL :

rtsp://user:password@192.168.0.44:554/live.sdp

Thanks!

hipersayanX, 2017-10-17:
Right now, no, but it is certainly possible, I need to to map the options to control the input buffer size to the GUI. Current input buffer size is 15MB.

doumdi, 2017-10-17:
Thanks for the quick answer! I have compiled the project following the build instructions on Windows.  However, I always have issues with AkQml not found when running from QtCreator. This is probably related to a PATH problem. Did I miss build instructions somewhere? Also, I can maybe test the change before you put too much energy into this. I figured this would be somewhere in the VideoCapture plugin. Is that the case?

hipersayanX, 2017-10-17:
When you are developing Webcamoid, plugins, qml and library paths differs from final release paths, so in that case you must pass the command line parameters to tell Webcamoid where are located those files. If you have Webcamoid already opened, go to _Projects > Build & Run > "The kit you are using to building Webcamoid" > Run_ and in **Command line arguments** write:

    -r -q "%cd%/webcamoid/libAvKys" -p "%cd%/webcamoid/libAvKys/Plugins"

Replace **%cd%** with the path where webcamoid folder is located.

Will update the wiki later.

doumdi, 2017-10-17:
You were referring to this ?https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VideoCapture/src/ffmpeg/src/convertvideoffmpeg.cpp#L150 
It seemed to me that this is just a buffer limit for accumulating ffmpeg packets and would not change anything in the latency as you decode frames as fast as you can in ConvertVideoFFmpeg::packetLoop and process the frame in ConvertVideoFFmpeg::dataLoop when a frame is available. However, I wonder if the time stamp could be the problem and the output to the Virtual driver delayed.

hipersayanX, 2017-10-17:
You are capturing an RTSP stream, so in this case you are using [MultiSrc](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/MultiSrc/src/ffmpeg/src/mediasourceffmpeg.cpp#L55), VideoCapture is for webcam streams only, MultiSrc is for local and remote files.
There you have [one thread](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/MultiSrc/src/ffmpeg/src/mediasourceffmpeg.cpp#L286) reading incoming packets and distributing to it's respective stream manager, so the the buffer will fill slow or faster depending on your connection speed. 
Then each stream manager will create two threads, one for [packet decoding](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/MultiSrc/src/ffmpeg/src/abstractstream.cpp#L269), and the other for [packet syncing and sending](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/MultiSrc/src/ffmpeg/src/abstractstream.cpp#L300), the decoding will add another delay, the syncing may also add some delay there. The virtual camera and the effects, if activated, may add some more delay.
It may require some tweaks here and there, but I have other priorities right now.

hipersayanX, 2017-10-17:
I forgot to ask, are you using some Android streaming app or similar for RTSP?

doumdi, 2017-10-17:
> I forgot to ask, are you using some Android streaming app or similar for RTSP?

 No, I am using a PTZ IP camera. 

> It may require some tweaks here and there, but I have other priorities right now.

Don't worry about it, I will try to compile and test it myself and come back to you if I find something. 

My goal is to make use of the IP/PTZ camera while videoconferencing with WebRTC / Skype / etc. This is where the virtual driver comes handy because they do not support IP cameras directly.

Thanks for the answers. Thank you also for your great software.

doumdi, 2017-10-23:
I have made some progress and removed the latency with RTSP sources. See [here](https://github.com/doumdi/webcamoid/tree/rtsp-dev). This needs more work, but basically it shows that the latency is due to the handling of pts.  Right now, I just ignore the time and send the data as fast as possible.

I have also converted some parts of the project to CMake. I find CMake easier to use and more flexible.  If you are interested, I could add CMake support to  the whole project.  

My goal is to do a [stand alone C++ minimalist console application](https://github.com/doumdi/webcamoid/tree/rtsp-dev/libAvKys/Demo) to understand the framework better using only the MultiSrc and VirtualCamera plugins (and submodules) and link them together to produce images for the virtual driver (dshow filter).  Which plugins should be compiled for a "bare minimum" system? Do you have any C++ example that I could look at for this kind of setup? 

I will continue my little experiment and keep you informed.

Thanks!

hipersayanX, 2017-10-24:
Great!

About CMake, please don't do that, I thought to do the same, but at the end it does not provide any real advantage to what Qmake offers, or at least I did not found any flaw there, and Qmake integrate very well with Qt Creator, I don't want to lose all the advantages offered by Qt framework. At least.

About the bare minimum, you need [libAvKys](https://github.com/webcamoid/webcamoid/tree/master/libAvKys/Lib), [MultiSrc and FFmpeg submodule](https://github.com/webcamoid/webcamoid/tree/master/libAvKys/Plugins/MultiSrc/src/ffmpeg), [VirtualCamera and FFmpeg and DirectShow submodules](https://github.com/webcamoid/webcamoid/tree/master/libAvKys/Plugins/VirtualCamera/src).
Almost all code required to load, connect and interact with plugins is more ore less documented in [AkElement](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Lib/src/akelement.h), if you are familiar with GStreamer programming, then you will catch very easily how libAvKys works, because it has a very similar interface.

    // Create the elements, no init code required.
    auto src = AkElement::create("MultiSrc");
    auto dst = AkElement::create("VirtualCamera");

    // Link the elements.
    src.link(dst);

    // start the pipeline.
    dst.setState(AkElement::ElementStatePlaying);
    src.setState(AkElement::ElementStatePlaying);

Each element has an iStream slot where it receives the data/frames, and an oStream signal where it send the processed data/frame to the next element. Everything else is basically interacting with the element like any other QObject. I didn't documented the API, because it is not stable at all, and I did not defined yet how script plugins will fit in this, and how it will be able to interact with everything else.

Hope it helps 😄

doumdi, 2017-10-24:
Thanks for the info. My only concern is how to set the media adequately for the MutiSrc AkElement. Since we have only a pointer to the base class, I have to cast the pointer to [AkMultimediaSourceElement](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Lib/src/akmultimediasourceelement.h) to access the [setMedia](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Lib/src/akmultimediasourceelement.h#L51) function.  I would rather try to avoid such a cast. Do you have any idea how this should be accomplished?

hipersayanX, 2017-10-24:
No cast needed, just set the object properties:

    auto src = AkElement::create("MultiSrc");
    src->setProperty("media", "video.webm");
    src->setProperty("loop", true);          // Loop the video/media if you need it.
    src->setProperty("showLog", true);       // Show play log in console, similar to MPlayer and ffplay.

doumdi, 2017-10-24:
Thanks! Of course! I was not thinking Qt / Meta-Object enough ;)

hipersayanX, 2017-10-24:
Exactly, is all designed to take advantage of Qt and C++ at 100%, it make much simpler to write the same implementation that exists in GStreamer, DirectShow and AvFoundation without all the code madness 😉

doumdi, 2017-10-25:
I got it working [link](https://github.com/doumdi/webcamoid/tree/rtsp-dev). Thanks for your  help.

hipersayanX, 2017-10-25:
Good! so [this](https://github.com/doumdi/webcamoid/blob/5d906e9a16aa5927ff91b2c07003959f437d4160/libAvKys/Plugins/MultiSrc/src/ffmpeg/src/mediasourceffmpeg.cpp#L637) is what control the delay, right?

doumdi, 2017-10-25:
You can set some options for ffmpeg/rtsp that may change the latency. But the biggest change is [here](https://github.com/doumdi/webcamoid/blob/rtsp-dev/libAvKys/Plugins/MultiSrc/src/ffmpeg/src/videostream.cpp#L135). I removed completely the time stamp check and output the data as soon as possible.

hipersayanX, 2017-10-25:
The solution would be adding the virtual getter and setters to enable disable video sync [here](https://github.com/doumdi/webcamoid/blob/rtsp-dev/libAvKys/Plugins/MultiSrc/src/mediasource.h)(enabled by default), implement in mediasourceffmpeg.h, and then map to multisrcelement.h.
An finally, map the new option to the [UI](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/MultiSrc/share/qml/main.qml).

Remember to keep in sync with the master, because right now I switching to QtQuick Controls 2 and your main.qml is affected by this, everything else is ok.

petergerten, 2018-02-25:
@doumdi Great work. Did you get further on setting up a commandline app for RTSP->virtual camera only? I am also very interested in that.

doumdi, 2018-02-25:
I hava made a small app here: https://github.com/doumdi/webcamoid/blob/rtsp-dev/libAvKys/Demo/main.cpp

petergerten, 2018-02-25:
@doumdi great, thanks. Will check it out :-)

petergerten, 2018-02-25:
@doumdi I see that you converted the entire project to cmake. Do you have a suggestion on  how to build this the easiest way? I am not very experienced with the build process on windows.

doumdi, 2018-02-26:
@petergerten, Let's continue the discussion here: https://github.com/doumdi/webcamoid/issues/1