webcamoid/webcamoid #86: libuvc permissions

reagentoo, 2017-10-22:
Please specify the required access permissions for work through libuvc in the Linux (Wiki or README.md).
```
$ webcamoid 
CaptureLibUVC: Access denied
```

hipersayanX, 2017-10-22:
It's specified in [libuvc wiki](http://wiki.ros.org/libuvc_camera) in **Usage** section, and also in [capturelibuvc.h](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VideoCapture/src/libuvc/src/capturelibuvc.h#L29), copied from there:

>libuvc requires RW permissions for opening capturing devices, so you must
>create the following .rules file:
>
>/etc/udev/rules.d/99-uvc.rules
>
>Then, for each webcam add the following line:
>
>SUBSYSTEMS=="usb", ENV{DEVTYPE}=="usb_device", ATTRS{idVendor}=="XXXX", ATTRS{idProduct}=="YYYY", MODE="0666"
>
>Replace XXXX and YYYY for the 4 hexadecimal characters corresponding to the
>vendor and product ID of your webcams.

reagentoo, 2017-10-22:
Ok. But for the average user it's enough to do
`usermod -a -G usb user`

hipersayanX, 2017-10-22:
Thank, I did not know!

reagentoo, 2017-10-23:
Although this is not very safe :) The best way to add 99-uvc.rules to libuvc upstream, containing a list of all possible uvc devices (for example for group 'uvcdev').

hipersayanX, 2017-10-24:
In that case I think is much better to fill a pull request or an issue in [libuvc](https://github.com/ktossell/libuvc) hub.

closed 2018-12-27 by hipersayanX