webcamoid/webcamoid #87: No video displayed from two available webcams

daveandes, 2017-10-22:
When I run Webcamoid 8.1.0 it fails to display webcam video from both the integrated webcam on my HP Zbook 15 G2 laptop and a Logitech C920 USB webcam.

I downloaded the source code ZIP file and built the application myself; here is the compile output displayed in Qt Creator: [webcamoid_compile.log](https://github.com/webcamoid/webcamoid/files/1405309/webcamoid_compile.log)

When I launch the application, the following is shown in the Application Output:

`Starting C:\Users\D01067561\Desktop\webcamoid\StandAlone\webcamoid.exe...
QML debugging is enabled. Only use this in a safe environment.
QObject::connect: No such signal MediaSource::oStream(const AkPacket &) in multisrcelement.cpp:214
QObject::connect: No such signal MediaSource::error(const QString &) in multisrcelement.cpp:219
QObject::connect: No such signal MediaSource::maxPacketQueueSizeChanged(qint64) in multisrcelement.cpp:223
QObject::connect: No such signal MediaSource::showLogChanged(bool) in multisrcelement.cpp:227
QObject::connect: No such signal MediaSource::loopChanged(bool) in multisrcelement.cpp:231
QObject::connect: No such signal MediaSource::mediasChanged(const QStringList &) in multisrcelement.cpp:235
QObject::connect: No such signal MediaSource::mediaChanged(const QString &) in multisrcelement.cpp:239
QObject::connect: No such signal MediaSource::streamsChanged(const QList<int> &) in multisrcelement.cpp:243
file:///C:/Programs/Qt/5.9.1/mingw53_32/qml/QtQuick/Dialogs/DefaultFileDialog.qml:210:25: QML ListView: Binding loop detected for property "model"
file:///C:/Programs/Qt/5.9.1/mingw53_32/qml/QtQuick/Dialogs/DefaultFileDialog.qml:210:25: QML ListView: Binding loop detected for property "model"
qrc:/Webcamoid/share/qml/OutputConfig.qml:39:5: QML GroupBox: Binding loop detected for property "implicitWidth"
qrc:/Webcamoid/share/qml/OutputConfig.qml:39:5: QML GroupBox: Binding loop detected for property "implicitWidth"
qrc:/Webcamoid/share/qml/OutputConfig.qml:39:5: QML GroupBox: Binding loop detected for property "implicitHeight"
qrc:/Webcamoid/share/qml/OutputConfig.qml:39:5: QML GroupBox: Binding loop detected for property "implicitHeight"
qrc:/Webcamoid/share/qml/OutputConfig.qml:39:5: QML GroupBox: Binding loop detected for property "implicitWidth"
qrc:/VideoCapture/share/qml/main.qml:168: ReferenceError: createControls is not defined
qrc:/VideoCapture/share/qml/main.qml:220: ReferenceError: controlsUpdated is not defined
qrc:/VideoCapture/share/qml/main.qml:221: ReferenceError: controlsUpdated is not defined
qrc:/VideoCapture/share/qml/main.qml:222: ReferenceError: createInterface is not defined
qrc:/VideoCapture/share/qml/main.qml:220: ReferenceError: controlsUpdated is not defined
qrc:/VideoCapture/share/qml/main.qml:220: ReferenceError: controlsUpdated is not defined
qrc:/VideoCapture/share/qml/main.qml:221: ReferenceError: controlsUpdated is not defined
qrc:/VideoCapture/share/qml/main.qml:221: ReferenceError: controlsUpdated is not defined
QImage::scaled: Image is a null image`

When I go to Configure Sources and toggle between the two cameras, I see that the corresponding camera's activity light will come on for a fraction of a second and then turn off. I surmise the program is attempting to connect to the camera, but something fails in that process and the light immediately turns back off.

Environment:
- Windows 10 Enterprise
- Webcamoid 8.1.0
- FFmpeg 3.3.4
- Qt 5.9.1 / Qt Creator 4.4.0
- GCC 5.3.0

hipersayanX, 2017-10-24:
You must configure the plugins search paths, [see here](https://github.com/webcamoid/webcamoid/issues/85#issuecomment-337098539). Will update the wiki later.

hipersayanX, 2018-07-05:
[wiki](https://github.com/webcamoid/webcamoid/wiki/Build-and-install-in-Windows) updated. Any chance to test it again?

markummitchell, 2018-11-06:
Hello,
I have, I believe, the same issue. No video shows in the user interface. Console errors are:

QObject::connect: No such signal MediaSource::oStream(const AkPacket &) in multisrcelement.cpp:237
QObject::connect: No such signal MediaSource::error(const QString &) in multisrcelement.cpp:242
QObject::connect: No such signal MediaSource::maxPacketQueueSizeChanged(qint64) in multisrcelement.cpp:246
QObject::connect: No such signal MediaSource::showLogChanged(bool) in multisrcelement.cpp:250
QObject::connect: No such signal MediaSource::loopChanged(bool) in multisrcelement.cpp:254
QObject::connect: No such signal MediaSource::mediasChanged(const QStringList &) in multisrcelement.cpp:258
QObject::connect: No such signal MediaSource::mediaChanged(const QString &) in multisrcelement.cpp:262
QObject::connect: No such signal MediaSource::streamsChanged(const QList<int> &) in multisrcelement.cpp:266

Environment
•Windows Server 2016 (make sense if you know the weird constraints I am working under)
•Webcamoid 8.0.0 and 8.1.0 were both tried
•FFmpeg 3.2.4
•Qt 5.11.2
•Visual Studio 2015 (64 bit)
•OpenSSL 1.0.2p (I put libeay32.dll and ssleay32.dll from https://indy.fulgan.com/SSL/ into Standalone directory)

I am quite willing to look into the code. Before that, I was wondering about the MediaSource class appearing in both Standalone/src/mediasource.h and libAvKys/Plugins/MultiSrc/src/mediasource.h. Maybe there is some confusion? Neither one offers the complete set of signals listed in the errors above.

Thanks for the nice tool. I really like the user interface. And the build went pretty nicely.

markummitchell, 2018-11-06:
I should mention that the command line options are, I think, properly set as
...
-r -p C:\users\Administrator\webcamoid\libAvKys\Plugins  -q C:\users\Administrator\webcamoid\libAvKys
...

markummitchell, 2018-11-06:
Good clues - downloaded installer version has same problems in Windows Server but works great in Windows 7. I will try building in Windows 7 and compare operation in the two operating systems.

hipersayanX, 2018-11-07:
@markummitchell, did you enabled shadow build? if that's the case the paths must point to _C:\users\Administrator\\[SHADOW FOLDER]_ instead of _C:\users\Administrator\webcamoid_, [here](https://github.com/webcamoid/webcamoid/wiki/Build-and-install-in-Windows#build-from-qt-creator) is the documentation.

markummitchell, 2018-11-07:
The immediate problem was that 'Update drivers' said I was ok, but installing hardware specific drivers from Logitech got the camera working. I still get the MediaSource signal errors, but will not assume they are the source of any serious problems. 

With the prepackaged 8.1.0 I get a crash after about 5 seconds of recording for AVI and MP4 formats.

With my personal build of 8.1.0 (I upgraded from 8.0.0) I do not get any of the options for selecting AVI versus MP4 versus others, which prevents me from getting to the point where I can record. I will investigate further.

I was not using a shadow build so the command line was fine. Although - the Build and Run page for Windows says the first argument is '--vcam' but I think the code expects '-vcam'. I removed the '-vcam <value>' portions.