webcamoid/webcamoid #89: Segmentation fault on FreeBSD 12

zhou0, 2017-10-23:
QKqueueFileSystemWatcherEngine::addPaths: kevent: Invalid argument
Segmentation fault

hipersayanX, 2017-10-24:
What Webcamoid version?

zhou0, 2017-10-24:
8.1.0

hipersayanX, 2017-10-24:
I did tried sometime ago building Webcamoid 8.0.0 in TrueOS, and it worked ok, but somewhere in 8.1.0 development, I had the exact same issue as you, it's maybe something related to Qt, any possibility for you to help debug it?

yurivict, 2017-12-01:
kevent is the bug in FreeBSD: https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=224011
Until it is fixed, /dev/video0 can't be monitored using kqueue.

hipersayanX, 2017-12-01:
@yurivict, thanks a lot! while this is solved, a solution would be patching your Wecamoid package and comment lines 69-78, 85, 1179-1180 and 1184-1185 in [capturev4l2.cpp](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VideoCapture/src/v4l2sys/src/capturev4l2.cpp). [audiodevoss.cpp](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/AudioDevice/src/oss/src/audiodevoss.cpp) is also affected by this, but since this submodule is not really important you can patch it or just disable with NOOSS=1.

yurivict, 2017-12-01:
@hipersayanX  Thanks for the tip.

I am worried about https://github.com/webcamoid/webcamoid/issues/100, it prevents me from creating the FreeBSD port.

yurivict, 2017-12-01:
Thanks for your tip!

I think, it would have been even better if you added build variables like NO_DEVICE_KEVENT_VIDEO and NO_DEVICE_KEVENT_OSS.

hipersayanX, 2019-01-22:
Fixed in 8.5.0.

closed 2019-01-22 by hipersayanX