webcamoid/webcamoid #90: Video not outputting

TheCNixon, 2017-10-23:
I'm trying to use a picture or video source on the output virtual camera, but the virtual camera is only showing the multi colored test picture even if I swap, restart the program, turn off and turn on the program, etc.

hipersayanX, 2017-10-24:
What operating system? what target program? need more info, please.

TheCNixon, 2017-10-24:
Windows 10 64x, Firefox's flash plugin is the target

hipersayanX, 2017-10-24:
First at all remove the virtual webcam in _Preferences > Output > Remove All_, then install 32 bits version of Webcamoid, then add the virtual webcam again and restart Firefox, it may be working now.

TheCNixon, 2017-10-24:
Ok, I tried to use the 32 bit version, uninstalled and reinstalled twice just to be sure with everything removed first. The issue has now gotten worse as my virtual camera doesn't even show up on firefox, same flash instance as well. I've restarted the browser and refreshed multiple times along with trying on a different browser (Brave) but the issue persists

hipersayanX, 2017-10-24:
Next version I will remove 64 bits version for sure, its just causing problems, and will add preview in **Outputs** instead of adding it to the media bar.

TheCNixon, 2017-10-24:
I don't know if this is clear but I did precisely what you put in steps.

hipersayanX, 2017-10-25:
Ok, be patient, since you made many changes, I'm trying to understand in which point are you.

Let's try this, install Webcamoid 32 bits **AND** 64 bits, yes both, install in default paths, Webcamoid 32 bit will be installed in _C:\Program Files (x86)\webcamoid\bin\webcamoid.exe_ Webcamoid 64 bit will be installed in _C:\Program Files\webcamoid\bin\webcamoid.exe_.

Give me a screen shot of the output configs, in both versions. Don't do anything more, don't touch anything else, just do that.

AhmedFaraz09, 2017-10-25:
My issue  [92](https://github.com/webcamoid/webcamoid/issues/92) was merged with this one. 
I followed the step by step guide you provided. It works now. I can play the video file on vlc or skype using the virtual camera, but now there is an other issue. The video which **vlc** plays is very gittery. The resolution of my video is **320x240** where as when vlc plays it via virtual camera it shows a resoultion of **1920*1080**.
There is also an issue with skype. I can't  make a video call after selecting virtual camera as my video device. Whenever I click on the button to enable the webcam during a call it keeps getting back in disable state. My skype version is 7.40.0.103
I also tried the latest version of skype but there is no option to select a webcam in it.

TheCNixon, 2017-10-25:
Seems to have been a bug, if I let the program crash and start it back up then turn off the video and turn it back on, it will end up working.

TheCNixon, 2017-10-25:
Though one issue still persists, the virtual webcam doesnt show up from the 32 bit version on firefox, only in chrome or discord.

hipersayanX, 2017-10-25:
@TheCNixon Ok, I've tested all programs, and now I can confirm all the problems, basically Windows is a mess of x86 and x64 programs, so some programs works with 32 bits versions, other with the 64 bits. I did tested the virtual webcam with [this demo](https://davidwalsh.name/demo/camera.php), and I can confirm it works well with Webkit/Blink based browsers, and freezes Firefox (tested with x86 version before and it worked well).

@AhmedFaraz09, the virtual webcam offers some predefined resolutions due to some restrictions in DirectShow format negotiations, end VLC probably selected the best resolution for you, try setting a different resolution in VLC. About Skype, I will need to make more tests on it.

In resume, solving it will be a pita.

Thanks both for reporting for the advice! 😄

TheCNixon, 2017-10-26:
I'm normally using Linux, so I'll have to try your software out on there. Windows is trash.

hipersayanX, 2018-03-13:
Progress report so far. This is more or less a list of programs I been tested and their status:

**Firefox and derivatives** :green_heart:: Working with both WebRTC and Flash Player.
**Chrome and derivatives** :green_heart:: Working with WebRTC.
**Chrome + Flash Player** :heart:: Not working, it does not even work with the integrated real camera, so I don't care.
**Skype desktop** :green_heart: : Working.
**Skype UWP (from Microsoft Store)** :heart: : Not working, my suspicion here is either because it uses Media Foundation instead of DirectShow (DirectShow filters does not works in Media Foundation), or due to sandboxing restrictions (like code signing), or both.
**VirtualBox** :heart:: It's listed but not working, the filter is never loaded. VirtualBox create a fake device that only show a blue frame, with some predefined formats and resolutions not even related to the configured one. In logs files it show **TrustedInstaller is not the owner** and when reassigned the filter to that user it shows something like filter is not signed (don't remember the exact message). Probably share some similarities with Skype UWP.
**VLC, OBS, Pidgin, aMSN and others** :green_heart:: and many other programs, working.

Now I need to write the server and the IPC mechanism to multiplex the frames from Webcamoid to other programs, and control the virtual camera.
Also, after some experiments it may be possible to create many virtual camera devices in the same way I did with GNU/Linux and Mac.

ROBERT-MCDOWELL, 2018-03-13:
> Chrome + Flash Player
the recent chrome versions blocks flash and devices by default.
it needs to allow first flash and then confirm again in parameters camera + mic, in some situation it also needs to put the website in whitelist

hipersayanX, 2018-03-13:
@ROBERT-MCDOWELL I did so, when enter a site it ask me to enable Flash Player, then it ask me to give permissions for the webcam, when I go to the webcam configuration page in FP it show the combobox with a blank option where is supposed to be the name of the integrated webcam, the virtual camera doesn't get listed there and the plugin is not even loaded. Also the site appears white listed by Chome. So I'm not sure if it's a bug in Chrome or FP :confused: 

Anyway, because the virtual camera works flawlessly with WebRTC in Chrome and Firefox, and Flash Player is soon to be dead I don't much care.

ROBERT-MCDOWELL, 2018-03-13:
> Flash Player is soon to be dead,
not really, FP will be in webassembly.... And many many companies still using flash because of various reasons, and for long time (I'm part of)
what version of Chrome?

hipersayanX, 2018-03-13:
@ROBERT-MCDOWELL 65.0.3325.162 (the latest one). This is what it show, the blank space is where is supposed to be the webcam name, as it does with Firefox.

[image]

Tested [here](http://www.testwebcam.com/).

ROBERT-MCDOWELL, 2018-03-14:
ok I had some issues with chrome 65, there is version 67 which should fix the problem (maybe, I'm in contact with adobe team who are in contact with chrome team)

hipersayanX, 2018-03-16:
I want to add this [link (Applicability of Virtual DirectShow Sources)](http://alax.info/blog/1722) that explains pretty much well all the situation related to virtual cameras in Windows, and all 32 bits, 64 bits and MMF mess.

ROBERT-MCDOWELL, 2018-03-16:
looks good indeed

arpcpro, 2018-03-19:
I have tried just now 8.1.0 and 7.2.1 both 64 bits on windows 8.1, and I cannot change the media file to play. I have used webcamoid in the past, I don't remember which version and it worked. I can add the virtual webcamera but I can only edit the description, not the file location. How are you going around this?
I've changed directly in the registry to the file location but it did not work.

hipersayanX, 2018-03-20:
@arpcpro, what do you mean with file location? do you want to play some video file? does it show at least the image pattern in the target program? what's the target program?

arpcpro, 2018-03-20:
@hypersayanX,  I was also on issue 92 and was redirected to this one. The target program is "graphstudionext64.exe". It creates an activemovie window with the test pattern. I've tried other program and it also displays the pattern.
I've also did some test and at a certain point is was asking for a password.

hipersayanX, 2018-03-20:
@arpcpro, yeah it seems you are affected by the same problem (I'm working right now to fix that), meanwhile, a hacky solution would be downloading both 32 and 64 bits portable Webcamoid, and [identifying which bitness](https://www.itechtics.com/4-ways-check-windows-program-32-bit-64-bit/) is the target program, if the target is 32 bits use the 32 bits Webcamoid, if it's 64 bits use the 64 bits Webcamoid. You can use Graph Studio Next to install the virtual camera filter that is inside webcamoid/bin/VirtualCameraSource.dll.
After you did that just play the source you want to send in Webcamoid.

hipersayanX, 2018-07-05:
Virtual webcam working, [here](https://github.com/webcamoid/webcamoid/issues/116#issuecomment-402799332) is more or less the change log. The virtual camera will work with any DirectShow based program, in the case of Skype you must use **Skype for Desktop**, Skype from Windows Store will not work for this.
Also the new virtual webcam works with **Wine** :green_heart:, for Windows programs, obviously.

graywolf336, 2019-01-20:
@hipersayanX do you have any developmental/nightly builds people can help test? Sadly I can't get the local items to build, so I can't help out that way. I am willing to help test Windows and MacOS. Thanks!

hipersayanX, 2019-01-21:
@graywolf336 I need to configure deployment, will check that after release.

hipersayanX, 2019-01-22:
Fixed in 8.5.0.

closed 2019-01-22 by hipersayanX

graywolf336, 2019-01-22:
@hipersayanX awesome! Thank you for all your hard work, this project is amazing!!