webcamoid/webcamoid #92: How to set a video file as virtual camera

AhmedFaraz09, 2017-10-24:
I want a video file to detected as a virtual camera on **Skype** 
I am using **windows 10 64-bit**. I have tried both 32-bit and 64-bit versions of **webcamoid** but I can't figure out how to do it. When I try to add a virtual camera in **preferences->output**, The only option which is available is of Description and there is no option to specify a video path.

hipersayanX, 2017-10-24:
Just add the camera, it will output whatever media you have selected for play.

AhmedFaraz09, 2017-10-24:
I added the camera but the edit button is disabled.
[image]

hipersayanX, 2017-10-24:
That's the virtual webcam, now you must select the file, desktop, or webcam you want to play, in the left bar, that is what other programs will see.

AhmedFaraz09, 2017-10-24:
Ok i tried that I added a file in the left bar and played it 
[image]

but Skype and vlc still play's that same image. 
[image]
 
I even tried VLC it also plays the same image when I select **TestCam** the (virtual camera i created in webcamoid)

hipersayanX, 2017-10-24:
You problem is the same as @TheCNixon in #90 and it was opened before, let close this and continue there to centralize information, ok?

closed 2017-10-24 by hipersayanX