webcamoid/webcamoid #93: Crash on start record

reagentoo, 2017-10-25:
**Platform:**
- Gentoo Linux ~amd64
- Qt-5.9.2
- ffmpeg-3.4
- libv4l-1.12.5

**Core dump:**
```
#0  0x00007fffcd325358 in avcodec_fill_audio_frame () from /usr/lib64/libavcodec.so.57
#1  0x00007fffc6026f72 in ConvertAudioFFmpegAV::convert (this=0x555555a70b50, packet=...)
    at src/convertaudioffmpegav.cpp:131
#2  0x00007fffc6231f8e in ACapsConvertElement::iStream (this=0x555555a2f1f0, packet=...) at acapsconvertelement.cpp:87
#3  0x00007ffff7b98f8b in AkElement::iStream (this=0x555555a2f1f0, packet=...) at src/akelement.cpp:854
#4  0x00007ffff7b92870 in AkElement::operator() (this=<optimized out>, packet=...) at src/akelement.cpp:818
#5  0x00007fffc4864edf in AudioGenElement::readFramesLoop (this=0x555555a2caa0) at src/audiogenelement.cpp:220
#6  0x00007fffc4865d3b in QtConcurrent::RunFunctionTask<void>::run (this=0x5555606c08a0)
    at /usr/include/qt5/QtConcurrent/qtconcurrentrunbase.h:136
#7  0x00007ffff61b3ccf in ?? () from /usr/lib64/libQt5Core.so.5
#8  0x00007ffff61b7960 in ?? () from /usr/lib64/libQt5Core.so.5
#9  0x00007ffff51cea43 in start_thread () from /lib64/libpthread.so.0
#10 0x00007ffff580190f in clone () from /lib64/libc.so.6
```

**Full backtrace:**
https://gist.github.com/reagentoo/ff8479b14909871aac3dae29e12227c8

reagentoo, 2017-10-25:
And also, under certain conditions webcamoid freezes after recording. The recording almost never ends successfully. :(

hipersayanX, 2017-10-25:
The problem seems to be in in AudioGen. try some of this:

- Set the audio input to the mic.
- Select a video and set the input to the video.
- And last disable audio recording.

Then try recording and test if still crash in these cases, try several times to be sure.

reagentoo, 2017-10-25:
Ok, I confirm that if the microphone is not set up, webcamoid crashs.

reagentoo, 2017-10-25:
So, I can not get microphone working. Webcamoid was build without pulseaudio and jack.
`[AudioData @ 0x7f47a57f92e0] invalid NULL pointer for src[0]`

hipersayanX, 2017-10-25:
Webcamoid works with ALSA and OSS too.

reagentoo, 2017-10-25:
I tested with all "audio capture/play" available backends (alsa, jack, pulseaudio, qtaudio). There is no positive result. Camera:
`Bus 003 Device 008: ID 046d:081b Logitech, Inc. Webcam C310`

hipersayanX, 2017-10-25:
And what about disabling audio recording?

reagentoo, 2017-10-25:
It seems without audio only video record works normal.

reagentoo, 2017-10-25:
The audio settings have been selected "HDA Intel PCH" instead of "UVC Camera". But I think that it should not lead to segfault in any way. And also by choosing the right audio source, I can not get a microphone to work correctly. A sound is either accelerated or there is none at all. :(
I would very much like that webcamoid would choose the most optimal settings automatically.

hipersayanX, 2018-12-29:
Crashing straightforward if using ffmpegsw in ACapsConvert, and leaking memory if using gstreamer, did not tried with ffmpegav. Recording with audio enabled is totally unusable. Hopefully it will not take that much time to fix.

hipersayanX, 2019-01-22:
Fixed in 8.5.0.

commit 164f3763 (2018-12-27): Check that ACapsConvert is valid in AudioGen, and frame size is ok in
ffmpegav (maybe this could fix #93?).
closed 2019-01-22 by hipersayanX