webcamoid/webcamoid #94: webcamoid appimages icon missing

Lvaskz, 2017-11-02:
hi, i am using webcamoid, but the icon is not appearing  in the system, can you provide one?
here i get one
[image]
[image]

hipersayanX, 2017-11-03:
Appimages are ELF executable binaries, and ELFs does not have an icon by design. Showing an icon depends absolutely on desktop implementation, I also remember there where some trick to show appimages icons with appimaged or something like that, but is much better if you as in their [official repository](https://github.com/AppImage/AppImageKit).

Lvaskz, 2018-02-23:
hi, i am using appimaged ( daemond ) and the icon in the app menu is missing. The app is in the app menu but not the icon. Can you ask some help to @probono ? the app is cool

hipersayanX, 2018-02-24:
Ok, I will ask but not sure it will work.

Lvaskz, 2019-03-04:
Hi, I am using appimagelauncher https://github.com/TheAssassin/AppImageLauncher and I have the same issue. Thanks for your efforts in this.
I can assigne an Icon from( linuxmint 19.1)  cinnamon menu manually is not a real solution but it is fine to me. But It would be great if some day can be resolved 
thanks

hipersayanX, 2019-03-05:
Oh, get it, that's the program that assign the icons to the AppImages, will check.

hipersayanX, 2019-03-26:
Fixed in 8.6.1.

closed 2019-03-26 by hipersayanX

Lvaskz, 2019-03-28:
thanks, works fine