webcamoid/webcamoid #95: It`s impossible to set out resolution of recorded video.

lostinwoods, 2017-11-12:
Webcamoid 8.1.0 for Linux on x64 desktop *buntu 16.04
It doesn`t matter what resolution and format I choose for webcam, I get 640x480 videos in result. I have c920 and want to get HD and fullHD recordings, but unfortunately that doesn`t work (
Another question is concerning default settings - is it possible to save my setup? Unfortunately I have to adjust everything each time I open the program.

hipersayanX, 2017-11-13:
If the preview resolution is ok, then it's most probably because the recording part is not aware of the resolution change, will check that.

>Another question is concerning default settings - is it possible to save my setup? Unfortunately I have to adjust everything each time I open the program.

Is already reported here #62, I will be working soon on that.

lostinwoods, 2017-11-13:
Yes, preview resolution is ok. Right now I checked Webcamoid 8.1.0 x64  portable version on win7 computer. And it has the same problem.

lostinwoods, 2017-12-28:
Is there a way to solve the problem modifying config file?

hipersayanX, 2017-12-28:
Right now, no, the recording plugin copy the resolution from source, if you change the source while recording it will (or may be) continue recording with the resolution of the old source.
I will add an option for configuring the output resolution for sure.

hipersayanX, 2019-01-22:
Fixed in 8.5.0.

closed 2019-01-22 by hipersayanX

holmesrichards, 2020-05-24:
I see the same issue in 8.7.1. Preview resolution is correct but output remains at 640x480.