webcamoid/webcamoid #97: Building on High Sierra or using virtual cam

SnelsSM, 2017-11-20:
Hi. Hepl me please!
I tried to use 8.1.0, but the virtual camera does not work in 8.1.0 with High Sierra.
Then I tried to build from source 8.5, but:
```
src/plugin.cpp:28:5:` error: use of undeclared identifier 'Ak'
    AkLoggerStart("~/AkVirtualCameraLog.txt");
    ^
src/logger.h:26:37: note: expanded from macro 'AkLoggerStart'
    #define AkLoggerStart(fileName) Ak::Logger::start(fileName)
```
How can I use virual cam?

hipersayanX, 2017-11-20:
Hi, the virtual camera is currently WIP, it will only able to install the plugin, but only a test dot pattern will show. I think it will took 1 or 2 weeks to be ready, maybe. 
Any way if you want to continue, you must edit [logger.h](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VirtualCamera/src/cmio/VirtualCamera/src/logger.h#L26) file and change the lines with **Ak::Logger** to **AkVCam::Logger**. I will push the changes later to fix this.

hipersayanX, 2017-11-20:
It may compile now.

closed 2017-11-20 by hipersayanX

SnelsSM, 2017-11-21:
Yes, now it's compiled. Thanks. But
[image]
If I try to run from StandAlone/webcamoid, mac os says "Can't open webcamoid bla-bla-bla" =((

hipersayanX, 2017-11-21:
You must run the StandAlone executable. Also be sure you have configured your [build environment](https://github.com/webcamoid/webcamoid/wiki/Build-and-install-in-Mac) properly.

[image]

unformatt, 2018-11-27:
Wondering where I can find a High Sierra Webcamoid 8.5.0 binary?

hipersayanX, 2018-11-28:
@unformatt 8.5.0 is next release so there is no binary and not tagged yet, but you can try [building it](https://github.com/webcamoid/webcamoid/wiki/Build-and-install-in-Mac) and report any error you find.
Hopefully I can get ready before the end the year.