webcamoid/webcamoid #98: Feature Request: FreezeFrame / Pause

camAtGitHub, 2017-11-23:
Hello,
I have a plugin request.  The plugin would "freeze/pause" the most current frame (when active?)
I.E. Pausing the output to the virtual camera etc.

So for a moment pretend you are in a video conference that is sub-par, you hit 'freezeFrame' and everyone else thinks you are still watching the video conference, when you need to speak you simply de-activate the plugin.

Perhaps an improvement of the above plugin would be a 'loop' plugin that when activated it would loop a pre-determined length of video ie. 5 seconds.

Anyway thanks for the great product and consideration of the plugins :-)

hipersayanX, 2017-11-23:
## For still frame ##

I can do that, my idea is to add a new plugin in the media bar that let you choose a file in your computer, and play that file as source.

## For Video ##

Storing one frame is ok, storing a few frames is also ok, the problem comes when you want to store a whole video sequence, let make some quick calculus, suppose you are using your webcam with a very conservative profile, ie. 640x480 at 30 FPS, and you want to save 5 seconds of video, let save them as RGB, 24 bits per pixel, that's 8 bytes per pixel. So you have:

BufferSize = bpp * whidth * height * fps * t = 3 * 640 * 480 * 30 * 5 = 138.24 MB

So a sequence of 5 seconds of video will take 138.24 MB of your RAM, now let optimize this a bit, and we will store the sequence as a YUV so each pixel will require just 2 bytes, that's 640x480 for Luma plane and 320x240 for UV. So as result the video buffer will now take 92.16 MB, that's a bit better. But the problem is that if a user can choose between lowest resolution and highest one, it will always choose the highest one for a video conference because looks better, the most common HD resolution for a webcam is 1920x1280, so now the buffer will take 737.28 MB, just for 5 seconds of video, you can redo the calculus for 10, 20, 30 or more seconds. The conclusion is that it gonna eat your RAM for nothing, because the user in the other side will soon discover you are using a looped video because of the short time period.

One solution would be encoding the frames and store the encoded frames, so the buffer size will depend solely in the bitrate configured, for a good quality video a recommended bitrate is between 1 a 1.5 Mbits/s. So for a 5 minutes of video you have:

BufferSize = t * bitrate / 8 = (5 * 60) * (1 ~ 1.5) / 8 = 37.5 ~ 56.25 MB

The other solution would be recording your self as minutes as your want in a video, and then replay that video when you go out, Webcamoid can already do that, you can record a video while in a video conference, and then you can save and add that video with the green **+** button in the media bar, so no plugin development required here.

>Anyway thanks for the great product and consideration of the plugins :-)

Thanks! :smile:

hipersayanX, 2017-12-12:
It will require an encoder and decoder plugin, this will probably come after adding scripted plugins support.

OscarSilve123, 2019-11-30:
I'd also like to be able to pause a video you're playing.  Right now all you can do is stop it, then when you hit play again it starts over from the very beginning.